Xastir/Xastir #7: Weather alert test cases fail

K2DLS, 2017-08-23:
Uncompressed weather alert test cases fail because of a 2009-07-28 "Temporary change" in which all weather alerts were treated as compressed.  This broke alerts following the standard which has an underscore as the third character in the county/zone.  Backing out the change allows the test cases to succeed.

tvrusso, 2017-08-23:
As has been pointed out on the Xastir mailing list, what's actually wrong here is that the 'standard' was defined only by what was then being done by Dale Huguley back when he was maintaining the weather alert gateway.  Dale got out of that business years ago, and it was taken up in limited form by Pete Loveall.   Pete adopted a slightly different approach -- injecting only compressed weather alerts -- and that forced the "temporary" change.  

Since then, Curt has also been injecting weather alerts into firenet by following Pete's 'new standard.'

Obviously, since nobody has since taken up the task of injecting the reports that Dale used to inject, the "temporary" change is pretty much permanent.

What ought to change is the comments in the code and the test cases.  If you are going to generate your own weather alerts, you should probably follow the examples provided by Pete and Curt in their separate feeds.

K2DLS, 2017-08-23:
Agreed.

K2DLS, 2017-08-23:
Further testing has revealed that that is more than a case of documentation and test cases.

On his page at http://www.aprs-is.net/wx/, Pete states that:

zcs are comma separated zone codes or county codes (see the databases
linked to the left to find the zone or county code for your area), or
statement text.

compressed zcs is a compressed listing of zones or counties (see
example below) (note underscore in To address instead of hyphen).

I have written my alerts to use the uncompressed zcs format.  Xastir
currently only processes the first zone in the list and ignores the
remainder.  If the alert uses compressed zcs, then all the applicable
zones are extracted and included in the alert.

Uncompressed, Xastir doesn't create an alert for each zone:

K2DLS-13>APRS::NWS-WATCH:280400z,Flash
Flood,NJZ012,NJZ013,NJZ014,NJZ015,NJZ016{7G000

Compressed, Xastir does create an alert for each zone:

K2DLS-13>APRS::NWS-WATCH:280400z,Flash Flood,NJZ012>016{7G000
K2DLS-13>APRS::NWS-WATCH:290400z,Flash Flood,NJZ014-020>023{IJJ00

K2DLS, 2017-08-24:
I've fixed the issue and tested against NWS-TEST.log.  It seems to be working now as expected.

K2DLS, 2017-08-26:
Resolved by merge #8.  Closing.

closed 2017-08-26 by K2DLS