Xastir/Xastir #10: Objects not following timing settings

kikosgc, 2017-10-16:
Like the title says, objects are not following timing settings. 
[image]
[image]

tvrusso, 2017-10-16:
This is not a bug, it's exactly as the feature is intended to work.  Objects are transmitted by Xastir using a decaying rate as described by Bob Bruninga in many hard-to-find places on his web site http://aprs.org, and also in the APRS spec http://aprs.org/doc/APRS101.PDF on page 10. 

The time setting in the dialog is only the MAXIMUM transmit interval.  It does not define a fixed interval for object transmission as you are expecting.

Whenever an object is first defined OR modified, the package is transmitted immediately, then a few seconds later (the actual interval is slightly randomized so that we don't retransmit lots of objects all at the same time).  The object is retransmitted at intervals that double each time until the maximum interval is reached, after which all subsequent transmissions are at the maximum interval.

The transmit interval is reset if any data in the object is changed.

The intent is that new objects are more important information than old objects, and should be beaconed frequently until they become old.

You're not seeing all of the retransmissions in your APRS-IS snapshot because they're being discarded as duplicates until they are at least 30 seconds apart.  

The actual algorithm is coded in the file objects.c, down around line 1519.  It is heavily commented.

kikosgc, 2017-10-16:
Oh, ok. Thanks for the quick explanation.

closed 2017-10-16 by tvrusso