Xastir/Xastir #18: How to configure xastir to work with TH-D74

lnitram, 2017-12-24:
Hi,

I am trying to connect my TH-D74E using bluetooth to my Ubuntu 17.10 laptop as I did with my old TH-D72. Is there a documentation how to set the settings in TH-D74 and how to configure xastir for this?

Thanks

Martin

dranch, 2017-12-24:
Hello Martin,

Are you using your D72 in KISS mode or COMMAND mode with Xastir?  If KISS mode, I would think the D74 will just work once you get it's serial port associated to the Ubuntu laptop.  The only gotcha could be that with the Xastir radio setup script, it's trying to do a KISS ON\n RESTART\n command sequence that the D74 won't need (it's now a KISS-only TNC).

I've been meaning to try out the BT access on my new D74 here so if you remain stuck, I can write this down and add it to the Xastir pages in the next day or two.

--David
KI6ZHD

tvrusso, 2019-03-17:
This would be better discussed on the Xastir mailing list, where there is a larger community of folks who might be able to help.  Github issues are the best place for code development issues, but questions like this are unlikely to get as much attention as they deserve here.

Please subscribe to the mailing list as pointed to on the Xastir web site, http://www.xastir.org/.

closed 2019-03-17 by tvrusso