Xastir/Xastir #23: All map processing in Xastir that uses graphicsmagick uses deprecated "GetIndexes" function.

tvrusso, 2018-02-16:
GraphicsMagick deprecated the GetIndexes function in January of 2018:
http://www.graphicsmagick.org/Changelog.html

As it says in the changelog above:

magick/pixel_cache.h: Mark GetPixels(), GetIndexes(), and GetOnePixel() as deprecated. Compilers may produce a warning if these functions are used.

I have already encountered some compilers that produce such a warning.

That code should be rewritten to do something other than use GetIndexes.  This may take some investigating to find the appropriate substitute.

tvrusso, 2018-02-16:
A really quick peek at the "pixel_cache.h" file from GraphicsMagick suggests that GetIndexes is deprecated in favor of AccessMutableIndexes, which has the same argument and return type.  This *might* be a drop-in replacement.  The only question is how long ago did they introduce the latter function, and do we have to play stupid ifdef games to deal with ancient installs of GraphicsMagick, like we used to do for \<hock>\<spit>ImageMagick\</spit>\</hock>?

tvrusso, 2018-02-16:
A quick googling shows that GraphicsMagick has had the AccessMutableIndexes function for over 6 years.  That makes it extremely likely that we don't have to worry about backward compatibility to handle people using versions too old to support it.

ChocolateFrogsNuts, 2018-07-02:
It's also deprecated on ImageMagick > 669 (possibly earlier).

I'm currently working on bringing a feature I did a few years ago and never got around to sharing into the current version, and part of that is a set of changes for ImageMagick 6.9.10.
Currently at https://github.com/ChocolateFrogsNuts/Xastir/commits/DLM
I'll throw a pull request in the near future.

tvrusso, 2019-04-05:
This issue was resolved by pull request #40.

closed 2019-04-05 by tvrusso