Xastir/Xastir #24: Building Xastir produces large numbers of warnings on newer compilers

tvrusso, 2018-02-16:
This was pointed out to me last week:  Xastir throws a very large number of compiler warnings on newer versions of GCC (specifically, GCC 6.x and newer).  It also throws a lot of warnings from Clang (which is my primary compiler).  Most of these are warnings about common errors that are not in fact real errors in Xastir code, but look like them (variables set but never used, variables possibly used uninitialized, but not ever actually used that way in Xastir, statements immediately after if statements that are indented as if they were intended to be guarded by the if, but which are in fact outside of the if, etc.).

I have started fixing these up, but wanted to record the issue here.

tvrusso, 2018-02-16:
To back up my claim of "a very large number of warnings,"  here are all the warnings emitted by GCC 6.3 on Raspian Stretch:

[pi_xastir_warnings.txt](https://github.com/Xastir/Xastir/files/1729939/pi_xastir_warnings.txt)

tvrusso, 2018-02-17:
This one is turning out to be really annoying.  It appears that the trend these days is to warn on just about everything that could possibly be a mistake, and not provide any means of shutting it up short of -Wno-\<warnaboutthis> or making crazy circumlocutions.

For example, certain functions like fgets and seteguid are set in their headers to be functions that should always warn if you don't use their results, e.g.:

      setegid(getgid());

Xastir has traditionally not warned or complained if this call fails, so it ignores the return value.  Some compilers (gcc 4.8.4 on Ubuntu 14.04) whine like mad about it --- but oddly gcc 6.3 on Raspbian Stretch doesn't.  One way to shut up the "unused result" warning is to assign the result to a variable and then don't do anything with it.  But then gcc 6.3 on Stretch complains that you've got a variable assigned but never used.  One can then make a nonsense use of the variable (e.g.:
    foobar=setegid(getgid());
    if (foobar != 0)
    {
      // do nothing
    } 

which does wonders for code readability.

fcc_data.c is *full* of uses of fgets that don't check their return values, and on Trusty with gcc 4.8 I get a ton of warnings.  Not on stretch with gcc 6.

Yay.

Anyhow, I've gotten all of the warnings in files from [a-i]*.c shut up on Stretch.  Not sure I should bother with the warnings from 4.8 on Trusty, especially all of those "unused result" warnings.  Then again, the steady stream of meaningless warnings does obscure the possibility of finding real errors, so it is probably worthwhile.

The problem is making sure I get all the code in ifdefs right.  For example, in interface_gui.c there is a bunch of X related code that saves return values of XtVaCreateManagedWidget and doesn't do anything with the returned widget (because the widget is set to be managed by some other).  In most cases, it's enough just to ignore the returned value.  But in the case of one such widget, the return value *IS* used, but only inside an ifdef.  In this case, silencing the warning means putting the variable declaration and the "variable =" inside the same ifdef that controls the later use of the variable.  Really, really ugly.  Whether it's uglier than having the warning is another question.

tvrusso, 2018-03-04:
Just a bump here.  I'm continuing to work on this issue, and am up to the "m"s.  Just finished picking through main.c and the huge number of warnings it produced on Debian Stretch is reduced to zero.   I am now down to picking through one or two files a week --- so it will be quite some time before I'm done with this.

KF7OML, 2018-10-18:
MR, tvrusso
I would like to get Xastir to install on my raspberry pi3 with the latest version of raspberrian installed on it.
Can you provide detailed instructions?  The install page they offer is way out of date.
Thanks

KF7OML, 2018-10-18:
> Just a bump here. I'm continuing to work on this issue, and am up to the "m"s. Just finished picking through main.c and the huge number of warnings it produced on Debian Stretch is reduced to zero. [...]

Hey can you provide Raspberry pi 3 install instrucitons?

tvrusso, 2019-04-07:
I have recently picked this issue back up, and now that gcc 8 is out I am finding that some of the files I had purged of all warnings now have new and exciting warnings coming up -- many of them irritating warnings about how the source string in an "xastir_snprintf" may be truncated down to the length of the destination buffer.  I call this an "irritating" warning because it was *PRECISELY* to make such a truncation happen if necessary that xastir_snprintf was used in all of these cases.  So warning about the potential for it is just confusing the issue.

I suspect I will be unable to get rid of those types of new warnings.

tvrusso, 2019-04-07:
There is a larger problem of repeated misuse of strncat's "count" argument.

strncat's third argument is *supposed* to be the maximum number of characters to copy to prevent overflow of the destination buffer.  In many, many cases in Xastir, it is instead used to specify the exact number of characters in the SOURCE buffer to copy, without regard to the size of the destination buffer.

In many of these cases, the code in question has NO IDEA how big the destination buffer is, and so cannot correctly guard against buffer overflow.  GCC 8 is warning about this misuse, because the count given is exactly the size of the source, and not a limitation based on the size of the destination.

tvrusso, 2019-04-07:
It looks like in many places strncat is used improperly as a substitute for memcpy --- it is frequently used to copy unterminated strings from one place to another by specifying the count of bytes to copy, and this is a misuse of strncat.

tvrusso, 2019-04-20:
I have basically gone back to square one on this issue, because by the time I came back to my old warning-removal branch, gcc8 came out and started throwing even more warnings.  And then I discovered that clang on OS X threw different warnings.  Instead of trying to tackle this all at once and all alone, I'm just going to make a bunch of PRs that pick at it piecemeal.

Other developers should feel free to make additional PRs to fix up warnings I've missed.

tvrusso, 2019-04-22:
Curt also addressed a bunch of these warnings in PR #68, but didn't mention this bug, so it didn't get tacked on here.  

I also addressed a handful of these issues in commit 1545996b, and also forgot to mention this issue.

tvrusso, 2019-04-23:
I removed one more warning (from Clang) in commit 51ef3f34 but neglected to mention this bug report in that commit message.

tvrusso, 2019-04-23:
I think we have come very close to the point of diminishing returns on this one.  Lots of progress this week!

we7u, 2019-04-26:
Running through all of the configure options one-by-one. Results:

Optional Packages:
  --without-festival        Disable festival features.              OK
  --without-gpsman          Disable gpsman features.                OK
  --without-libcurl         Disable libcurl features.               OK: Is it checked in DLM code?
  --without-libproj         Disable libproj features.               OK
  --without-shapelib        Disable shapelib features.              OK
  --without-pcre            Disable pcre features.                  Compile WARNING!
  --without-geotiff         Disable geotiff features.               OK
  --without-ax25            Disable ax25 features.                  OK
  --without-dbfawk          Disable dbfawk features.                Compile WARNING!
  --without-rtree           Disable spatial indexing of shapefiles. OK
  --without-map-cache       Disable Map Caching.                    OK: Is it checked in DLM code?
  --without-graphicsmagick  Disable graphicsmagick features.        Compile WARNINGS!
  --without-imagemagick     Disable imagemagick features.           Compile WARNINGS!

DEVELOPER OPTIONS:
  --with-errorpopups        Send error popups to stderr.            OK: Switch is backwards? 
  --with-libgc              Enable libgc features.                  OK: Needs "gc-devel" package
  --with-profiling          Enable profiling features.              OK
  --with-lsb                Enable Linux Standard Base.             Compile ERROR!
  --with-postgis            Enable Postgresql with PostGIS.         Compile WARNINGS!
  --with-mysql              Enable MySQL, with spatial support if available.    Compile WARNINGS!

tvrusso, 2019-04-26:
Just one comment:  There is almost no point to having both "--with-pcre" and "--with-dbfawk":  pcre is used *only* for dbfawk, so there should just be one option.  --with-dbfawk should require pcre, and refuse to configure if pcre isn't found.  That reduces one thing to test.

we7u, 2019-04-26:
Results from new Travis-CI config which runs through many configure flags (Linux only):

CONFIG 1: ./configure --without-dbfawk CPPFLAGS="-I/usr/include/geotiff"
  CC       map_shp.o
map_shp.c: In function `draw_shapefile_map':
map_shp.c:2324:38: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
                         if (lanes != (int)NULL) {
---------------------------------------------------
CONFIG 2: ./configure --without-graphicsmagick CPPFLAGS="-I/usr/include/geotiff"
Lots of imagemagick-related warnings, included "deprecated" and "declaration isn't a prototype"
---------------------------------------------------
CONFIG 3: ./configure --without-graphicsmagick --without-imagemagick CPPFLAGS="-I/usr/include/geotiff"
  CC       map_OSM.o
map_OSM.c:292:13: warning: `get_OSM_local_file' defined but not used [-Wunused-function]
 static void get_OSM_local_file(char * local_filename, char * fileimg){
             ^
map_OSM.c:395:13: warning: `xastirLat2pixelLat' defined but not used [-Wunused-function]
 static long xastirLat2pixelLat(
             ^
map_OSM.c:433:13: warning: `pixelLat2xastirLat' defined but not used [-Wunused-function]
 static long pixelLat2xastirLat(long osm_lat, int osm_zoom) {
             ^
map_OSM.c:442:13: warning: `xastirLon2pixelLon' defined but not used [-Wunused-function]
 static long xastirLon2pixelLon(
             ^
map_OSM.c:462:13: warning: `pixelLon2xastirLon' defined but not used [-Wunused-function]
 static long pixelLon2xastirLon(long osm_lon, int osm_zoom) {
             ^
  CC       map_tif.o
In file included from map_tif.c:87:0:
/usr/include/geotiff/xtiffio.h:68:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 extern void CPL_DLL XTIFFInitialize();
---------------------------------------------------
CONFIG 4: ./configure --without-shapelib --without-graphicsmagick --without-imagemagick CPPFLAGS="-I/usr/include/geotiff"
Same warnings as CONFIG 3.
---------------------------------------------------
CONFIG 5: ./configure --without-imagemagick --without-graphicsmagick --without-shapelib --without-pcre --without-dbfawk --without-rtree --without-map-cache --without-libcurl --without-ax25 --without-libproj --without-geotiff --without-festival --without-gpsman CPPFLAGS="-I/usr/include/geotiff"
Same warnings as CONFIG 3 minus the xtiffio.h warning.
---------------------------------------------------
CONFIG 6: ./configure --with-libgc CPPFLAGS="-I/usr/include/geotiff"
  CC       main.o
main.c: In function `main':
main.c:26879:5: warning: `GC_find_leak' is deprecated [-Wdeprecated-declarations]
     GC_find_leak = 1;
     ^
In file included from /usr/include/gc.h:2:0,
                 from leak_detection.h:86,
                 from main.c:161:
/usr/include/gc/gc.h:132:31: note: declared here
 GC_API GC_ATTR_DEPRECATED int GC_find_leak;

tvrusso, 2019-05-01:
I'm thinking that the number of warnings we are getting now is no longer "large" and we can call this done, picking off the remainder that are just warnings later, and fixing the ones that are actual bugs as part of the open bug reports that already exist for them.  Thoughts?

we7u, 2019-05-01:
Fine by me.

godfreja, 2019-05-01:
Closing this based on the progress we've made seems reasonable to me.

closed 2019-05-01 by godfreja