Xastir/Xastir #29: AX.25 header corrputed when data is relayed from APRS-IS to HF

DL7AD, 2018-02-19:
Hi,

I tried to send APRS messages through the APRS-IS. I figured out, that Xastir appends an additional char "}" in front of the whole packet at message packets and corrupts the AX.25 header. The bug occurs only if the packet was received from APRS-IS and sent over VHF by Xastir (with my TH-D72).

APRS-IS connected client DL4MDW-15 (Telnet in my case) => APRS-IS => Xastir (AF5LI) => TH-D72 => Receiver (GQRX in my case)

I send these messages directly into my Telnet Terminal:
```
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:53:48 GMT T2RADOM 193.106.216.154:14580
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:54:08 GMT T2RADOM 193.106.216.154:14580
DL4MDW-15>APX209,TCPIP*,qAC,T2UKRAINE::DL7AD-13 :ping{3B}
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:54:28 GMT T2RADOM 193.106.216.154:14580
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:54:48 GMT T2RADOM 193.106.216.154:14580
DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1A
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:55:08 GMT T2RADOM 193.106.216.154:14580
DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1A
DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1B
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:55:28 GMT T2RADOM 193.106.216.154:14580
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:55:48 GMT T2RADOM 193.106.216.154:14580
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:56:08 GMT T2RADOM 193.106.216.154:14580
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:56:28 GMT T2RADOM 193.106.216.154:14580
# aprsc 2.1.4-g408ed49 19 Feb 2018 18:56:48 GMT T2RADOM 193.106.216.154:14580
```

Xastir sends it
```
 0:TNC-> }DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1A
 0:TX -> AF5LI>APX209,:}DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1B
 0:TNC-> MYCALL AF5LI
 0:TNC-> MYCALL   was AF5LI
 0:TNC-> UNPROTO APX209 VIA WIDE2-1
 0:TNC-> UNPROTO  was APX209 VIA WIDE2-1
 0:TNC-> k
 0:TNC-> }DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1B
 0:TX -> AF5LI>APX209,WIDE2-2:=5230.55N/01329.86E-
```

And this is received with GQRX:
```
19:54:14$ fm AF5LI-0 to APX209-0 via WIDE2-1 UIv PID=F0
          }DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{3B}
19:54:59$ fm AF5LI-0 to APX209-0 via WIDE2-1 UIv PID=F0
          }DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1A
19:55:24$ fm AF5LI-0 to APX209-0 via WIDE2-1 UIv PID=F0
          }DL4MDW-15>APX209,TCPIP,AF5LI*::DL7AD-13 :ping{1B
```

Messages sent directly from Xastir affected:
```
19:55:50$ fm AF5LI-0 to APK003-0 via WIDE1-1,WIDE1-1 UIv PID=F0
          :DL7AD-13 :hallo5{63
```

73 Sven DL7AD / AF5LI

tvrusso, 2019-02-16:
Have you checked that this issue is still broken?  We found a problem a few months ago where a bug had been introduced into Xastir that was causing it to improperly pollute TNC/RF data with information from network interfaces.  That got fixed in November with commit 8d403fe, and I'm wondering if it also fixed this issue.

godfreja, 2019-04-21:
I am going to close this issue on the assumption that it was fixed by the commit Tom mentioned in the previous comment. If this is still an issue please reopen.

Thank you

closed 2019-04-21 by godfreja