Xastir/Xastir #32: Update Geofabrik dbfawk files to match current shapefiles

tvrusso, 2018-03-04:
In commit dd99425, Curt added a couple of DBFAWK files to support using shapefiles from http://download.geofabrik.de/ in Xastir.

Unfortunately, the shapefiles currently available on that web site do not match the DBFAWK files anymore.  The dbf signature is different, and they have clearly changed how the feature type is stored (it's now in an fclass field instead of a type field)  I have not checked what range of values the fclass field takes.  It may be that the only difference is in the name of the field, but if they reorganized the values of the field, too, then this may take a little more work.

Opening this bug to solicit community assistance in getting this fixed.  Going to mark it "Help Wanted" and "Good first issue".

mthassler, 2018-10-09:
tvrusso,
I have dbfawk files that are working with geofabrik files as of 10/8/18.  Still working on getting some decent colors for some of them.  Road colors I have setup on mine, are similar to the colors on open street map.  I have also set some map levels on them.  Some also show the red X, but no errors.  I haven't cleaned up the comments in the files yet either.  Attached what I have at this time.

[geofabrik-dbfawk-for-shapefiles.zip](https://github.com/Xastir/Xastir/files/2458560/geofabrik-dbfawk-for-shapefiles.zip)

tvrusso, 2018-10-09:
I do not have time to work on testing this issue, nor downloading and viewing geofabrik shapefiles.  The community will very much appreciate your efforts.

If you are willing to work on this issue, please create a pull request, and maybe support it by posting screenshots and other supporting evidence that you've addressed the incompatibilities of the current set of dbfawk files in Xastir with the current set of shapefiles available from geofabrik.

I have time to review pull requests, but unfortunately no spare time to spend working on Xastir myself these days.

Tyler-2, 2022-03-22:
[image]
I tried out the dbfawk's that were offered up and they are definitely an improvement over just the raw lines!

tvrusso, 2022-03-22:
Please create a pull request for us to review.  It's great that there's a zip file, but it's been there for a few years since the OP said he was going to clean them up.

tvrusso, 2025-12-10:
It may be possible (though tedious) to convert the osm files that geofabrik *does* still supply (and keep current) to shapefile with qgis.

tvrusso, 2026-01-07:
In fact, the zip file provided by the OP appears to work just fine with the shapefiles as they are available today in 2026, and I will probably use them and credit the OP.

Some tweaks may be necessary to optimize colors, as the OP had stated.

closed 2026-01-11 by tvrusso