Xastir/Xastir #33: Unable to configure imagemagick support on mint linux 18.3

P3Hosting, 2018-04-11:
I have run the configure command (output at bottom of post) using the apt-get install version of image magic
and after removing the old version and re-installing using the directions found at
https://www.imagemagick.org/script/install-source.php.
After doing the manual install the following is output by testing the new version
~$ identify -version
Version: ImageMagick 7.0.7-28 Q16 x86_64 2018-04-11 http://www.imagemagick.org
Copyright: © 1999-2018 ImageMagick Studio LLC
License: http://www.imagemagick.org/script/license.php
Features: Cipher DPC HDRI OpenMP 
Delegates (built-in): bzlib djvu fftw fontconfig freetype gvc jbig jng jpeg lcms lqr lzma openexr png tiff wmf x xml zlib
~ $ /usr/local/bin/Magick++-config --version
7.0.7 Q16 HDRI

The old version ( apt-get install)
output the following
~/xastir $ /usr/bin/Magick-config --version
6.8.9 Q16 

So from what I can tell ImageMagick is installed. What do I need to do to get ImageMagick support installed

./config command output for manual install below

george@YOGA-910 ~/xastir $ ./configure --with-ImageMagick=YES
configure: WARNING: unrecognized options: --with-ImageMagick
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes

Configuring xastir 2.1.1

checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking dependency style of gcc... (cached) gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
[... 262 lines cut ...]

xastir 2.1.1 has been configured to use the following
options and external libraries:

MINIMUM OPTIONS:
  ShapeLib (Vector maps) ................. : yes

RECOMMENDED OPTIONS:
  GraphicsMagick/ImageMagick (Raster maps) : no
  pcre (Shapefile customization) ......... : yes
  dbfawk (Shapefile customization) ....... : yes
  rtree indexing (Shapefile speedups) .... : yes
  map caching (Raster map speedups) ...... : no
  internet map retrieval ................. : yes (libcurl)

FOR THE ADVENTUROUS:
  AX25 (Linux Kernel I/O Drivers) ........ : yes
  libproj (USGS Topos & Aerial Photos) ... : yes
  GeoTiff (USGS Topos & Aerial Photos) ... : no
  Festival (Text-to-speech) .............. : yes
  GDAL/OGR (Obtuse map formats) .......... : no
  GPSMan/gpsmanshp (GPS downloads) ....... : yes

xastir will be installed in /usr/local/bin.
Type 'make' to build Xastir (Use 'gmake' instead on some systems).

we7u, 2018-04-11:
On Wed, 11 Apr 2018, P3Hosting wrote:

> I have run the configure command (output at bottom of post) using the apt-get install version of image magic
> and after removing the old version and re-installing using the directions found at
> https://www.imagemagick.org/script/install-source.php.

Check out the install docs for Linux Mint here:

   http://xastir.org/index.php/Installation_Notes#Linux_Mint

If that doesn't work, subscribe to the xastir list and ask there. (www.xastir.org)

-- 
Curt, WE7U.        http://we7u.wetnet.net
"The world DOES revolve around me:  I picked the coordinate system!"
Coordinate Converter (Android): http://www.sarguydigital.com

P3Hosting, 2018-04-12:
Thanks.
I followed the doc you suggested for mint linux. With the following exception. I used the git repository not the source forge repository. I believe that the source forge repository is no longer being used (xastir.org main page big green highlighted area at top says as of 7/ 2016) so the doc for mint linux is out of date at best.  As far as using the mailing list i searched it and all info on imagemagick is dated before 2008 so at best it is also out of date.

I have been looking at the configure script and believe that there is an issue with finding the correct location of the imagemagick modules or this may be a DUA. Is there any accurate current documentation on how to install xastir and what if any parameters I would need to supply to get imagemagick support installed. Because what is shown and pointed to on the website is not current and I have done what they suggest. using apt-get install installs older releases and still has the same problem.

Thanks
George

tvrusso, 2018-04-12:
The platform-specific installation notes on the Xastir wiki are user contributions, and clearly the person who wrote th mint instructions did not keep them up to date.  

The Xastir configure script tries to get Magick's configuration info by trying to find Magick-config in your path.  It does not use an explicit path name itself.  Is Magick-config in your path?

Most Xastir users have moved away from using ImageMagick and use GraphicsMagick instead, because ImageMagick tends to have an unstable API and used to break a lot.  But Xastir's configure script searches for GM the same way --- it expects the GraphicsMagick-config program to be in your path.  From the configure output you post, it does not appear that either is in your path.

Curt's comment stands:  you should ask your questions on the Xastir mailing list, not in GitHub issue requests --- only a handful of Xastir developers monitor here, and your question is more likely to be answered if you ask it of the wider Xastir community, some of whom might even be running Xastir on Linux Mint already.

tvrusso, 2018-04-12:
Ugh.  And another thing:  You have configured ImageMagick with HDRI support.  This will break Xastir, because it changes a pixel data type to a double instead of an integer, and Xastir can't deal with that.  You won't be able to compile Xastir.  

Before you use ImageMagick with Xastir you'll need to rebuild it with HDRI support disabled.  This is mentioned in the FAQ (https://github.com/Xastir/Xastir/blob/master/FAQ), under "Why does compilation of map_geo, map_WMS or map_tiger fail with "invalid operands to binary <<" errors?"