Xastir/Xastir #35: get-pop

kc2cbd, 2018-05-31:
The get-pop script is returning a 404.

Eric-Sparks, 2018-06-25:
I looked at this earlier today and the GNIS_SITE listed at uah.edu is no longer there (main domain is being redirected to something that isn't a file server).

godfreja, 2019-04-21:
I'm not familiar with the GNIS functionality that get-pop references. How is this different from the get-gnis script?

kg4wsv, 2019-08-04:
That server (www.eng.uah.edu) was terminated 5 years ago, and the data was at least 5 years old at that time.

I may be able to resurrect the data, find a home for it, and get the script working again, but is it worth it for 10+ year old data?

At a glance it appears get-gnis suffered a similar fate. I tried the URL in the script and poked around a bit. It seems the data has moved from usgs.gov to nationalmap.gov. I didn't do much of a search at nationalmap.gov to find it.

What data are you looking for? There may be a modern alternative.

kc2cbd, 2019-08-04:
I was just trying to run the script.
I'm not really sure what the script is suppose to retrieve.

godfreja, 2024-08-02:
I am thinking about deleting the get-pos script and also going through and find other scripts that are no longer worth keeping and deleting them.

Any objections?

tvrusso, 2024-08-02:
I am all for that.

Xastir has had a lot of one-off features added in its life that either stop working or rot for some reason or another.   Their presence can confuse users (as, for example, did "get-maptools.sh" and "get_shapelib.sh" that had stuck around for years after they stopped being maintained, and caused more than one users to make use of them only to break their setup).

Doing some house cleaning is not a bad idea.  

>  As an aside, there are other ancient, broken features that probably should get removed, such as the use of the geocoder that once enabled us to process TIGER/line data to create an address database with georeferencing, so you could look up street addresses and have Xastir center on them.   Those TIGER/Line files are long obsolete, and the README.md file even points users at an FTP site for downloading preprocessed geocoder files that is long gone.  Another issue should probably be opened to call for clearing *that* out.

PR #224 (2024-08-02): Delete get-pop and update get-NWSdata
Remove the get-pop script as it points to data that is no longer there.

Update the get-NWSdata script to point at the current versions of the files. New versions go into effect on September 10th and the script should be updated again then.

Closes #35.
commit c762b665 (2024-08-02): Delete get-pop and update get-NWSdata

Remove the get-pop script as it points to data that is no longer
there.

Update the get-NWSdata script to point at the current versions of the
files. New versions go into effect on September 10th and the script
should be updated again then.

Closes #35.
closed 2024-08-02 by godfreja