Xastir/Xastir #37: Messages continually transmitted on RF even after they've been ack'd on Inet

Eric-Sparks, 2018-06-02:
I sent a message to a friend this morning and watched the message screen show that the message had been acknowledged.  I then watched the "Display Packet Data" screen for the next few minutes and noted that Xastir continued to transmit the message over RF even though it had already been acknowledged.

Version 2.0.8 as compiled and packaged for Fedora.

tvrusso, 2018-06-02:
I'm having trouble reproducing this.  I have sent a number of messages to stations that are out of my local area, and received ACKs over APRS-IS.  Each ack immediately ends my retransmissions over all interfaces. 

Can you provide a little more information about what you're doing and what you're seeing happen?  Screen shots, exact steps needed to reproduce, etc. would be helpful.

Eric-Sparks, 2018-06-02:
I'm unable to duplicate with the latest build.  I'll continue to watch for it and see if I can reproduce it.

closed 2018-06-02 by Eric-Sparks