Xastir/Xastir #38: Local NOGATE messages being sent to IS

Eric-Sparks, 2018-06-02:
Just sent a message with the path digi,digi,NOGATE.  To me, this would have told Xastir to not use the IS path and only use the RF path.  Because the packet wasn't gated the IS servers never saw the path with the NOGATE so it never knew that the packet shouldn't have been there.

WG3K>APX211,TCPIP*,qAC,WG3K-NJ::WJ3K-10  :http://thelifeofkenneth.com/aprx{06}

Seeing this in version 2.1.1 (539bbf84).

tvrusso, 2018-06-03:
While it might seem to you that NOGATE and RFONLY would mean that your station would not send this data to APRS-IS, it is a misunderstanding on your part about what these mean.

NOGATE and RFONLY apply only to IGate functions.  A packet received by an IGate from RF with NOGATE or RFONLY in the digi path will not be gated to APRS-IS.  XASTIR does have this sort of check in its igate functions, and will respect this bit of pathing.  APRS-IS servers don't check the NOGATE or RFONLY path elements --- if a packet gets to them, they take it.  It's the job of igates not to send it on.

But that's not what's going on here.  You have your station configured to transmit both on RF and APRS-IS, and your local packets are always sent out on all of your configured interfaces.  NOGATE and RFONLY do not apply in this case --- you're not igating, you're just transmitting, and when you transmit local packets, all configured interfaces get the information.

If you don't want your packets sent to APRS-IS, disable the "Allow Transmitting?" check box in the APRS-IS interface properties dialog before sending the messages you intend not to go to the internet.  Since you've got NOGATE in the path, none of the IGates that hear your station will gate it, and you won't push it to the internet yourself.

It is not out of the question that we could add checking for NOGATE and RFONLY before transmitting local packets to APRS-IS over configured interfaces, but it is not really likely to get much attention given how few people are actually working on XASTIR these days.  I'll leave this issue open in case someone would like to explore the issue and maybe implement such a check.  But in the meantime, you'll simply have to avoid transmitting your unwanted packets to the internet yourself using the properties dialog.

Eric-Sparks, 2018-06-07:
I'm familiar with how NOGATE and RFONLY works.  I'm talking about my expectation of how my message would be treated from the client side.  At a higher view, I would like to see better mechanisms for which circuits a message gets transmitted without having to go into the circuits menu to turn off, and then back on, different circuits every time I want to send a message a specific way.