Xastir/Xastir #43: Instruction page for Raspberry pi install are not functional.  Many errors!!

KF7OML, 2018-10-18:
Instruction page for Raspberry pi  3 install are not functional.  Many errors!!
Please update instructions page.

we7u, 2018-10-18:
On Wed, 17 Oct 2018, KF7OML wrote:

> Instruction page for Raspberry pi  3 install are not functional.  Many errors!!
> Please update instructions page.

Please subscribe to the Xastir mailing list and begin a discussion on it. There are quite a few users there that may be able to help.

I don't recall who wrote the Rpi instructions but I know they pre-date the Rpi3 and most likely pre-date the latest version of Raspian as well.

Mailing list is at xastir.org

-- 
Curt, WE7U.        http://we7u.wetnet.net
APRS Wiki:  http://info.aprs.net/
Coordinate Converter (Android): http://www.sarguydigital.com

KF7OML, 2018-10-18:
Ok,
I subscribed.

tvrusso, 2019-03-17:
I have installed Xastir on Raspbian using the instructions at http://xastir.org/index.php/HowTo:Raspbian, and have not found it to contain "many errors."  It works fine.  The place to discuss Wiki changes is the Xastir mailing list, not GitHub, so I'm closing this issue.

closed 2019-03-17 by tvrusso

KF7OML, 2019-03-17:
I loaded a new copy of the OS and the install went fine. Thanks