Xastir/Xastir #44: .geo maps not showing

LiborBerka, 2018-11-10:
Hello my hamradio callsign is PY2ZEN. I upgraded xastir to version 2.1.1. compiled from the source.
The ./configure process passed well and all librares were accepted:

xastir 2.1.1 has been configured to use the following
options and external libraries:

MINIMUM OPTIONS:
  ShapeLib (Vector maps) ................. : yes

RECOMMENDED OPTIONS:
  GraphicsMagick/ImageMagick (Raster maps) : yes (GraphicsMagick)
  pcre (Shapefile customization) ......... : yes
  dbfawk (Shapefile customization) ....... : yes
  rtree indexing (Shapefile speedups) .... : yes
  map caching (Raster map speedups) ...... : yes
  internet map retrieval ................. : yes (wget)

FOR THE ADVENTUROUS:
  AX25 (Linux Kernel I/O Drivers) ........ : yes
  libproj (USGS Topos & Aerial Photos) ... : yes
  GeoTiff (USGS Topos & Aerial Photos) ... : yes
  Festival (Text-to-speech) .............. : yes
  GDAL/OGR (Obtuse map formats) .......... : yes
  GPSMan/gpsmanshp (GPS downloads) ....... : yes

xastir will be installed in /usr/local/bin.

make and make install passed normal.

Maps .geo are correctly placed in /usr/local/share/xastir/maps/Online directory and are seen in choose
table
[image]
Please can somebody help me to resolve this issue?
Thank you
Libor

tvrusso, 2018-11-10:
Which map have you tried to load?
Can you see any error messages on the standard output? (you would have to run Xastir from the command line so that these show up)
It is sadly the case that many online map sources get taken off line without notice, and Xastir's geo files simply stop working for that source.  The OSM_tiled_mapnik.geo file generally works for me, so you might try that one and see if it displays.
If not, the information you've provided isn't enough to diagnose what the problem is.

LiborBerka, 2018-11-10:
How can I run xastir ftrom terminal without display?

LiborBerka, 2018-11-10:
Maybe the correspond .inf file must be present together with .geo files in maps directory. Will try it.

LiborBerka, 2018-11-11:
So Itried to run xastir from terminal as user i.e $ xastir & as you said. And the result is this:
pi@raspberrypi:~$ xastir &
[1] 28664
pi@raspberrypi:~$ 

Built-in map types:
      gnis   USGS GNIS Datapoints
       pop   USGS GNIS Datapoints w/population
       map   APRSdos Maps
       map   WinAPRS/MacAPRS/X-APRS Maps
       pdb   PocketAPRS Maps

Support for these additional map types has been compiled in: 
       geo   Image Map (ImageMagick/GraphicsMagick library, many formats allowed)
       geo   URL (Internet maps via wget)
       geo   URL (OpenStreetMaps via wget
                  Copyright OpenStreetMap and contributors, CC-BY-SA)
       shp   ESRI Shapefile Maps (Shapelib library)
       tif   USGS DRG Geotiff Topographic Maps (libgeotiff/libproj)
       tab   MapInfo TAB
       mid   MapInfo MID
       mif   MapInfo MIF
       ddf   Spatial Data Transfer Standard (SDTS)
       s57   International Hydrographic Organization (IHO) S-57
       dgn   MicroStation DGN
       rt1   US Census Bureau TIGER/Line

festival_client: connect to server failed
SayText: Couldn't open socket to Festival
Error: Can't open display:

Is the problem with Festival library. I ll try to install the Festival manually from source then.
About .geo origonal files I have the same ones installed as were in 2.0.4 version. But I have
also my made .geo files with the script inf2geo. They worked well in 2.0.4

LiborBerka, 2018-11-11:
Hm Fstival has nothing with showing the maps. It serves for voice server...I'll not touch it.

LiborBerka, 2018-11-12:
I have resolved today the issue with re-installing xastir to version 2.0.8. There all .geo maps commenced to be shown normally. That previous problematic version 2.1.1 I deleted from raspberry linux.

closed 2018-11-12 by tvrusso

we7u, 2018-11-12:
I just noticed: You're posting to Github with questions.

Try going to xastir.org and subscribe to the Xastir mailing list: You'll have a lot more people there to ask questions of.

-- 
Curt, WE7U.        http://we7u.wetnet.net
APRS Usage Pages:  http://we7u.wetnet.net/usage.html
Coordinate Converter (Android): http://www.sarguydigital.com