Xastir/Xastir #45: Symbols and maps do not install correctly on raspberry pi compass

nouse4anick, 2019-01-01:
here's the script I use for building (note all the libraries are already installed)
```
git clone https://github.com/xastir/xastir.git
cd xastir
./bootstrap.sh
./configure CPPFLAGS="-I/usr/include/geotiff"
make
sudo make install
```
this installs correctly except it doesn't copy maps or the symbols directory

tvrusso, 2019-01-01:
Also, can you please give a little more detail about what OS you're running on your Pi?  I cannot find any information about "raspberry pi compass" except articles about making an actual digital compass out of a Raspberry Pi.

tvrusso, 2019-01-01:
Sure enough, when I build Xastir on my Raspberry Pi 2 running Raspbian Stretch, it built and installed everything, including the maps and symbols directories (the old copies of which I had moved out of the way just to make sure they were properly reinstalled).

In order to help you out, we'll need a lot more information.  I have a spare Pi 3 that is gathering dust and I'm willing to play around with it to see if I can reproduce your issue, but at the moment I cannot.

nouse4anick, 2019-01-01:
when i do make I see this towards the end of the build:
```
Making all in symbols
make[2]: Entering directory '/home/pi/xastir/symbols'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/pi/xastir/symbols'
```
this is the symbols install:
```
Making install in symbols
make[1]: Entering directory '/home/pi/xastir/symbols'
make[2]: Entering directory '/home/pi/xastir/symbols'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/xastir/symbols'
 /usr/bin/install -c -m 644 symbols.dat 2x2.xbm 25pct.xbm 13pct.xbm alert.xbm flood.xbm red_flag$
make[2]: Leaving directory '/home/pi/xastir/symbols'
make[1]: Leaving directory '/home/pi/xastir/symbols'
```
so after looking at this I figured out the main problem.... when i go to run xastir its looking for everything in /usr/share/xastir.... its NOT looking in /usr/local/share/xastir!

tvrusso, 2019-01-01:
This is because you have previously installed and used a binary package from apt-get, so your Xastir config is set up to point to the wrong place.  This is actually pretty well documented all over the wiki, notably at http://xastir.org/index.php/HowTo:Raspbian under the section "If you already installed the binary version".

You have several options.  

1.  edit ~/.xastir/config/xastir.cnf and replace all occurances of "/usr/share/" with "/usr/local/share/"
2.  simply rename ~/.xastir/config/xastir.cnf and restart Xastir.  Xastir will create an all-default xastir.cnf file, and you'll have to completely reconfigure your station.

closed 2019-01-01 by tvrusso

nouse4anick, 2019-01-02:
ok, I see what happened, thanks for your understanding and help! I'll have to forward that to everyone else in the groups i'm in!

johnhays, 2019-04-16:
The current build instructions don't show running make install in symbols directory

godfreja, 2019-04-16:
Perhaps I am misunderstanding what you are getting at, but there should be no need to separately run "make install" in the symbols directory. It will be done from the top level make.

johnhays, 2019-04-16:
It doesn't work on Raspbian unless you run it separately.

tvrusso, 2019-04-16:
I am testing this out on Raspbian right now, but it seems very odd that Raspbian would behave any differently than any other system, on which this behaves properly.  Doing "make install" anywhere other than top-level should not be necessary.

tvrusso, 2019-04-16:
My build and install on Raspbian from today's code base, and I am unable to reproduce any failure to install.

I can confirm that the build system does indeed install maps and symbols when "make install" is invoked from the top of the build directory, even on Raspbian:
```
> sudo mv /usr/local/share/xastir /usr/local/share/xastir_preserved
> ls /usr/local/share
ca-certificates  doc  emacs  fonts  man  sgml  xastir_preserved  xml
```

I then configured and built Xastir per the wiki instructions:
```
> mkdir build
> cd build
> ~/src/Xastir/configure CPPFLAGS="-I/usr/include/geotiff"
> make
```
 and finally installed by invoking the appropriate command:
```
bogopi[992]  (17:17:22)> sudo make install
```

after which /usr/local/share/xastir was properly populated

```
> ls /usr/local/share
ca-certificates  doc  emacs  fonts  man  sgml  xastir  xastir_preserved  xml
> ls /usr/local/share/xastir
config  Counties  fcc  GNIS  help  maps  scripts  sounds  symbols
> ls /usr/local/share/xastir/symbols
13pct.xbm  alert.xbm  red_flag.xbm  torndo.xbm        winter_weather.xbm
25pct.xbm  flood.xbm  snow.xbm      wind.xbm
2x2.xbm    icon.png   symbols.dat   winter_storm.xbm
```

If you continue to have issues, please open a new bug report and show your detailed steps, so we can try to understand and reproduce the problem you're seeing.