Xastir/Xastir #46: Debian Repo

xJMV, 2019-01-08:
Will the debian repository be updated again with a more up-to-date version of Xastir in the future ?

I actually use the repo version (2.0.8) on all my setup because it is just easy to install and maintain every systems (all can be done with a one-liner instead of many operations for each systems).

Hibby, 2019-01-08:
Hey there -

If this is the Debian distro repo as opposed to an xastir-run one, yeah -
I'm releasing an update in the next day or so so it's in before freeze. I
did the work a while back and asked some folks to test it, then we all
forgot!

I'll bounce a ping to this list when it's up:)

Cheers,
Dave

we7u, 2019-01-08:
You're discussing this on Issue #46 on the Xastir Github repo. Xastir developers don't do binary releases so it's up to those who have favorite Linux distributions to see to that themselves. Sounds like someone is thinking about it for Debian and Debian-based distros. More power to you!

we7u, 2019-01-08:
I replied on Github directly but didn't see it come through email, so will reply here as well.

You're discussing this on Issue #46 on the Xastir Github repo. Xastir developers don't do binary releases so it's up to those who have favorite Linux distributions to see to that themselves. Sounds like someone is thinking about it for Debian and Debian-based distros. More power to you!

xJMV, 2019-01-08:
@Hibby Thanks mate! Glad to read it! I'll get myself ready to update my devices! :)

Hibby, 2019-01-10:
Hi all,
So I did some research - I committed 2.1.0 to our Repos back in Feb 2018, it's in the 'Testing' distribution which is heading for a freeze in the coming weeks, so if you're running stable you'll get it at the next major release. It's already in Ubuntu and other Derivatives.

It's a little late in the cycle to do a formal backport of 2.1.0 to the current Debian stable, sadly, but it'll come down the pipeline in due course!

I've just committed a new update that includes some architectural changes to make the package a little better for Debian users, and adds a reproducible build patch as we're quite hot on that at the moment.  

@we7u We can probably close this one now, I'll pass on ther reproducible build patch shortly, if I haven't already!

Cheers,
DH

closed 2019-01-10 by we7u