Xastir/Xastir #47: Support for PROJ 6.0.0

sebastic, 2019-03-13:
Xastir still uses `proj_api.h` which is deprecated in PROJ 6.0.0 and will be removed in PROJ 7.0.0 (scheduled for March 2020).

When using `proj_api.h` with PROJ 6.0.0 `-DACCEPT_USE_OF_DEPRECATED_PROJ_API_H=1` needs to be added to `CFLAGS` to prevent a compiler error.

Xastir should be updated to (also) support the `proj.h` API.

tvrusso, 2019-03-13:
Thank you for your report.  All of my systems still have PROJ 5.

Would you be willing to contribute patches to get this additional support in place?  We would need to maintain backward compatibility with older versions, so a configure probe for the proj version would be required, along with any ifdefs necessary to make older API calls continue to function if they're available.

sebastic, 2019-03-13:
I'm not able to contribute patches for PROJ 6 support unfortunately.

As maintainer of the proj package in Debian I went over all reverse dependencies that will require changes for PROJ 6, and updated the [list maintained by the PROJ project](https://github.com/OSGeo/proj.4/wiki/proj.h-adoption-status).

tvrusso, 2019-03-14:
Xastir's use of proj.4 and its proj_api.h is limited to the map_tif.c code, and is isolated in a single function.

Apparently, some 14 years ago, it was observed that if Xastir was compiled with both GeoTIFF and GDAL support, the GTIFProj4FromLatLong function would break.  The solution at that time was to use a private GTIFProj4FromLatLong copied (with permission) from the libgeotiff library, rename it, and use *that* name everywhere.  The code is conditionally compiled only if GDAL is in use, otherwise, the new name is just "#define"d to the original name, falling back on the one in the library.

The best path forward for resolving this bug would be to determine if this hack is still needed.  If not, removing it would solve the problem completely.

If it is still necessary to use this hack, then the next best solution might be to copy the GTIFProj4FromLatLong function from the upcoming libgeotiff 1.5.0, which has been updated to use the proj5 API.  

Another problem is our inclusion of an ancient version of shapelib, to which we fall back if the target system doesn't have shapelib installed.  This contains a program "shpproj" that has been removed from more recent shapelib releases.  I see two paths forward to this:  1) remove our fallback shapelib, because it was a relic of the days when shapelib was kind of hard to come by in package repos, and that's no longer true  or 2) remove shpproj from our shapelib/contrib directory.

shpproj is of very, very limited utility and we shouldn't need to be distributing it, especially since shapelib has dropped it themselves.

tvrusso, 2019-03-15:
Current thinking is that the best solution would be to remove the GDAL/OGR support in Xastir, which doesn't really work well and isn't used much.  This would enable us to remove the problematic copied GTIFProj4FromLatLong function altogether, and thereby completely remove all direct calls to the proj library.  We would still depend on proj, but indirectly through libgeotiff --- and so long as libgeotiff stays up-to-date with proj development, we'll be golden.

tvrusso, 2019-03-15:
I've renamed this issue --- rather than add support for Proj 6.0.0 API, we really just need to stop using proj directly in the one place where it's used.

closed 2019-03-17 by tvrusso

sebastic, 2019-07-12:
xastir 2.1.2 still links to libproj via `configure`, is that intentional?

tvrusso, 2019-07-12:
Yes, it is still intentional.  

Xastir uses libgeotiff and libgeotiff requires proj libraries.  By explicitly adding it to the configure probes, we make sure that we can use libgeotiff, even if it is a static library.

PR #48 (2019-03-17): Feature removegdal
This set of changes would resolve #47.

It removes the support in Xastir for OGR and GDAL, none of which ever really worked well and none of which is in much use by any Xastir users.  By doing this, we can remove the hackish inclusion of a copy of libgeotiff's GTIFProj4FromLatLong in map_tif.c, which was the only place in all of Xastir that any proj API was used directly --- what we *really* want to be doing, and were doing before GDAL support was added, is to use the libgeotiff projection conversion code, which itself uses proj.

The libgeotiff project is on top of the whol [... cut]