Xastir/Xastir #50: Greek language support has introduced annoying warnings due to locale setting

tvrusso, 2019-03-21:
Ever since commit bab28a79 (integrating Greek language support into Xastir), Xastir has been spewing annoying warnings:

>  Warning:
>      Name: create_appshell text_input2
>      Class: XmTextField
>      Charater '\260' not supported in font. Discarded

On many US-based Linux systems, the default LANG setting is "en_US.UTF-8" (or something like that).  Prior to integration of the Greek language in Xastir, Xastir had hard-coded some settings of certain locale components (LC_NUMERIC and LC_CTYPE) to "C" using the "setlocale" function.  To support Greek, these were removed and replaced by some set/restore locale functions that only *sometimes* force us into C locale.

This led to the warnings in question, which are all about drawing the degree symbol using character code 260 (octal), which is the extended ASCII encoding for this symbol.  Using any other encoding can break Xastir in some contexts.

I addressed *some* of the problem in commit 910d6c4a, which forced us into C local every time we try to draw the degree symbol, but it doesn't always work.  For example, if your LANG setting is en_US.UTF-8, the degree symbol doesn't show up in a station info dialog next to the station bearing, but if LANG="C" or LANG="", it does.

My position is that the Greek stuff was accepted and merged into Xastir a little too casually, and has caused a great deal of headache for users, and a lot of traffic on the Xastir mailing list as a result, while primarily benefiting only a small user base.  It may be better to maintain it as a branch until it can be fully debugged rather than merging it into the main code base.  I think it should be reverted (by reverting all the commits that implemented it on master), then the commits that implemented it should be cherry-picked onto a new branch, which should be kept rebased off of master, but not merged in without extensive testing.

Since Curt merged these changes into the master branch, I'm assigning the reversion/fixes to him.

we7u, 2019-03-21:
Received. Thanks. Will see if I can look into this in the next week. I'm
back from vacation.

we7u, 2019-03-26:
Have reverted all of the Greek patches and created a new local branch with those same patches in it. I've also forked the Xastir project into my personal Github account and will most likely put the Greek patches into either a feature branch or the master branch there for those that want Xastir with the Greek language added. Once I get everything working properly w/o generating errors to STDERR, I'll issue a pull request to the master Xastir project and get it merged back into the mainline.

closed 2019-03-26 by we7u