Xastir/Xastir #53: Unable to get GPS data via GPSD

opg4759, 2019-04-04:
I've complied the most recent xastir build (2.1.1). I've enable Networked GPS (via GPSD) and the interface show UP. However I'm not getting the GPS data to read and update my position. I've tested using YAAC and I am getting updated GPS data there via GPSD. I have verified that GPSD is running

gpsd.service - GPS (Global Positioning System) Daemon
   Loaded: loaded (/lib/systemd/system/gpsd.service; enabled; vendor preset: ena
   Active: active (running) since Thu 2019-04-04 13:49:10 PDT; 30min ago
  Process: 500 ExecStart=/usr/local/sbin/gpsd $GPSD_OPTIONS $OPTIONS $DEVICES (c
 Main PID: 503 (gpsd)
   CGroup: /system.slice/gpsd.service
           └─503 /usr/local/sbin/gpsd -n /dev/ttySC0 /dev/pps0

I am running on a RBI 3B+ and using a DRAWS hat for my GPS source.

tvrusso, 2019-04-08:
Could you please provide more information about your Xastir system and GPSD?  

I am having trouble reproducing this problem.  I've currently got gpsd running on one system and Xastir connecting to it from another system, and there is no difficulty getting the data, although in one case I found it took a while for Xastir to report that it had valid data, even though xgps on the same machine showed it was fine.

How long have you waited to see if Xastir starts noticing the valid gps data?

opg4759, 2019-04-09:
I have left my Pi up and running days and it never passes the data through to xastir, even though the interface says its up. I know gpsd is working in other applications.

Below is the console output when i launch xastir.

Built-in map types:
      gnis   USGS GNIS Datapoints
       pop   USGS GNIS Datapoints w/population
       map   APRSdos Maps
       map   WinAPRS/MacAPRS/X-APRS Maps
       pdb   PocketAPRS Maps

Support for these additional map types has been compiled in: 
       geo   Image Map (ImageMagick/GraphicsMagick library, many formats allowed)
       geo   URL (Internet maps via libcurl library)
       geo   URL (OpenStreetMaps via libcurl library
                  Copyright OpenStreetMap and contributors, CC-BY-SA)
       shp   ESRI Shapefile Maps (Shapelib library)
       tif   USGS DRG Geotiff Topographic Maps (libgeotiff/libproj)
       xpm   X Pixmap Maps (XPM library)
festival_client: connect to server failed
SayText: Couldn't open socket to Festival
Indexing maps...
Finished indexing maps
*** Reading WX Alert log files
*** Done with WX Alert log files
convert_from_xastir_coordinates:X out-of-range (too low):4272240745
convert_from_xastir_coordinates:X out-of-range (too low):4269153266
convert_from_xastir_coordinates:Y out-of-range (too low):4294882319
convert_from_xastir_coordinates:X out-of-range (too low):4285992775
Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
Tue Apr 09 11:08:22 PDT 2019:
	Interface Error!  Error opening interface 0 Hard Fail

Socket connection for interface 0 type (2, 1, 6) failed: Connection refused
Tue Apr 09 11:09:00 PDT 2019:
	Interface Error!  Error opening interface 0 Hard Fail

tvrusso, 2019-04-10:
There are tons of errors on your console, so perhaps one of those is related --- and if not, you certainly have some other problems, too.  What interface is the failed interface 0?  Could that be your "Networked GPS" interface?

I'm also curious why Magick is segfaulting on you, which indicates a problem with map display.  This has nothing to do, of course with the GPSD issue.

By the way, while this is unrelated to your original question, are you running the version of Xastir from the Pi repository (e.g. "apt-get install xastir") or  did you build it yourself from source?  I ask because you have clearly enabled festival text-to-speech, but are not apparently running the festival daemon that it requires to work.  Unless you want to run that daemon, you should probably turn off that feature (turn it off by de-selecting the various radio buttons under File->Configure->Speech).

tvrusso, 2019-04-10:
Other information that would be helpful:
   - Is gpsd running on the same machine as Xastir?
   - What port is your Xastir interface set to read from?  Is this the same port as gpsd is listening on?
   - If you change the debug level to 2, does it show any information related to gpsd as the port is turned on?

opg4759, 2019-04-10:
i built myself from source
yes it it is running on the same machine
port is default 2947

here is the output from debug level 2
*** Reading WX Alert log files
fill_in_new_alert_entries start
fill_in_new_alert_entries end
*** Done with WX Alert log files
Port Available 0
Opening a network GPS using gpsd
Creating new thread
net_connect_thread start
We have a socket to use
after setsockopt
calling connect(), port: 2947
10
connect result was: 0
net_connect_thread():Net up, port 0
net_connect_thread terminating itself
10
Net ok: 1, port 0
*** net_init is returning a 1 ***
CMD:R

*** add_device: ok: 1 ***
Start port 0 threads
*** Startup of read/write threads for port 0 ***
Port 0 read start
End port 0 threads
Port 0 write start
CMD:?WATCH={"enable":true,"nmea":true}

tvrusso, 2019-04-10:
Sorry, I'm having trouble reproducing this issue, as I've just built Xastir from source on my own Pi, started gpsd with an attached gps, and connected Xastir to it.  Both xgps and Xastir show data coming in from the GPS, and Xastir claims to be getting a valid fix.  My debug level 2 output looks just like yours.  I did nothing special, and it Just Works.

I don't even know what else to ask to try to find the real source of the issue.  You might be better off asking this question on the Xastir mailing list, where someone with more Pi experience might point you at some obvious problem that I'm overlooking.

we7u, 2019-04-10:
Could be group ownership on uucp or dialout group? May need to add your
login to that group in /etc/group and reboot.

Could also be a firewall issue. If you're running a firewall make sure
you've opened up that port. For a quick test you can drop the firewall
entirely and see if Xastir connects to gpsd. Is that hard failure on the
gpsd interface in Xastir?

tvrusso, 2019-04-10:
Since gpsd is running fine and getting data from his gps, and other programs can read data from gpsd just fine, it should not be a group permissions issue.  And when I installed gpsd on my own pi, no such changes were necessary to get it to work.  It just worked right out of the box.  There must be something very fishy going on, but it should not be necessary to be in the dialout group just to connect to gpsd's network port (only for gpsd to have permission to read the serial port (or ttyUSB device) with the GPS on it, which it clearly already has since it is working with other programs).  But the default user on the pi already *is* in the dialout group.  Perhaps you're running as a non-default user?  If so, Curt's suggestion is worth a try, but I don't think it'll help.

His gpsd is running on the very same machine as his xastir, so I don't see how it could be a firewall issue.

The hard failure on port 0 at start-up does concern me --- but the debug level 2 output he posted clearly shows a successful connection to gpsd on port 0 later on.

tvrusso, 2019-04-11:
Basil Gunn has reported in email to me and Curt that the model number of the GPS module used on the DRAWS hat is the SkyTraq S1216F8-GL.  The spec sheet for this module shows that it does not in fact produce the GPGGA and GPRMC sentences that Xastir expects, but instead emits the completely equivalent GNGGA and GNRMC sentences.

It should not be a difficult task to add support for GNGGA and GNRMC sentences to Xastir.
I'm on it.

tvrusso, 2019-04-11:
OK, I believe I have made the changes necessary to support GNGGA and GNRMC sentences as being essentially equivalent to GPGGA and GPRMC.  I have NOT merged them into the Xastir code base yet, because they might not work as I do not have a GLONASS receiver to test on.  But I have pushed the code to my fork and made a draft pull request that folks can use to test this out.   It is pull request #58.

To grab my changes, you need to pull the "feature-supportglonass" branch from my fork of Xastir.  First, clone the normal Xastir repository.  Then, cd to your clone and issue the following commands:

git checkout -b tvrusso-feature-supportglonass master
git pull https://github.com/tvrusso/Xastir.git feature-supportglonass

This will copy the contents of my "feature-supportglonass" branch into a new branch of your clone.  You can then build the code and test it.  If I did this right, your GPS data from the DRAWS hat should start being recognized by Xastir.

n7nix, 2019-04-11:
After build I am running Xastir V2.1.1 (d3ae0719)
Before build I was running Xastir V2.0.8
Running View ->GPS Status : Sats/View:12 Fix:DGPS

I believe this version of Xastir is now working for the the DRAWS hat GLONASS GPS.

Two questions:
- Is there a way to display  Xastir version number with a command line option?
- I am challenged going from a Raspbian package install of Xastir to a git repo build and getting maps to display at all. I'm familiar with /usr vs /usr/local Debian filesystem standards.
Maps display fine from a package install and then not at all after a manual build install.

For reference my console commands follow:
 git checkout -b tvrusso-feature-supportglonass master
 git pull https://github.com/tvrusso/Xastir.git feature-supportglonass
 cd build/
 make -j3 2>&1 | tee make.log
 cp src/xastir ~/tmp/xastir-glonass
 sudo make install
 cd ~/tmp
 sudo cp xastir-glonass /usr/bin/xastir

pi@mhams:~/tmp $ ls -salt xastir*
3696 -rwxr-xr-x 1 pi   pi   3781092 Apr 11 10:22 xastir-glonass
3696 -rwxr-xr-x 1 pi   pi   3780648 Apr 11 10:16 xastir-normal
1176 -rwxr-xr-x 1 root root 1202212 Apr 11 10:03 xastir-pkg

Thank you Tom for all your time & support!

we7u, 2019-04-11:
No on the command-line option, but it could be easily added.
With respect to the maps, look in ~/.xastir/config/xastir.cnf and change the path if you switch between local build and a package install.

tvrusso, 2019-04-11:
Glad to know these changes fix your GPS issues with the DRAWS hat.

For your direct questions:

  - There is no way to get the Xastir version number from the command line at this time.  It's only displayed in the Help->About box.  That wouldn't be an unreasonable feature to add, though, but "-v" is already taken for specifying debug "verbosity" level.
  - The problem with maps is directly related to the /usr/ vs. /usr/local/ business.  When you installed Xastir from the package and ran it, it created a config file that had /usr/ paths coded in it --- and from then on it is looking for maps in /usr/share/xastir/maps.  But when you uninstall the package and build Xastir from source, that directory has been deleted, but the new build still wants to look for the maps there due to the existing config file left around in your .xastir/config directory.

This is why the very first part of the Xastir wiki page on building Xastir for Raspbian has a section telling you what to do if you had previously installed from a package:
http://xastir.org/index.php/HowTo:Raspbian

Note that the first step of that process is to preserve your maps directory, and the last step is to edit your xastir.cnf file.

Oh, and for the record, rather than copying executable files directly so that you can support multiple builds of Xastir with different executable names, you can do it just by having multiple build directories and using the "--program-suffix" option to configure:

```
   (assume your git clone is in ~/src/Xastir)
   cd ~/src/Xastir
   git checkout master
    mkdir build_master
    cd build_master
     ../configure --program-suffix="-master" [other options]
    make && sudo make install
   mkdir ../build_glonass
   cd ..
   git checkout feature-supportglonass
    cd build_glonass
  ../configure --program-suffix="-glonass" [other options]
   make && sudo make install
```

You'll now have two versions of xastir installed, with different suffixes.  You can rebuild as needed just by cding to the appropriate build directory after switching branches, and redoing "make && sudo make install" --- the two will never interfere with each other.

tvrusso, 2019-04-11:
Quick comment:  PR #58 now contains additional support for GLONASS, Beidou, and Gallileo sentences in addition to the GNSS sentences supported initially (and mistakenly referred to as GLONASS support).

closed 2019-04-11 by tvrusso

opg4759, 2019-04-12:
Thanks everyone for all the work and the quick resolution. I rebuilt from git this evening  and am now up and running on 2.1.1 with no GPSD issues and the DRAWS pi hat. 
73's
Josh KB6FJ.

tvrusso, 2019-04-12:
Glad you're up and running.  Thank you for bringing this issue to our attention, and thanks to Basil for providing the GPS module part number that was the key to identifying where the real problem lay.  Use it in good health!

PR #58 (2019-04-11): Add support for GNGGA and GNRMC sentences
GLONASS receivers send $GNGGA sentences instead of $GPGGA sentences, and
$GNRMC instead of $GPRMC.  But the only difference between these
sentences is the GN instead of GP --- the data is identical.

This appears to be the reason that users of the DRAWS hat are unable
to use this device with Xastir, even though it appears to work with
all other programs that consume NMEA sentences.

This commit attempts to recognize GGA and RMC sentences irrespective
of whether they begin with "$GP" or "$GN".  It is tested so far only
with GPS receivers, and it doesn't [... cut]
commit 2264a035 (2019-04-11): Merge pull request #58 from tvrusso/feature-supportglonass

Add support for sentences from GLONASS, GNSS, Beidou, and Gallileo geolocation receivers.

Closes #53.