Xastir/Xastir #65: Duplicate xastir windows appeared

godfreja, 2019-04-20:
While testing some code changes I saw multiple xastir windows appear. The stdout output indicated that one copy started and at a certain point multiple copies were competing for network ports. I ultimately fixed the problem by killing some old xastir processes that were running even though there were no xastir windows.

I suspect something isn't handled well (in the network code?) where a fork happens and if there is an error (due to say something else having a port open) the forked copy returns to the "normal" control flow instead of behaving as a proper child and two copies show on the screen. This is pure speculation.

This also raises the question of why I had some old xastir processes hanging around with no windows open.

godfreja, 2019-04-20:
Looking at the code I see this exact scenario (fork and then return to the normal control flow) can happen with the UDP spider server. (I haven't checked the TCP server.) The output on stdout indicates that this happened.

I will dig into this,

we7u, 2019-04-21:
That's a long-standing problem. Has to do with another fork or thread not
dying when Xastir stops running, then can interfere by keeping a port open
that the second Xastir wants to open.

tvrusso, 2019-04-21:
David Ranch (@dranch) reported also that this issue showed up after Xastir had crashed while manipulating the map (subject of bug #66).  The crash is a separate issue, but the stray processes it left are tripping the multiple-window issue of this bug.

PR #67 (2019-04-21): Ensure x_spider children don't return
If the UDP x_spider server couldn't open any ports (such as due to
another xastir process having them open) the child would return to
the main control logic of xastir, leading to two copies of xastir
running and duplicate windows appearing. Add an exit(1) after the
call to UDP_Server() so this can't happen. This fixes #65.

The TCP server didn't use return, but I added an exit(1) after the
call to TCP_Server() just to be safe.

I also fixed some warnings on MacOS compiles. This helps with
issue #24.
commit 87e291a8 (2019-04-21): Merge pull request #67 from godfreja/godfrejja/issue65

Ensure x_spider children don't return

Fixes #65.
commit aec59e38 (2019-04-21): Ensure x_spider children don't return

If the UDP x_spider server couldn't open any ports (such as due to
another xastir process having them open) the child would return to
the main control logic of xastir, leading to two copies of xastir
running and duplicate windows appearing. Add an exit(1) after the
call to UDP_Server() so this can't happen. This fixes #65.

The TCP server didn't use return, but I added an exit(1) after the
call to TCP_Server() just to be safe.

I also fixed some warnings on MacOS compiles. This helps with
issue #24.
closed 2019-04-21 by tvrusso