Xastir/Xastir #66: Xastir crashes on map manipulation, leaves stray processes

tvrusso, 2019-04-21:
David Ranch (@dranch) reports he had Xastir crash, leaving zombie processes.  As a result, when he fired Xastir up again he had the issue of double-windows reported in bug #65.

The actual crash, though, is distinct from the double-window issue, so I'm creating a new bug report about the crash itself.

From David's original report:
----
I was zooming in on a part of the map tonight and Xastir crashed.  Below is the abbreviated backtrace.  I then tried restarting Xastir and it threw the below errors and the double windows as reported in this thread.  After I closed one window, the other crashed with it and I saw two stray xastir processes running though there weren't any GUI windows open.

======= Backtrace: =========                                                             
/lib64/libc.so.6[0x33afa75e5e]                                                           
/lib64/libc.so.6[0x33afa78cad]                                                           
/usr/lib64/libXm.so.4(XmTextFieldSetString+0x464)[0x33b5768df4]                          
xastir[0x4813b3]                                                                         
xastir[0x47ee2f]                                                                         
/lib64/libpthread.so.0[0x33b0207aa1]                                                     
/lib64/libc.so.6(clone+0x6d)[0x33afae8c4d]  
. . .
7fffe8815000-7fffe8836000 rw-p 00000000 00:00 0                          [stack]
7fffe88a0000-7fffe88a2000 r--p 00000000 00:00 0                          [vvar]
7fffe88a2000-7fffe88a4000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Magick: abort due to signal 6 (SIGABRT) "Abort"...
Aborted

--------
# Now restarting Xastir that results in errors and two windows open

$ xastir                                                                                  
Found pid file: /home/dranch/.xastir/xastir.pid                                                                                   
Other Xastir process, pid: 25440 does not appear be running.                                                                      

Built-in map types:
      gnis   USGS GNIS Datapoints
       pop   USGS GNIS Datapoints w/population
       map   APRSdos Maps                     
       map   WinAPRS/MacAPRS/X-APRS Maps      
       pdb   PocketAPRS Maps                  

Support for these additional map types has been compiled in: 
       geo   Image Map (ImageMagick/GraphicsMagick library, many formats allowed)
       geo   URL (Internet maps via libcurl library)                             
       geo   URL (OpenStreetMaps via libcurl library                             
                  Copyright OpenStreetMap and contributors, CC-BY-SA)            
       shp   ESRI Shapefile Maps (Shapelib library)                              
       tif   USGS DRG Geotiff Topographic Maps (libgeotiff/libproj)              
       xpm   X Pixmap Maps (XPM library)                                         
x_spider: Can't bind local address for AF 2: 98 - Address already in use         
x_spider: Can't bind local address for AF 2: 98 - Address already in use         
Either this OS maps IPv4 addresses to IPv6 and this may be expected or           
could some processes still be running from a previous run of Xastir?             
Either this OS maps IPv4 addresses to IPv6 and this may be expected or           
could some processes still be running from a previous run of Xastir?             
x_spider: Can't bind local address for AF 10: 98 - Address already in use        
Either this OS maps IPv4 addresses to IPv6 and this may be expected or           
could some processes still be running from a previous run of Xastir?             
x_spider: Can't bind local address for AF 10: 98 - Address already in use        
Either this OS maps IPv4 addresses to IPv6 and this may be expected or           
x_spider: Couldn't open any sockets                                              
could some processes still be running from a previous run of Xastir?             
Unable to setup any x_spider server sockets.                                     
x_spider: Couldn't open any sockets                                              
Unable to setup any x_spider UDP server sockets.                                 
Could some processes still be running from a previous run of Xastir?             
UDP_Server process died.                                                         
x_spider: Couldn't set read-end of pipe_udp_server_to_xastir non-blocking        
Could some processes still be running from a previous run of Xastir?             
festival_client: connect to server failed                                        
festival_client: connect to server failed                                        
SayText: Couldn't open socket to Festival                                        
SayText: Couldn't open socket to Festival                                        
[... 98 lines cut ...]
$ ps aux | grep xastir
dranch   25441  2.9  0.0 159440   436 pts/4    S    14:38  11:28 xastir
dranch   25442  0.0  0.0 159440   420 pts/4    S    14:38   0:00 xastir

Running an strace on the first one shows this, scolling off the screen fast:
--
poll([{fd=4, events=POLLIN}, {fd=5, events=POLLIN}], 2, 0) = 0 (Timeout)
read(3, 0x7fffe88325cf, 1)              = -1 EAGAIN (Resource temporarily unavailable)
nanosleep({0, 1000000}, NULL)           = 0
poll([{fd=4, events=POLLIN}, {fd=5, events=POLLIN}], 2, 0) = 0 (Timeout)
read(3, 0x7fffe88325cf, 1)              = -1 EAGAIN (Resource temporarily unavailable)
nanosleep({0, 1000000}, NULL)           = 0
poll([{fd=4, events=POLLIN}, {fd=5, events=POLLIN}], 2, 0) = 0 (Timeout)
read(3, 0x7fffe88325cf, 1)              = -1 EAGAIN (Resource temporarily unavailable)
nanosleep({0, 1000000}, NULL)           = 0
poll([{fd=4, events=POLLIN}, {fd=5, events=POLLIN}], 2, 0) = 0 (Timeout)
read(3, 0x7fffe88325cf, 1)              = -1 EAGAIN (Resource temporarily unavailable)
nanosleep({0, 1000000}, NULL)           = 0
poll([{fd=4, events=POLLIN}, {fd=5, events=POLLIN}], 2, 0) = 0 (Timeout)
read(3, 0x7fffe88325cf, 1)              = -1 EAGAIN (Resource temporarily unavailable)
nanosleep({0, 1000000}, ^CProcess 25441 detached
. . . 
--

_Originally posted by @dranch in https://github.com/Xastir/Xastir/pull/64#issuecomment-485223279_

tvrusso, 2019-04-21:
I am quite concerned about this map manipulation crash, which I have so far been unable to reproduce.  The crash appears to be related to Magick calls, but that is not crystal clear.

Could you please provide the following additional information about your Xastir build:
  - What git sha-1 is reported in your Help->About box  (e.g. "Xastir V2.1.1 (1234567)")
  - Are you building with GraphicsMagick or ImageMagick?
  - What does your "About" box say about your Magick version (second line from the bottom)?
  - What OS?
  - What exactly where you doing with the map at the time it crashed?  Can you reproduce it?
  - What maps were selected in your map chooser?

tvrusso, 2019-04-21:
I'm very curious if this issue is the result of the merge of #40, which updated a lot of Magick code to discontinue use of deprecated API, and also added the download manager.

If it is possible to reproduce this crash easily, one might try rolling back to git commit 64dcb206, which was the last one before the download manager was merged in, to see if it is still reproducible.

I do not think the backtrace you posted will be helpful to pin down the issue.  You might have to build Xastir making sure to enable "-g" and NOT do a "make install-strip", then run the code making sure that it drops core when it crashes (that is, make sure you've got your "ulimit -c" set to unlimited, and do NOT use the "-t" option to Xastir, which disables core dumps).  Then, you can poke around with gdb in the core file to see where it *actually* is crashing.

dranch, 2019-04-24:
Hello Tom,

I can try working on reproducing this soon but until then, the requested details:

- In the about box, it only shows "Xastir v2.1.1" - there isn't any Git detail in there but looking in the git clone directory, "git log" shows commit 220ae44399cdfcb86de27c6fbad99e3b66cebe3c

- GraphicsMagick-1.3.31-2.el6.x86_64

- About Xastir shows the same 1.3.31 version

- OS:   Centos 6.10 with a custom kernel to enable AX.25 support

- At the time of the crash, I was trying to zoom in on a smaller section of the map.  The one caveot I can state is that I might have had the other stray xastir processes running with the two Xastir windows running as well but I didn't know it.  Seems this is consistently happening on my station since I updated to this newer Xastir.  Seems I might need to downgrade back to v2.0.9 unless fixes are coming soon.

- Maps:  I have Online/OSM_tiled_mapnik.geo and Online/nationalmap.gov/Topo1-10.geo but playing with this right now, it seems this nationalmap.gov map doesn't work (doesn't display anything).

we7u, 2019-04-24:
I'm running one of the OSM tiled mapnik maps but from another test server
elsewhere. Shouldn't be a difference there. Also running GraphicsMagick.
I'm on OpenSuSE-42.3.

How was Xastir installed? Via Git? If so you might want to update as there
have been a TON of changes in the last week, including adding the commit
number in parens after the version number int the About box. One of the
changes included making sure processes terminated.

How were you zooming in? Mouse drawn-box? PageUp/PageDown keys?
Pre-selected zoom levels using right-mouse-menu?

tvrusso, 2019-04-24:
Jason G. looks like he might have found a multithreading issue where the download manager may be attempting to write to the status line in an unsafe manner.  This memory issue could be the source of this bug.  Should know more when Jason has a little more time to hunt it down.

If the git hash does not appear in your about box, there may be something goofy in how your code is getting built -- the build process is *supposed* to put the current git hash into the box if the code was built out of a git clone.

As Curt mentioned just this second, the multi-window thing got fixed a few days ago, 8 days after the git version you just quoted.

If you continue to experience map crashes, you could try an experiment:  down on line 607 of the file "dlm.c" you'll find a call to "statusline".  Try commenting out this line and see if the map crashes continue.  If they go away, then my speculation (based on Jason's observation) that this crash is due to a non-gui thread accessing thread-unsafe functions in the gui thread may be on the right track.

tvrusso, 2019-04-24:
The commit number in the about box isn't new --- I added that ages ago in commit ba3b8079.

godfreja, 2019-04-24:
> One of the changes included making sure processes terminated. 

Just to clarify, my fix wasn't to make sure stray processes terminated. They will still be around messing up Xastir's server functionality, but with my change a second copy (and thus another window) won't be created during startup if this condition arises.

godfreja, 2019-04-24:
"killall xastir" before starting Xastir after a crash is a good step until we figure out how to make sure Xastir's children go away on a crash.

godfreja, 2019-04-25:
I tried triggering the address sanitizer crash I saw earlier in an environment where I would get a better backtrace tonight. No luck. I'll keep trying when I have time.

tvrusso, 2019-04-25:
The DLM thread writes its download status into the status line and so does the GUI thread, and I believe that this is really unsafe.  If you look at other places in Xastir where the "statusline" function is called, they're all running from the main thread --- but DLM is its own thread, so its use of statusline is kinda unique.  Your backtrace (from private email) shows that one thread is allocating data using XmSetTextField (or whatever) and also freeing it and another thread is accessing the data after it's been freed.

My speculation is that this crash requires that the download manager thread and the gui thread both be trying to write to the status line at once.  So if my speculation is right, to make it happen with high probability you'd want to be connected to an interface that has an almost constant stream of packet data (say, a full, unfiltered feed from an APRS-IS server) *and* be doing tons of map movement that triggers map downloads.  

I still have never seen this crash, so I can't help out.  I even tried adding an unfiltered feed from CWOP and doing some map panning and zooming (with lots of tiles downloaded) and didn't see it.

we7u, 2019-04-25:
As I recall any writing to the GUI has to be done through the main thread.
I tried years ago to write to it in another thread or two and it didn't
work. I don't recall if things crashed or things just didn't get written.

tvrusso, 2019-04-25:
I am no pthreads OR an X11 programmer but looking over the code in dlm.c, it appears to me that the function DLM_transfer_thread, where the statusline call lives, is the one that is being run in the non-gui thread.

I am definitely seeing status line updates by the download manager indicating that it is still downloading tiles and how many it still has left to fetch.

But I believe I have read many times that it is a bad idea to be calling X functions from anywhere but the main thread, because they are not thread safe.  A quick googling comes up with many hits about that fact.

So I believe that if this isn't *actually* the reason for the crashes, it's somewhere close to it.  Jason's partial stack trace from email was this, and you can see that the read after free is happening in thread 0 downstream of a statusline call, but the free of the memory is in thread 4, also downstream of a statusline call.

==34621==ERROR: AddressSanitizer: heap-use-after-free on address 0x6060000629c0 at pc 0x00010fdf25e8 bp 0x7ffee0de5400 sp 0x7ffee0de4ba0
READ of size 23 at 0x6060000629c0 thread T0
==34621==WARNING: external symbolizer didn't start up correctly!
==34621==WARNING: Failed to use and restart external symbolizer!
    #0 0x10fdf25e7 in wrap_memcpy (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/10.0.0/lib/darwin/libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x1c5e7)
    #1 0x10f830f7e in XDrawString (/opt/X11/lib/libX11.6.dylib:x86_64+0x1ef7e)
    #2 0x10f643637 in DrawText (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xc5637)
    #3 0x10f6432ae in DrawTextSegment (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xc52ae)
    #4 0x10f639738 in RedisplayText (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xbb738)
    #5 0x10f63b8e0 in XmTextFieldSetString (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xbd8e0)
    #6 0x10eee9913 in statusline (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000dc913)
    #7 0x10ef892c4 in draw_OSM_tiles (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10017c2c4)
    #8 0x10ef7ba5b in draw_geo_image_map (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10016ea5b)
    #9 0x10ef6562c in draw_map (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10015862c)
    #10 0x10ef6dbd4 in load_maps (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x100160bd4)
    #11 0x10eee90a0 in create_image (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000dc0a0)
    #12 0x10eee4285 in new_image (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000d7285)
    #13 0x10ef4d185 in UpdateTime (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x100140185)
    #14 0x10f7c619d in XtAppProcessEvent (/opt/X11/lib/libXt.6.dylib:x86_64+0x1919d)
    #15 0x10f7bb636 in XtAppMainLoop (/opt/X11/lib/libXt.6.dylib:x86_64+0xe636)
    #16 0x10ef4bfd2 in main (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10013efd2)
    #17 0x7fff6f8d1084 in start (/usr/lib/system/libdyld.dylib:x86_64+0x17084)

0x6060000629c0 is located 0 bytes inside of 53-byte region [0x6060000629c0,0x6060000629f5)
freed by thread T4 here:
    #0 0x10fe2d10d in wrap_free (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/10.0.0/lib/darwin/libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x5710d)
    #1 0x10f63b830 in XmTextFieldSetString (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xbd830)
    #2 0x10eee9913 in statusline (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000dc913)
    #3 0x10eee0d34 in DLM_transfer_thread (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000d3d34)
    #4 0x7fff6fac333c in _pthread_body (/usr/lib/system/libsystem_pthread.dylib:x86_64+0x333c)
    #5 0x7fff6fac62a6 in _pthread_start (/usr/lib/system/libsystem_pthread.dylib:x86_64+0x62a6)
    #6 0x7fff6fac2424 in thread_start (/usr/lib/system/libsystem_pthread.dylib:x86_64+0x2424)

previously allocated by thread T0 here:
    #0 0x10fe2cf53 in wrap_malloc (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/10.0.0/lib/darwin/libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x56f53)
    #1 0x10f7added in XtMalloc (/opt/X11/lib/libXt.6.dylib:x86_64+0xded)
    #2 0x10f63c0b8 in ValidateString (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xbe0b8)
    #3 0x10f63b841 in XmTextFieldSetString (/usr/local/opt/openmotif/lib/libXm.4.dylib:x86_64+0xbd841)
    #4 0x10eee9913 in statusline (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000dc913)
    #5 0x10ef892c4 in draw_OSM_tiles (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10017c2c4)
    #6 0x10ef7ba5b in draw_geo_image_map (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10016ea5b)
    #7 0x10ef6562c in draw_map (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10015862c)
    #8 0x10ef6dbd4 in load_maps (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x100160bd4)
    #9 0x10eee90a0 in create_image (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000dc0a0)
    #10 0x10eee4285 in new_image (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x1000d7285)
    #11 0x10ef4d185 in UpdateTime (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x100140185)
    #12 0x10f7c619d in XtAppProcessEvent (/opt/X11/lib/libXt.6.dylib:x86_64+0x1919d)
    #13 0x10f7bb636 in XtAppMainLoop (/opt/X11/lib/libXt.6.dylib:x86_64+0xe636)
    #14 0x10ef4bfd2 in main (/Users/godfrey/xastbuild/src/./xastir:x86_64+0x10013efd2)
    #15 0x7fff6f8d1084 in start (/usr/lib/system/libdyld.dylib:x86_64+0x17084)

Thread T4 created by T0 here:
[... cut ...]
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==34621==ABORTING

godfreja, 2019-04-25:
When I triggered the above partial backtrace I was running xastir over a remote connection. I wonder if the added latency helps trigger the issue.

we7u, 2019-04-25:
Quick test to see if we're on the right track: Comment out the DLM code that tries to write to the GUI. If the problem goes away we've got it and can work on thread-safe code to do that later.

tvrusso, 2019-04-25:
Yes, I had suggested commenting out that code a few comments back.

we7u, 2019-04-25:
Trying that out on this branch:

https://github.com/we7u/Xastir/tree/dlm-threading

I haven't seen the problem yet myself though.

dranch, 2019-04-29:
Hello Curt,

Thanks for working on this.  Should we test using the "dlm-threading" branch or the main "Dev" branch?  I've refrained from trying any of the new branches recently as you all seem to be very busy doing compiler warning cleanups which can be disruptive at times.  Please let me know what you'd like us to pull and test.

we7u, 2019-04-29:
https://github.com/Xastir/Xastir
master branch

We're mostly through the work of fixing compiler warnings for various
compilers and OS'es. The ones we know about anyway. There are a few minor
ones remaining on the issues list.

tvrusso, 2019-05-04:
@dranch :  Have you had a chance to test out the new master branch to see if it is still crashing for you?  This open bug is important enough that we don't want to release the next version until it's fixed, but we need confirmation either that it's squashed or that we were on the wrong track and it still needs work.

dranch, 2019-05-04:
I have some spare time today so I'll try out the master branch now

dranch, 2019-05-05:
Things are looking good here in terms of zoom in/out, panning, etc.

we7u, 2019-05-06:
Shall we consider this one fixed for good? If so I'd like to remove the code completely that I commented-out in the DLM module: Will make sure it won't get re-enabled in the future.

tvrusso, 2019-05-06:
It's tempting to declare it fixed.  But since it is so intermittent and 
Jason only saw the memory access errors in rare circumstances with his
clang memory debugger tool, I am a bit scared to jump to the conclusion.
It seems plausible that removing the GUI code that might have been thread
safe (and was showing up in Jason's crash trace) was enough to completely
eliminate the problem, but since it was so hard to reproduce in the first 
place, we can't really be so sure.

Removing the code you commented out is still the right thing to do, because
it was definitely unsafe, and while it is unfortunate that the DLM won't
display any progress information, it is probably difficult to find 
a thread safe way to do so.

we7u, 2019-05-06:
One method: A queue that the main thread reads and deletes from / Any other
thread can can only write to. The main thread is able to update the GUI
status line. It's too much work to go through in this case, but in theory
should work just fine. Keep it in mind for later if we want to have other
threads do similar things.

tvrusso, 2019-05-06:
Agreed, probably more work to go through for this than it's worth.  So long as we're not crashing, we should leave it alone.

Let's give this a few more days of testing (and @dranch:  please keep trying to use it so we can have more confidence it's fixed) and then call it done.

we7u, 2019-05-06:
Hah. Was already implented:

statusline() at main.c:3938
-and-
stderr_and_statusline at main.c:3953

main.c:UpdateTime() calls check_statusline_timeout() periodically in the
main loop.

We could add code back into dlm.c, switching to statusline() instead of
trying to write directly to the status line from another thread like the
old code did.

tvrusso, 2019-05-06:
I would suggest moving a little more cautiously.  We have already said we want a release soon, and this bug is a blocker to that.  Let's make sure that the statusline call from the DLM thread was the real culprit, and leave it out if we are convinced.  

After the next release, we can put a thread-safe statusline update back in.

we7u, 2019-05-06:
I'm ok with waiting until after the release to put that code in.

we7u, 2019-05-06:
Looking at the code that was pulled out: He was using statusline() but then
did a Motif call afterward, presumably to speed up the display of the
message. That Motif call most likely caused the threading issue, the rest
was probably ok. I created a local branch with the correct fixes, it just
won't do status updates as fast as one might wish in that case. I'll hold
off on doing a pull request on it, but statusline() is used all over our
code and should be thread-safe, as other threads can write to it, but only
the main process loop reads/deletes from it and updates the GUI.

godfreja, 2019-05-06:
statusline() calls XmTextFieldSetString(). Doesn't that leave us in the same place with two threads calling X functions? Or is XmTextFieldSetString supposed to be thread safe?

tvrusso, 2019-05-06:
Maybe it is the case that the call to XmUpdateDisplay was causing a problem.  Still, Jason's stack trace shows the main thread trying to read data in XmTextFieldSetString that was deleted in the dlm thread in XmTextFieldSetString -- and in both cases the call is from statusline. 

It is worth experimenting with, but since it's been hard enough to reproduce this problem (only David has seen it regularly, and only Jason has ever reproduced it and then rarely), I think we need a little more to go on before trying to reinstate the feature on the speculation that it's the XmUpdateDisplay call that's the real root of the problem.

tvrusso, 2019-05-08:
Recommend we go ahead and close this, as the last thing we have heard is that the map zoom/pan is no longer causing crashes.

Any other issues surrounding this bug (such as reinstating status line updates in a thread-safe manner) should just be separate bug reports for after this next release.

tvrusso, 2019-05-08:
Closing this based on current thinking that we've got it addressed.  If it resurfaces, we can reopen or create a new issue referencing this one.

closed 2019-05-08 by tvrusso