Xastir/Xastir #72: Potential problem with mixture of long/unsigned long/abs in map_tif.c

tvrusso, 2019-04-22:
map_tif.c contains some lines of code that cause clang on OS X (and clang in general on other OSen) to throw some warnings about use of abs() on an unsigned expression.

Investigating, this is not just a warning that should be removed while working on #24, but an actual usage error that probably needs fixing correctly, not just warning silence effort.  It could (but does not in any case I've experimented with) produce wrong answers in addition to warnings.

A typical line in map_tif.c that causes this warning is:
```
        xastir_left_x_increment = (float)
            (1.0 * abs(SW_x_bounding_wgs84 - NW_x_bounding_wgs84)   // Need to add one pixel worth here yet
            / abs(SW_y - NW_y));
```
In this code, xastir_left_x_increment is of type float, SW_x_bounding_wgs84 and NW_x_bounding_wgs84 are both unsigned longs, and SW_x and NW_x are both signed ints.

The warning is this:

```
map_tif.c:1999:20: warning: taking the absolute value of unsigned type 'unsigned long' has no effect [-Wabsolute-value]
            (1.0 * abs(SW_x_bounding_wgs84 - NW_x_bounding_wgs84)   // Need to add one pixel worth here yet
                   ^
map_tif.c:1999:20: note: remove the call to 'abs' since unsigned values cannot be negative
            (1.0 * abs(SW_x_bounding_wgs84 - NW_x_bounding_wgs84)   // Need to add one pixel worth here yet
                   ^~~

```
The problem here is that according to standard C expression typing rules, the difference between two unsigned ints is itself an unsigned int --- so even if SW_x_bounding_wgs84 is smaller than NW_x_bounding_wgs84, the difference between them is NEVER negative --- it's an unsigned value, and is carried out in unsigned arithmetic.  abs() is supposed to take a signed int and return a signed int

This code has existed in this way in map_tif.c since map_tif.c was split out from maps.c in commit 2bcc1e9, and existed in maps.c since the beginning of history in commit fa09e5594.

But it's  technically wrong (according to C type conversion rules) whenever the difference in the numerator's abs call, but **_clang's suggestion to remove the call  (which does indeed silence the warning) actually causes it to produce wrong answers_**, which is even worse.

The little test program below, when compiled with clang, throws the same warning.  In this case, a-b in "correct" arithmetic is -100, and in correct arithmetic foo1 should be 0.8333....  It actually gets that with the abs call, even though the warning is thrown.  The warning from clang *tells* us to remove the abs, but doing so produces a ridiculous result.  

However, casting the unsigned longs into longs before doing the subtraction both removes the warning *and* produces correct result.  Further, the line where that's done replaces "abs" with "labs" because "abs" is supposed to work on ints, and "labs" should be used on (signed) longs.  Clang bitches about that, too.

GCC compiles this with no complaints, and produces the same results.  

I am opening this as a bug report because the correct solution should probably be up for discussion, rather than  just fixing it carelessly.  The original code was clearly designed to handle cases where a map's bounding box might have been inverted left-right/up-down in some way (hence using abs to try to determine the step size per pixel), but it should be carefully reviewed and the fix done right so we both get rid of warnings like this *and* get the code to work correctly in all cases.

```
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char **argv)
{

  unsigned long a = 100;
  unsigned long b = 200;
  int c = 50;
  int d = 170;
  float foo1;
  float foo2;
  float foo3;

  foo1 = (float)(1.0*abs(a-b)/abs(c-d));
  foo2 = (float)(1.0*(a-b)/abs(c-d));
  foo3 = (float)(1.0*labs((long)a-(long)b)/abs(c-d));

  printf("With abs, %f, without, %f, with casts, %f\n",foo1,foo2,foo3);
}

```

tvrusso, 2019-04-22:
Slightly modified test program.  Here, it appears that labs(a-b) in the second printf gets the right result only because we're counting on the representation of the unsigned long being just right so that when labs gets it and thinks it's a signed long, it has the correct value of "100" as far as it's concerned.  But that's a dangerous thing to count on.
```

#include <stdio.h>
#include <stdlib.h>

int main(int argc, char **argv)
{

  unsigned long a = 100;
  unsigned long b = 200;
  int c = 50;
  int d = 170;
  float foo1;
  float foo2;
  float foo3;

  foo1 = (float)(1.0*abs(a-b)/abs(c-d));
  foo2 = (float)(1.0*(a-b)/abs(c-d));
  foo3 = (float)(1.0*labs((long)a-(long)b)/abs(c-d));

  printf("With abs, %f, without, %f, with casts, %f\n",foo1,foo2,foo3);
  printf("a = %lu, b=%lu, a-b=%lu, labs(a-b)=%ld\n",a,b,(a-b),labs(a-b));
}

```

we7u, 2019-04-28:
I did some fixes in PR #104 for the OSX warnings. Now going back and reading this issue I may have done it incorrectly for the things you're talking about here. The compiler warnings are gone now, but we need to go back over the code to determine whether additional changes are needed.

tvrusso, 2019-04-29:
Yes, if you just followed Clang's instructions to remove the "unnecessary" abs, you broke the code, because Clang's wrong.  The "right" thing to do is to cast the things to (long) before subtracting, then take labs of the difference.

The problem introduced by simply removing the abs won't show up unless you happen to have a raster whose NW and NE corners are somehow swapped so the NW pixel has a smaller X coordinate than the NE  (can that ever happen?  The code is clearly written specifically to take care of just that case, and it's in exactly that case that removing the abs breaks it).

As indicated by the little test programs above, subtracting a larger unsigned from a smaller yields a wrong but still positive unsigned (because the difference between two unsigneds is unsigned) --- and taking abs of that simply casts the unsigned long result into an int (and as written, not even a long int), and it just happens to be OK because the binary representation of the unsigned result happens to be the same as the binary representation of the desired negative value.  This is probably going to be true on all modern and even future hardware (it is unlikely the binary representation of negative ints will ever change), but it is still an incorrect thing to be relying on, and the compiler is right to warn about it.

we7u, 2019-04-29:
Can go back to Travis-CI build 108 for OSX and see the original warnings, although of course the line numbers will be slightly different in the current code.  I may want to check out the other files that were mentioned in that build (and were fixed in my pull request) to assure those were fixed properly also.

tvrusso, 2019-04-29:
I have gone ahead and identified the two problematic commits that were included in #104.  These could easily be reverted and fixed in the manner described in that comment, i.e. by taking abs((int)a-(int) b) in the case of a and b being "unsigned" or labs((long)a -(long)b) where a and b are unsigned long.

Another choice (which eliminates the whole issue) is to not use unsigneds in the first place.

A third approach, needed, if the extra bit of range really is necessary and signed variables just won't do, would be not to use "abs" or "labs" to compute the absolute value of their difference, swap them so that you're always subtracting the smaller from the larger.

we7u, 2019-04-29:
Closed by PR #107.

closed 2019-04-29 by we7u