Xastir/Xastir #79: Interface_gui can try to sprintf into a null pointer

tvrusso, 2019-04-23:
The following warnings are thrown by GCC8 on FreeBSD:
```

../../Xastir/src/interface_gui.c: In function 'Config_TNC.part.5':
../../Xastir/src/interface_gui.c:664:17: warning: argument 1 null where non-null expected [-Wnonnull]
                 sprintf(tmp, langcode("WPUPCFT029"), (int)device_type);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../Xastir/src/snprintf.h:27,
                 from ../../Xastir/src/interface_gui.c:28:
/usr/include/stdio.h:285:6: note: in a call to function 'sprintf' declared here
 int  sprintf(char * __restrict, const char * __restrict, ...);
      ^~~~~~~
../../Xastir/src/interface_gui.c:664:17: warning: null destination pointer [-Wformat-overflow=]
                 sprintf(tmp, langcode("WPUPCFT029"), (int)device_type);
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I've looked over the code, and it looks like this:

```
    tmp=(char *)NULL;                                                       
                                                                                
    if(!config_TNC_dialog) {
//...
        switch(device_type) {
            case DEVICE_SERIAL_TNC:
                tmp=langcode("WPUPCFT001");
                break;
//...
            default:
                sprintf(tmp, langcode("WPUPCFT029"), (int)device_type);
                break;
        }

```
This means that under most circumstances, the code is properly copying a pointer returned by langcode into the tmp variable, but if it falls through to the default, it will try to sprintf into a null pointer.  It may be the case that this default block is never executed (it means a device type, which can only be selected from an enumerated list, has a value outside of the range of enumerated values).  But the code should NOT be written so that this default block would cause the code to call sprintf to "print" using an invalid pointer.    An actual temporary array should be used, and tmp should have that array's pointer copied into it.

tvrusso, 2019-04-30:
I don't really think this qualifies for the "3 Major" label.  While sprintfing into a null pointer would be serious, I believe that the way the code is structured it really should never be reachable as the code currently exists.  Tripping it would require that a developer adds a new device type and forgets to expand this switch statement to handle it.

we7u, 2019-04-30:
Yea, change at will. I said earlier it was an initial attempt.

we7u, 2019-05-07:
Closed by PR #114

closed 2019-05-07 by we7u