Xastir/Xastir #88: Use after free in list_gui.c

godfreja, 2019-04-24:
Encountered a use after free bug while running xastir with clang's -fsanitize=address

To trigger this I opened the "Own objects and items" view and then clicked on one of the icons of my objects.

```
==26037==ERROR: AddressSanitizer: heap-use-after-free on address 0x60200000d610 at pc 0x000109acaa61 bp 0x7ffee61f4e40 sp 0x7ffee61f4e38
READ of size 1 at 0x60200000d610 thread T0
    #0 0x109acaa60 in Call_locate_station list_gui.c:409
    #1 0x10a3a96ea in XtCallCallbackList (libXt.6.dylib:x86_64+0x16ea)
    #2 0x10a1ce67f in ActivateCommon (libXm.4.dylib:x86_64+0x5667f)
    #3 0x10a3db12a in HandleActions (libXt.6.dylib:x86_64+0x3312a)
    #4 0x10a3d8f4a in _XtTranslateEvent (libXt.6.dylib:x86_64+0x30f4a)
    #5 0x10a3b5a6c in XtDispatchEventToWidget (libXt.6.dylib:x86_64+0xda6c)
    #6 0x10a3b6246 in _XtDefaultDispatcher (libXt.6.dylib:x86_64+0xe246)
    #7 0x10a3b5dd8 in XtDispatchEvent (libXt.6.dylib:x86_64+0xddd8)
    #8 0x10a3c10fb in XtAppProcessEvent (libXt.6.dylib:x86_64+0x190fb)
    #9 0x10a3b6636 in XtAppMainLoop (libXt.6.dylib:x86_64+0xe636)
    #10 0x109b49092 in main main.c:27601
    #11 0x7fff6f8d1084 in start (libdyld.dylib:x86_64+0x17084)

0x60200000d610 is located 0 bytes inside of 1-byte region [0x60200000d610,0x60200000d611)
freed by thread T0 here:
    #0 0x10aa0f10d in wrap_free (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x5710d)
    #1 0x109acb94c in Station_List_fill list_gui.c:620
    #2 0x109aceb8e in update_station_scroll_list list_gui.c:1038
    #3 0x109b4a85a in UpdateTime main.c:11643
    #4 0x10a3c119d in XtAppProcessEvent (libXt.6.dylib:x86_64+0x1919d)
    #5 0x10a3b6636 in XtAppMainLoop (libXt.6.dylib:x86_64+0xe636)
    #6 0x109b49092 in main main.c:27601
    #7 0x7fff6f8d1084 in start (libdyld.dylib:x86_64+0x17084)

previously allocated by thread T0 here:
    #0 0x10aa0ef53 in wrap_malloc (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x56f53)
    #1 0x10a3a8ded in XtMalloc (libXt.6.dylib:x86_64+0xded)
    #2 0x10a2351df in XmTextFieldGetString (libXm.4.dylib:x86_64+0xbd1df)
    #3 0x109acb98d in Station_List_fill list_gui.c:621
    #4 0x109ad282f in Station_List list_gui.c:2481
    #5 0x10a3a96b1 in XtCallCallbackList (libXt.6.dylib:x86_64+0x16b1)
    #6 0x10a1d19ea in ActivateCommonG (libXm.4.dylib:x86_64+0x599ea)
    #7 0x10a1a9983 in _XmDispatchGadgetInput (libXm.4.dylib:x86_64+0x31983)
    #8 0x10a25e035 in _XmMenuBtnUp (libXm.4.dylib:x86_64+0xe6035)
    #9 0x10a3db12a in HandleActions (libXt.6.dylib:x86_64+0x3312a)
    #10 0x10a3d9409 in HandleSimpleState (libXt.6.dylib:x86_64+0x31409)
    #11 0x10a3d8ae7 in _XtTranslateEvent (libXt.6.dylib:x86_64+0x30ae7)
    #12 0x10a3b5a6c in XtDispatchEventToWidget (libXt.6.dylib:x86_64+0xda6c)
    #13 0x10a3b6246 in _XtDefaultDispatcher (libXt.6.dylib:x86_64+0xe246)
    #14 0x10a3b5dd8 in XtDispatchEvent (libXt.6.dylib:x86_64+0xddd8)
    #15 0x10a3c10fb in XtAppProcessEvent (libXt.6.dylib:x86_64+0x190fb)
    #16 0x10a3b6636 in XtAppMainLoop (libXt.6.dylib:x86_64+0xe636)
    #17 0x109b49092 in main main.c:27601
    #18 0x7fff6f8d1084 in start (libdyld.dylib:x86_64+0x17084)

SUMMARY: AddressSanitizer: heap-use-after-free list_gui.c:409 in Call_locate_station
Shadow bytes around the buggy address:
  0x1c0400001a70: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
  0x1c0400001a80: fa fa fd fd fa fa fd fd fa fa fd fa fa fa fd fd
  0x1c0400001a90: fa fa 00 00 fa fa 00 00 fa fa 00 fa fa fa fd fd
  0x1c0400001aa0: fa fa fd fd fa fa fd fd fa fa fd fa fa fa fd fd
  0x1c0400001ab0: fa fa fd fa fa fa fd fa fa fa fd fa fa fa fd fd
=>0x1c0400001ac0: fa fa[fd]fa fa fa 00 00 fa fa 00 00 fa fa 00 fa
  0x1c0400001ad0: fa fa fd fa fa fa fd fa fa fa 01 fa fa fa fd fd
  0x1c0400001ae0: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
  0x1c0400001af0: fa fa fd fd fa fa fd fd fa fa 00 00 fa fa 00 00
  0x1c0400001b00: fa fa 00 fa fa fa fd fd fa fa fd fd fa fa fd fd
  0x1c0400001b10: fa fa fd fd fa fa fd fd fa fa fd fd fa fa fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==26037==ABORTING

```

godfreja, 2019-04-24:
Not sure if it is worthwhile opening issues for things I find right away before really trying to debug them, but I figure more documentation is better than less. For now I plan on filing bugs unless the fix is immediately obvious. Let me know if I should show more restraint.

tvrusso, 2019-04-24:
This actually sounds like a bug to me, and documenting it as an issue
at least lets us keep it in our sights for a fix.

I prefer reporting bugs when they're found, even if there's nobody to hunt
down the fix immediately.

we7u, 2019-04-24:
I think what you're doing is the correct method.

godfreja, 2019-04-24:
I have close to zero experience programming X guis outside of other wrapper libraries. Is the following  expected behavior?

I added a printf at line 624 of list_gui.c to show whenever a callback is registered. 
`printf("Adding callback for type %d row %d (%s)\n", type, row, SL_callback[type][row]);`
When I opened the own objects at items dialog I saw:

Adding callback for type 6 row 0 ()
Adding callback for type 6 row 1 ()
Adding callback for type 6 row 2 ()
Adding callback for type 6 row 0 (HUGO)
Adding callback for type 6 row 1 (MSFINISH)
Adding callback for type 6 row 2 (WILLOWRVR)

I wouldn't be shocked if the fact that each row is setup twice is related to our use after free bug. But what I wasn't expecting is that several seconds later with the program just sitting there I once again got:
Adding callback for type 6 row 0 (HUGO)
Adding callback for type 6 row 1 (MSFINISH)
Adding callback for type 6 row 2 (WILLOWRVR)

And then some time later another batch of:
Adding callback for type 6 row 0 (HUGO)
Adding callback for type 6 row 1 (MSFINISH)
Adding callback for type 6 row 2 (WILLOWRVR)

Would we expect Station_List_fill to get called multiple times with the program just siting there?

tvrusso, 2019-04-24:
I believe Station_List_fill has to be called repeatedly, because the station
list may be update as new data comes in.  Looking over all the places where
it may be called (and therefore all the places where line 624 might get run),
it is not at all surprising to me that it is doing this, though it shouldn't
be doing any accessing of data after freeing it.

PR #90 (2019-04-24): Fix use after free bug in Station_List_fill
Station_List_fill gets called repeatedly. Each run it
would add a callback to the icon button. On the next run
it would free the string that was passed to the callback
but not delete the old callback. Over time we would develop
a stack of old callbacks on the icon button that would
point to freed strings.

Add logic to remove the existing callback before we free
the string from the previous run.

Fixes #88
commit 00174886 (2019-04-24): Fix use after free bug in Station_List_fill

Station_List_fill gets called repeatedly. Each run it
would add a callback to the icon button. On the next run
it would free the string that was passed to the callback
but not delete the old callback. Over time we would develop
a stack of old callbacks on the icon button that would
point to freed strings.

Add logic to remove the existing callback before we free
the string from the previous run.

Fixes #88
closed 2019-04-24 by we7u