Xastir/Xastir #91: AddressSanitizer: stack-buffer-overflow in display_station db.c:2889

godfreja, 2019-04-25:
Using clang's -fsanitize=address option after double clicking on the icon of a repeater in the "all stations" dialog the following error was encountered:

```
==43728==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffee23c107f at pc 0x00010d858aac bp 0x7ffee23c0e50 sp 0x7ffee23c0e48
READ of size 1 at 0x7ffee23c107f thread T0
    #0 0x10d858aab in display_station db.c:2889
    #1 0x10d85ac11 in display_file db.c:3910
    #2 0x10d9184b1 in create_image main.c:3431
    #3 0x10d913325 in new_image main.c:13072
    #4 0x10d97c225 in UpdateTime main.c:11475
    #5 0x10e1f219d in XtAppProcessEvent (libXt.6.dylib:x86_64+0x1919d)
    #6 0x10e1e7636 in XtAppMainLoop (libXt.6.dylib:x86_64+0xe636)
    #7 0x10d97b072 in main main.c:27601
    #8 0x7fff6f8d1084 in start (libdyld.dylib:x86_64+0x17084)

Address 0x7ffee23c107f is located in stack of thread T0 at offset 543 in frame
    #0 0x10d855d6f in display_station db.c:2640

  This frame has 16 object(s):
    [32, 52) 'temp_altitude' (line 2641)
    [96, 116) 'temp_course' (line 2642)
    [160, 180) 'temp_speed' (line 2643)
    [224, 244) 'dr_speed' (line 2644)
    [288, 346) 'temp_call' (line 2645)
    [384, 434) 'wx_tm' (line 2646)
    [480, 510) 'temp_wx_temp' (line 2647)
    [544, 584) 'temp_wx_wind' (line 2648) <== Memory access at offset 543 underflows this variable
    [624, 644) 'temp_my_distance' (line 2649)
    [688, 708) 'temp_my_course' (line 2650)
    [752, 772) 'temp1_my_course' (line 2651)
    [816, 866) 'temp2_my_gauge_data' (line 2652)
    [912, 920) 'tmp' (line 2658)
    [944, 952) 'ambiguity_coord_lon' (line 2666)
    [976, 984) 'ambiguity_coord_lat' (line 2666)
    [1008, 1012) 'temp' (line 3048)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism or swapcontext
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow db.c:2889 in display_station
Shadow bytes around the buggy address:
  0x1fffdc4781b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1fffdc4781c0: 00 00 00 00 00 00 00 00 00 00 00 00 f1 f1 f1 f1
  0x1fffdc4781d0: 00 00 04 f2 f2 f2 f2 f2 00 00 04 f2 f2 f2 f2 f2
  0x1fffdc4781e0: 00 00 04 f2 f2 f2 f2 f2 00 00 04 f2 f2 f2 f2 f2
  0x1fffdc4781f0: 00 00 00 00 00 00 00 02 f2 f2 f2 f2 00 00 00 00
=>0x1fffdc478200: 00 00 02 f2 f2 f2 f2 f2 00 00 00 06 f2 f2 f2[f2]
  0x1fffdc478210: 00 00 00 00 00 f2 f2 f2 f2 f2 00 00 04 f2 f2 f2
  0x1fffdc478220: f2 f2 00 00 04 f2 f2 f2 f2 f2 00 00 04 f2 f2 f2
  0x1fffdc478230: f2 f2 00 00 00 00 00 00 02 f2 f2 f2 f2 f2 00 f2
  0x1fffdc478240: f2 f2 00 f2 f2 f2 00 f2 f2 f2 f8 f3 00 00 00 00
  0x1fffdc478250: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==43728==ABORTING

```

tvrusso, 2019-04-25:
Pretty clear what's up there.  The code in question at the line indicated
is:

           if (temp_wx_wind[strlen(temp_wx_wind)-1] == ' ') {
                temp_wx_wind[strlen(temp_wx_wind)-1] = '\0';  // delete blank at EOL
            }

If strlen(temp_wx_wind) is zero.

The same pattern happens again a few lines down.

It may be that there are some stations where the weather pointer is not
null, but many of the entries in it are zero, in which case nothing gets
sprintf'd into temp_wx_wind.

Looks like one could just change this (in both cases) to:

          if (strlen(temp_wx_wind)>0 && temp_wx_wind[strlen(temp_wx_wind)-1] == ' ') {
                temp_wx_wind[strlen(temp_wx_wind)-1] = '\0';  // delete blank at
 EOL
          }

and clear the memory access error.

tvrusso, 2019-04-25:
Looks like when the DataRow's weather element is first set up (which happens on first receipt of a weather packet), it is initialized with basically all string elements empty.  So if a station doesn't HAVE a wind sensor or humidity sensor (say it just has a temperature sensor), it may be that nothing ever gets written in there, and so the block of code that would write to temp_wx_wind never does anything.  I think adding the strlen check before trying to access characters of the array based on strlen is the right fix.

godfreja, 2019-04-25:
Doing a quick scan through db.c (a 19,542 line file!) there may be a similar issue around line 18507.  I'm wondering if the best option in that case is to return in line is length 0. In any case that should probably be a separate issue.

tvrusso, 2019-04-25:
Yes, there are some truly enormous files in Xastir.  db.c isn't the largest --- it's second only to main.c at 27,614 lines.  You know, we could shorten it if we removed lines of commented out code from 15 years ago... ;)

As for the lines at 18507, yeah, looks like if that function gets an empty string to decode it should just return FALSE --- just as it does if the line is too long right above line 18504.

It should probably also be fixed, and probably as a separate issue.

PR #92 (2019-04-25): Fix strlen()-1 buffer overflow in display_station()
If strings were empty a accessing strlen(-1) goes out of bounds.
Add a strlen() > 0 check.

Fixes #91
commit 49b9625f (2019-04-25): Fix strlen()-1 buffer overflow in display_station()

If strings were empty a accessing strlen(-1) goes out of bounds.
Add a strlen() > 0 check.

Fixes #91
closed 2019-04-25 by tvrusso