Xastir/Xastir #93: Missing map display bookmarks

tvrusso, 2019-04-25:
As of very recently, my map display bookmarks is missing entries that I use all the time.  If I try to re-create them with a new name, the old one shows up again, and the new name doesn't.

I think we have a memory error in the list gui code.

tvrusso, 2019-04-25:
The ~/.xastir/config/locations.sys file shows all of my bookmarks, including the new one.  It's the one at the end of the list that is no longer showing up.

This appears to be a consequence to the change in location_gui.c of PR #75.  If I revert commit 57735df7, the full list of all my bookmarks reappears.

I have tried this without doing that revert, but by removing the "locations_db.dat" file that gets created and letting it be regenerated, but the behavior remains --- if commit 57735df7 is not reverted, the last one in the input database is missing.  If it is reverted, they all show.

tvrusso, 2019-04-25:
I see what the problem is.

Commit 57735df changed the "200" in the call to sort_input_database at line 192 of location_gui.c to "sizeof(name)".  This results in writing a locations_db.dat file with 100 character records.  But no other "200" was changed in this way, and we're apparently trying to *read* 200 character records all over the place.

I'm going to revert 57735df now, then come back to this bug when it's not an obscene hour to try to hunt down the correct fix.  It may be as simple as changing the 100 character "name" array to 200 chars, but there should probably not be a bunch of magic numbers all over this code.

tvrusso, 2019-04-25:
I neglected to mention this issue in commit 0f534e59, my revert of 57735df.  So instead I'll mention the commit here.

tvrusso, 2019-04-25:
Actually, it's not just the last bookmark that's missing, it's every other one --- because the code was reading 200 byte records, but only writing 100 byte records.

godfreja, 2019-04-25:
Interesting. I specifically looked for the missing every other bookmark when I tested the change because I was concerned about that possibility and I didn't (and still don't) see it. I do have the missing last entry which I missed before. Sorry about that.

tvrusso, 2019-04-25:
Since the revert of the commit from #75 has restored bookmark functionality, I'm closing this bug.

If you want to go back over the code that has the buffer overrun issue and fix it so that all the reads and writes of the database use the same size records (and possibly centralize definition of that constant instead of having it hard-coded several times all over the file), go for it.  I have to withdraw for a few days from activity on Xastir.  This memory overrun has been with us since the dawn of time in commit fa09e55, so if it gets fixed, great, but if it has to wait it's not like anyone's noticed it before in the 20-odd years that Xastir has existed.

closed 2019-04-25 by tvrusso

we7u, 2019-04-25:
Should we open another bug so this doesn't get forgotten, plus reference this closed bug report in it?

tvrusso, 2019-04-25:
Yes, I think opening a new bug that's specific to the buffer overrun issue is appropriate.  This bug, being specific to the map display bookmarks  (which is now working correctly), is rightly closed.  Referring back to it so that you can see what issues can arise is perfectly reasonable.