Xastir/Xastir #98: Festival speech synthezier + systemd + PulseAudio: No sound

we7u, 2019-04-25:
Changes must be made to festival to allow the binary "festival" to talk to Pulseaudio. Changes are documented here:

http://xastir.org/index.php/HowTo:Festival

Once you make those changes "festival" will be able to use Pulseaudio but "festival_server" run by systemd cannot. The festival binary is run as the user, while festival_server is run as the "festival" user. The "festival" user has no permissions to talk to any other user's Pulseaudio. Solutions may involve tweaking permissions for the user's Pulseaudio or running "festival_server" as root.

tvrusso, 2019-04-25:
Just a comment:  The instructions there are extremely system specific, even as far as the initial setup.  On both of the systems I have that use festival, they're inapplicable because the setup specifically assumes you're running pulseaudio *and* that the system wasn't already configured when the package was installed.  That's not an assumption that is really clear.  Also, there should be a "siteinit.scm" that's meant to hold site-specific modifications of the configuration rather than editing the main config files installed by the festival package.

On BSD I just have to run festival --server and it Just Works with no configuration changes needed, and BSD doesn't normally need pulseaudio to handle sound.  On Ubuntu 16.04 it's also just a matter of running that same command, and I haven't ever had to tweak its installation at all.  Your instructions are primarily important when festival is being started up by systemd and the festival package has not been configured to work out of the box, neither of which was the case for me on Ubuntu.

You might want to clarify that this particular set of HowTos certainly applies to your system, and tell what that system is.  It may have wider applicability, but shouldn't read as a general set of instructions implying it's what's necessary to get it working.  It's just what it was necessary for *you* on your Suse system.

we7u, 2019-04-25:
I was working with Ken Koster on this one and we both run OpenSUSE. Yours is the first report I've heard that Festival works these days, so that's new information. The Wiki page is a work in progress and I've spent almost zero time on it. I'll add OpenSuSE notes to it and mention that it works on BSD and Ubuntu-16.04.

tvrusso, 2019-04-25:
Festival never stopped working for me, I just rarely bother with it since there's so little for it to say to me around here anymore (APRS traffic 'round these parts is 99% digipeaters, short-lived receive-only igates, fixed "openspot" beacons, and one-way trackers these days).

we7u, 2019-06-05:
This doesn't appear to be an Xastir issue but a Festival / OS-config issue. Will document changes necessary to get Festival running on the Xastir.org HowTo:Festival page.

closed 2019-06-05 by we7u