Xastir/Xastir #108: Config not finding Xpm library

we7u, 2019-04-29:
OpenSuSE-42.3: I have /usr/include/X11/xpm.h and /usr/lib/libXpm.so.4 and libXpm.so.4.11.0

configure is not linking in the Xpm library during it's test for XPM, so I'm not able to do Snapshots from Xastir. I have verified that the "XpmWriteFileFromPixmap" function is in the libXpm.so.4 library.

I haven't figured out how to get configure to link against the -lXpm library to verify functionality and compile that library in.

Configure.ac has this:
```
# Check for XPM
if test "${use_lsb}" = "no"; then 
  AC_CHECK_HEADERS(X11/xpm.h, AC_SEARCH_LIBS(XpmWriteFileFromPixmap, Xpm, AC_DEFINE(HAVE_LIBXPM, , Define to 1 if you have the `Xpm' library (-lXpm). )))
  AC_CHECK_HEADERS(xpm.h, AC_SEARCH_LIBS(XpmWriteFileFromPixmap, Xpm, AC_DEFINE(HAVE_LIBXPM, , Define to 1 if you have the `Xpm' library (-lXpm). )))
  AC_CHECK_HEADERS(Xm/XpmI.h, AC_SEARCH_LIBS(XmeXpmWriteFileFromPixmap, Xm, AC_DEFINE(HAVE_LIBXPM_IN_XM, , Define to 1 if the `Xm' library (-lXm) is used for Xpm. )))
else
  AC_DEFINE(HAVE_LIBXPM, , Define to 1 if you have the `Xpm' library (-lXpm). )
  AC_DEFINE(HAVE_LIBXPM_IN_XM, , Define to 1 if the `Xm' library (-lXm) is used for Xpm. )
  AC_DEFINE(HAVE_X11_XPM_H, , Define to 1 if you have 'X11/xpm.h' )
fi
```

config.log shows this:

```
configure:11285: checking X11/xpm.h usability
configure:11285: gcc -c -g -O2 -pipe -W -Wall -Wpointer-arith -Wstrict-prototypes -Wno-unused-parameter -pthread -I/usr/local/include -I/usr/include/libgeotiff  conftest.c >&5
configure:11285: $? = 0
configure:11285: result: yes
configure:11285: checking X11/xpm.h presence
configure:11285: gcc -E -I/usr/local/include -I/usr/include/libgeotiff  conftest.c
configure:11285: $? = 0
configure:11285: result: yes
configure:11285: checking for X11/xpm.h
configure:11285: result: yes
configure:11290: checking for library containing XpmWriteFileFromPixmap
configure:11321: gcc -o conftest -g -O2 -pipe -W -Wall -Wpointer-arith -Wstrict-prototypes -Wno-unused-parameter -pthread -I/usr/local/include -I/usr/include/libgeotiff  -L/usr/local/lib   conftest.c -lrt -lXm -lXt -lXext -lm   -lSM -lICE -lX11  -lcurl -lproj -lshp -lpcre >&5
conftest.c:169:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 char XpmWriteFileFromPixmap ();
 ^
conftest.c:171:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 main ()
 ^
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: /tmp/ccHOHP7w.o: in function `main':
/home/archer/src/we7u/xastir/github-we7u/Xastir/build/conftest.c:173: undefined reference to `XpmWriteFileFromPixmap'
collect2: error: ld returned 1 exit status
configure:11321: $? = 1
```

we7u, 2019-04-29:
As a brute-force experiment I edited config.h to tell it we did have LIBXPM:
```
/* Define to 1 if you have the `Xpm' library (-lXpm). */
#define HAVE_LIBXPM 1
```
```
ldd xastir | grep -i xpm
libXpm.so.4 => /usr/lib64/libXpm.so.4 (0x00007f0b5fbe5000)
```
Interesting: If I go through the normal build process Xastir gets linked against that library anyway, but config.h variables don't show that XPM is available. This means that map_OSM.c / map_WMS.c / map_dos.c / map_geo.c / map_gnis.c / map_pdb.c / map_shp.c / map_tif.c and maps.c get compiled without XPM support and we lose snapshots. We probably also lose minimal support for XPM maps if IM or GM aren't present.

tvrusso, 2019-04-29:
Just to confirm:  This is only a problem when you've got imagemagick and graphicsmagick both disabled, right?

My guess is you have to figure out what package is supposed to provide libXpm.a, and just having the shared libraries and headers isn't enough.  The configure script is trying to link against libXpm.a, I think, and I don't think there's a way to make it find things if only the shared library is around.

we7u, 2019-04-29:
This used to work, and it should be trying against the shared library, not the static library. This is on a system with graphicsmagick installed and Xastir is finding it, so this is a completely normal installation with no extra flags enabling/disabling things. I'm horrible with the configure.ac stuff and was hoping you could jump in and help fix this one.

we7u, 2019-04-29:
I'm thinking we should probably have an output in the summary for XPM as well (maybe called "XPM Snapshots") as otherwise there's no indication that we lost this functionality. I can do the formatting of the summary piece if someone else figures out the configure.ac pieces to discover the library properly. I'd just need a final variable that was a combination of all the other XPM variables we have that tells for sure whether we're using it.

tvrusso, 2019-04-29:
Never mind about my question about magick --- I see clearly that configure is simply not picking up libXpm when it tries to compile the little test program it creates, and just says you don't have libXpm at all.

What's really strange, though, is that it hasn't put "-lXpm" into the gcc line you've posted.

I think AC_SEARCH_LIBS first tries to link its test program with no libraries added, to see if the current content of LIBS already satisfies the dependency.  Only if that fails does it start trying the libraries listed in the the second argument.  Looks like the snippet of config.log you posted is just that first try, and not the second.  Could you please attach the full config.log to help track down what's really going on.  The secret is probably in the lines following the ones you've posted.

tvrusso, 2019-04-29:
Confirmed by looking at my own config.log that -lXpm is not found the first time AC_SEARCH_LIBS tries its compile, because "-lXpm" is not present in LIBS yet.  The second time it tries, with "-lXpm" appended, it is found, and so "-lXpm" is added to LIBS.

What we have to figure out is why that second try doesn't work for you.  The full config.log will point the way, or at least it should.

we7u, 2019-04-29:
[config.log](https://github.com/Xastir/Xastir/files/3129170/config.log)

tvrusso, 2019-04-29:
Also, your configure.ac appears to have an extra line that mine does not:
  AC_CHECK_HEADERS(xpm.h, AC_SEARCH_LIBS(XpmWriteFileFromPixmap, Xpm, AC_DEFINE(HAVE_LIBXPM, , Define to 1 if you have the `Xpm' library (-lXpm). )))

Did you add that?  It's wrong, I think, because it's searching for "xpm.h" without a prefix, but we never include that file without the X11/ prefix (inside an #ifdef HAVE_X11_XPM_H).

we7u, 2019-04-29:
Yes. I was screwing with things trying to get it to work and didn't set it
all the way back to defaults. Doing so now and will attach a new config.log.

tvrusso, 2019-04-29:
Your config.log clearly shows libXpm being found , and also showing HAVE_LIBXPM being set.  You should even see that happening on the console, where it is saying "-lXpm" after "Checking for library containing XpmWriteFileFromPixmap" (which happens after "Checking X11/xpm.h presence" and "Checking for X11/xpm.h", both of which are saying "yes").

we7u, 2019-04-29:
Yea, but look at config.h... HAVE_LIBXPM is not set, so the code doesn't
get compiled in.

tvrusso, 2019-04-29:
I can't see your config.h --- you didn't post it.  Config.log *says* it's putting it into confdefs.h,  so if it's not getting into config.h, something is really screwy.

we7u, 2019-04-29:
Fun. "We don't support that file type." I'll copy/paste the text:

```
$ grep -i xpm config.h
/* Define to 1 if you have the `Xpm' library (-lXpm). */
#define HAVE_LIBXPM /**/
/* Define to 1 if the `Xm' library (-lXm) is used for Xpm. */
/* #undef HAVE_LIBXPM_IN_XM */
/* Define to 1 if you have 'X11/xpm.h' */
#define HAVE_X11_XPM_H 1
/* Define to 1 if you have the <Xm/XpmI.h> header file. */
/* #undef HAVE_XM_XPMI_H */
```

tvrusso, 2019-04-29:
OK.  So there's HAVE_LIBXPM being defined, right there along with HAVE_X11_XPM_H.

There is no problem in evidence here.  Do XPM snapshots in fact not work?

we7u, 2019-04-29:
They were not working for me. I thought you had to have a "1" after the
define in config.h. Not true?
I tried snapshots earlier today and they did not work. I can give it
another shot in a bit.

tvrusso, 2019-04-29:
Not true.

#define foobar

#ifdef foobar
   blah
#endif
will work just fine.

If it had not found libXpm, it would have had "#undef HAVE_LIBXPM".

Look at all the other lines of config.h where it's saying it found stuff successfully.  They all have those "/* */" things.  The error here is in the comments that somebody added to the AC_DEFINE call, saying to put 1 in there.  That's wrong, it should just say "Define if... " not "Define to 1 if..."

we7u, 2019-04-29:
Ok. Just killed and restarted, now can generate kml or png snapshots, so
chalk it up to user error plus me not understanding the defines. I told you
I suck at the configure.ac stuff!

we7u, 2019-04-29:
I'd still like to add a line to the summary about XPM/Snapshots.

tvrusso, 2019-04-29:
That would be fine to add to the summary, since the failure to find Xpm is so quietly ignored elsewhere, and has this effect.

closed 2019-04-29 by tvrusso