Xastir/Xastir #110: Create tagging system for Xastir project

we7u, 2019-04-30:
I implemented a simple system of 5 priority tags on 04/30/2019 based on this URL:
https://github.com/isaacs/github/issues/472
Reference the post by andrewschultz that also mentions PyGithub.

Other ideas that may be better:

https://softwareengineering.stackexchange.com/questions/129714/how-to-manage-github-issues-for-priority-etc

https://medium.com/@dave_lunny/sane-github-labels-c5d2e6004b63

I like that last one the best so far.

tvrusso, 2019-05-01:
My personal opinion is that for as small a project as we have, as few active developers as we have, and as short our bursts of activity are, the five-level priority system you've implemented is completely adequate.

I think that the other ideas, while they are very good, are overkill for this project's current needs.

we7u, 2019-05-01:
No problemo. Agreed.

closed 2019-05-01 by we7u