Xastir/Xastir #116: Reformat sources to new coding standard

we7u, 2019-05-08:
Reformat Xastir sources per our new coding standards. The new standards were set May 7, 2019, 12:35 PM on the Xastir and Xastir-Dev lists after a vote:
```
2-space indents
Vertical brace alignment
Always use braces even when optional
```
See: http://xastir.org/index.php/HowTo:Developer_Guidelines_and_Notes
for more details. The Git project has Vim and Emacs config files added which help indentation.

NOTE: On hold until -AFTER- our next release.

we7u, 2019-05-08:
Not that this only applies to our *.c and *.h files, not our Makefiles or others.

we7u, 2019-06-06:
Looking at alert.c: Are we not going with the vertically-aligned braces for enums and similar too? I did for Davis/Lacrosse/callpass code.

tvrusso, 2019-06-06:
On Thu, Jun 06, 2019 at 09:55:27AM -0700, we recorded a bogon-computron collision of the <notifications@github.com> flavor, containing:
> Looking at alert.c: Are we not going with the vertically-aligned braces for enums and similar too? I did for Davis/Lacrosse/callpass code.

I didn't do that, but have nothing against it.  You can fix those if you like.
And I'll make sure to do it for the next round of files I attack.  I'm working
on awk.c and awk.h now, and I'll get those.

-- 
Tom Russo    KM5VY
Tijeras, NM  

 echo "prpv_a'rfg_cnf_har_cvcr" | sed -e 's/_/ /g' | tr [a-m][n-z] [n-z][a-m]

we7u, 2019-06-06:
See the response with 77 votes here:
https://stackoverflow.com/questions/2355834/how-can-i-autoformat-indent-c-code-in-vim
They use "astyle" from inside VIM. I may try that 'cuz it'll make a big job easier. At least if it can understand #ifdef's.

we7u, 2019-06-06:
So far it seems to work. Added this to Xastir/src/.vimrc:

" Using 'Artistic Style'or 'astyle' to autoformat/autoindent
" In VIM type 'gg' to go to the top of the file, then 'gqG'
autocmd BufNewFile,BufRead *.c,*.cpp,*.h set formatprg=astyle\ -A1\ -s2\ -S\ -xt0\ -xW\ -F\ -p\ -P\ -j

See http://astyle.sourceforge.net/astyle.html for details on the options.

we7u, 2019-06-06:
Tom: What's the most complicated set of #ifdef's you saw in our code? I want to run "astyle" against that and see what it does.

tvrusso, 2019-06-06:
Sorry, the only "complicate" ifdef issue I found was this sort of thing:

     #ifdef FOO
          if (bar) {
     #else 
          if (baz) {
     #endif 
            stuff
          }

that is, where the opening brace was inside the ifdef and the closing wasn't.  Emacs went nuts over that and couldn't tell that it shouldn't indent it as:

     #ifdef FOO
          if (bar) {
     #else 
            if (baz) {
     #endif 
              stuff
            }

I already took that out in the place where I encountered it during our last indentation fire drill (where I had brought lots of files back into compliance with our older standard), during our race to remove warnings.

Can't recall any other "difficult" ifdef issues.

we7u, 2019-06-06:
Take a look at we7u/Xastir, the "reformat" branch. I reformatted everything in Xastir/src and Xatir/src/rtree using:
  astyle -A1 -s2 -S -xt0 -xW -F -p -P -j
Which is:
  --style=allman (-A1)
  --indent=spaces=2 (-s2)
  --indent-switches (-S)
  --indent-continuation=0 (-xt0)
  --indent-preproc-block (-xW)
  --break-blocks=all (-F)
  --pad-oper (-p)
"   --pad-paren (-P)
"   --add-braces (-j)
Docs are at: http://astyle.sourceforge.net/astyle.html

we7u, 2019-06-06:
I added two more:
  --lineend=linux (-z2)
  --convert-tabs (-c)
To give a total command of:
  astyle -A1 -s2 -S -xt0 -xW -F -p -P -j -z2 -c

we7u, 2019-06-07:
" Using 'Artistic Style'or 'astyle' to autoformat/autoindent
" In VIM type 'gg' to go to the top of the file, then 'gqG'
" Long options:
"   --style=allman (-A1)
"   --indent=spaces=2 (-s2)
"   --indent-switches (-S)
"   --indent-continuation=0 (-xt0)  SKIP?
"   --indent-preproc-block (-xW)
"   --break-blocks=all (-F)         SKIP?
"   --pad-oper (-p)                 SKIP?
"   --pad-paren (-P)                SKIP?
"   --add-braces (-j)
"   --convert-tabs (-c)             SKIP?
"   --lineend=linux (-z2)
I now have a test branch "reformat3" where I ran: "astyle -A1 -s2 -S -xW -j -z2" across the *.h and *.c files,  doing fewer changes to the spacing.

we7u, 2019-06-09:
Lastest is now the "reformat4" branch at we7u/Xastir. Please check it out.

closed 2019-06-10 by tvrusso

we7u, 2019-06-10:
I updated https://xastir.org/index.php/HowTo:Developer_Guidelines_and_Notes to show the .vimrc and command-line versions of the reformatting command used. For future reference.

tvrusso, 2019-06-10:
I have also updated the emacs stuff on the wiki, which had been incomplete.  

The settings in the directory-local variables file can't set all the 
indentation customization, as some of that is not just variable setting, but 
rather function invocation.  If only the directory-local variables get set, 
then you actually don't get the correct indentation we want (braces indented 
wrong, case statements are not how you like them).

we7u, 2019-06-10:
FWIW I decided NOT to add the .vimrc invocation of "astyle" 'cuz when I tried it inside VIM it changed formatting of comments and I didn't want to go through all the code again to see what else it might have changed. Better to invoke "astyle" from the command-line and get repeatable results each time. Glad this is behind us but we may need to invoke it periodically over coming years if style starts deviating. Can refer to the Developer_Guidelines_and_Notes later for the exact command.

commit e9d10e62 (2019-06-10): Merge pull request #145 from we7u/reformat4

Format code to new coding standards

Closes #116