Xastir/Xastir #121: NMEA sentences not beginning with the usual $GP

IZ1GCL, 2019-05-09:


IZ1GCL, 2019-05-09:
A previous release of Xastir was dealing only with sentences beginning with $GP.
My GPS was sending sentences starting with $II (Integrated Instrumentation), but there are many more prefixes. So the program was not recognizing the GPS.
I want to be sure that the new release will not consider the first two characters of the sentences.
May be this issue has already considered before
73
Giuseppe
IZ1GCL

tvrusso, 2019-05-09:
This is a duplicate of issue #53 and was resolved in pull request #58.  The new release will work correctly with all current generation

tvrusso, 2019-05-09:
Xastir was updated to support $GN, $GL, $GA, and $GB with pull request #58 in response to issue #53.

If there is a need for $II support it'll have to be added later, and with more checking to make sure that the RMC and GGA sentences with that prefix are compatible with the others.

tvrusso, 2019-05-09:
Fixing this would only require the two functions isGGA and isRMC in gps.c to be updated.  Currently, they look for the $G prefix, then the GGA or RMC suffix (respectively) and then the constellation identifier in the third character.  These functions were created precisely to make it easier to detect GGA and RMC sentences without hard-coding "$GPGGA" and "$GPRMC" in several places throughout the code.

This won't be in the next release, though, but it should be possible to add it quickly afterward (so it'll be available in a master-branch build).

tvrusso, 2019-05-09:
We should probably modify isGGA and isRMC to ignore the talker ID entirely, as there are many more than previously encountered:

http://www.catb.org/gpsd/NMEA.html#_talker_ids

Until this release, we were rejecting everything but $GP.  There's no good reason for that.  If it starts with "$", is followed by any two characters, and ends with "GGA" or "RMC" we should probably accept it as a GGA or RMC.

godfreja, 2019-05-09:
I was going to suggest the same thing.

tvrusso, 2019-05-09:
PR added.  This was so simple it might be appropriate to just do it for the next release, even though we were starting to start about locking it down.  I'll leave it up to Curt and Jason to decide whether it should be in the release, in which case they should add this to the Kanban "In progress" column, then merge it if appropriate.

tvrusso, 2019-05-09:
Fixed by merge of PR #122

closed 2019-05-09 by tvrusso

godfreja, 2019-05-09:
Fixed with PR 122