Xastir/Xastir #123: Create some release notes for the next release

tvrusso, 2019-05-10:
Our next release should contain release notes describing the changes since the last release.  These should be pasted into the text box for the release description when we create the actual release  (see, for example, https://github.com/Xastir/Xastir/releases/tag/Release-2.1.0).

Here's what I have in my notes for what we've done since 2.1.0.  I have certainly overlooked something, and this is too wordy to be used directly as the release notes:

- National Weather Service shapefile script updated, dbfawk files added

- Compilation errors in OS X earlier than 10.6 fixed

- Tiger WMS server maps added

- A new set of National Map .geo files has been added for access to current versions of the National Map WMS products, replacing those that referenced WMS products that have gone off-line since the last release.

- GDAL and OGR support removed (it was never very functional)

- Xastir builds are now reproducible (builds no longer contain the build date and time)

- Deprecated GraphicsMagick and ImageMagick function calls replaced with new API

- Download of tiled maps is now handled by a download manager running in a separate thread, dramatically improving map update speed.

- Obsolete interface to old Tiger/Line map server removed (the server went down years ago)

- OpenTrac decoding support removed

- Support for the old PocketAPRS (PalmOS based APRS) map format has been removed.  Users of these old map types should transition to a more modern format.

- Xastir no longer ships with a copy of an old version of shapelib source.  Shapelib support must now always be provided by an external library, since most distros have shapelib in their package management systems.

- The embryonic "Xastir-qt" project has been moved to a separate repository.

- "Xastir -V" will now cause Xastir to report its version number and exit.  The version number will contain the git SHA-1 hash if Xastir was built directly from sources in a git clone.

- Support for many additional satellite navigation (GNSS) receivers has been added to the existing GPS support by expanding the range of NMEA GGA and RMC sentences we recognize (irrespective of NMEA "Talker ID").  Prior to this release, only NMEA sentences starting with $GP were recognized.

- Many compilation warnings from recent GCC and Clang compilers have been eliminated, resulting in a cleaner build.

- Our Makefiles now default to "silent rules": during compilation, only the compilation phase (CC, LINK, etc) is printed with the name of the current file being processed.  The old, noisier behavior may be invoked by adding "V=1" to the "make" line at build time.

we7u, 2019-05-10:
Added: "Xpm / Snapshots" to configure summary.

tvrusso, 2019-05-23:
I propose the following, which could just be cut and pasted directly into the appropriate box when we do the github release.  It is basically all the stuff mentioned above, with wording pared down slightly.

- National Weather Service shapefile script updated, dbfawk files added.

- Compilation errors in OS X earlier than 10.6 fixed.

- Tiger WMS server maps added.

- A new set of National Map .geo files replaces those that referenced WMS products that have gone off-line since the last release.

- GDAL and OGR support removed.

- Xastir builds are now reproducible (builds no longer contain the build date and time).

- Deprecated GraphicsMagick and ImageMagick function calls replaced with new API.

- Download of tiled maps is now handled by a download manager running in a separate thread, dramatically improving map update speed.

- Obsolete interface to old Tiger/Line map server removed (the server went down years ago).

- OpenTrac decoding support removed.

- Support for the old PocketAPRS (PalmOS based APRS) map format has been removed.  Users of these old map types should transition to a more modern format.

- Xastir sources no longer contains a copy of an old version of shapelib source.  Shapelib support must now always be provided by an external library.

- The embryonic "Xastir-qt" project has been moved to a separate repository.

- "Xastir -V" will now cause Xastir to report its version number (with git sha-1 hash appended if built from a git clone) and exit. 

- Support for many additional satellite navigation (GNSS) receivers has been added to the existing GPS support.  Prior to this release, only NMEA sentences starting with $GP were recognized.

- Many compilation warnings from recent GCC and Clang compilers have been eliminated, resulting in a cleaner build.

- Our Makefiles now default to "silent rules" to reduce uninteresting output during compliation.  The old, noisier behavior may be invoked by adding "V=1" to the "make" line at build time.

- The configure summary now shows the status of its probe for Xpm libraries under "Xpm / Snapshots".

tvrusso, 2019-05-30:
Hearing no objection to the suggestions above, I'm going to declare the previous comment our "Release notes" and will use them as is when I push out the release this evening.

closed 2019-05-30 by tvrusso