Xastir/Xastir #126: Festival should not be saving the speech new station selection.

n7ipb, 2019-05-14:
In the speech settings Xastir should never save the status of the 'new station' flag.   

Conditions:   1. New build from Git-Hub,  
2. Delete .xastir/config directory so the config file is created new.
3. Start Xastir and enter your callsign.
4. Turn on New Station speech flag
5. Exit Xastir
6. Restart Xastir and the flag will still be on

A quick look shows the code to initialize it to off still exists (complete with comment about why) so something must be over writing it.

I currently don't have time to continue looking.

tvrusso, 2019-05-15:
The comment you refer to (which you wrote) has been in Xastir since its very earliest import into sourceforge's CVS on 2 Feb 2002, and presumably predates that import by a long while.

The code to save the festival_speak_new_station setting was put into xa_config.c in commit fcdb2aa9 less than two weeks after that initial import (on 14 Feb 2002) and has been there ever since.

The reading and writing of this flag from the config file could easily be removed by taking out two lines of xa_config.c.

That it's been in place for 17 years without much comment suggests that either nobody turns this on routinely, or that festival isn't used much, or that folks aren't running buffered connections much and haven't been having trouble with the very issue that makes saving this flag a bad idea.

Still, fixing it is a two line change, and might be worth doing.

n7ipb, 2019-05-15:
Now that I reflect on it I seem to remember some discussion at that time about allowing the user to bite themselves in the butt this way.   I just never expect it to bite me.

While we could change it I agree it's probably not used that often or those that do use speech just have ignored it.  Rather than change it now lets leave it be.   The whole speech portion could use an overhaul especially given the state of festival support and I have some ideas on that just waiting for enough ATU's.

closed 2019-05-15 by n7ipb