Xastir/Xastir #129: Upgrade to a safer string library

we7u, 2019-05-17:
The string functions in the standard C library are lacking as well as unsafe: Some can't guarantee strings will be null-terminated, some can write past the end of buffers, overwriting other data structures or code. In the past we attempted to make string operations a bit safer by using xastir/src/snprintf.c. It was a step in the right direction but didn't go far enough. Newer compilers show hundreds of possible issues in string operations with the current Xastir code.

I propose we look for a better string library, replacing most/all of our string operations with said library. My initial proposal to the other Xastir admins was [Better String Library](https://github.com/websnarf/bstrlib) but that may require extensive changes to the code to implement. Some current contenders (not an exhaustive list):

- Better String Library
- Apache Portable Runtime or APR
- GLib
- Simple Dynamic Strings Library
- SafeStr
- ISO/IEC TR 24731 Functions
- strlcpy() and strlcat()
- Managed String Library
- C++ std::string
- Annexia's c2lib library
- stralloc/qmail
- wxString
- Vstr
- libmib astrings
- fireString
- Safe C String Library
- Libsrt
- SDS
- Unicode libraries (there are many)

Here's an article comparing some of the above and listing even more: [String library comparison](http://www.and.org/vstr/comparison). Comments are welcome from anyone who has used string libraries in place of the standard library functions.

we7u, 2019-05-21:
Section 6 of this is a very good read:
[Secure Programming HOWTO](https://dwheeler.com/secure-programs/Secure-Programs-HOWTO.html)
plus check out the man-page for asprintf() / vasprintf() which look like good additional functions to use, mentioned in section 6.2.4 of the above link.

tvrusso, 2020-02-23:
While I understand why this issue was opened, I think it should be removed as a blocker to the next release.  It's still listed in the ToDo column of https://github.com/Xastir/Xastir/projects/2.