Xastir/Xastir #138: Rework GUI dialogs: Add manager, and scrollbars when necessary

we7u, 2019-06-04:
Dialogs were added/extended from the original code which used hard-coded connections beween the window frame and the widgets. There was no included window manager and most dialogs are also missing scrollbars. Dialogs were forced to fixed sizes so resizing wouldn't mess up the widget stackup. This made it difficult to use on small screen sizes as portions of the dialogs would be off-screen and with no scrollbar people couldn't get to parts of the dialog.

Change dialogs to include a manager widget / add scrollbars where needed / make them resizeable.

tvrusso, 2019-06-04:
Is this issue minor enough that you know how to fix it for the next release?  If so, might want to add it to the ToDo column of the 2.2.0 kanban.

we7u, 2019-06-04:
There are quite a few dialogs so I don't know quite when they'll all be done. Might take me a while.

tvrusso, 2019-06-04:
Well, at this point we've been releasing less often than once a year.  Your call whether to commit to it as a 2.2.0 feature.  If you decide on it, add it to the KanBan, and you can always remove it if it ain't gonna happen by next year.

we7u, 2020-02-24:
First tweaks pushed as branch "dialogs". More to do yet.

godfreja, 2020-02-27:
I believe Curt still has work to do on this and this should be reopened. Curt can close this if I am wrong.

tvrusso, 2020-02-27:
Yes, indeed, this was mistakenly closed when I foolishly added it as a "related issue" to the PR, thinking that when Github said "Merging this PR may close these issues" it meant "may" not "will."

I will not do that again.

we7u, 2020-02-28:
I have code running that checks screen size and dialog size, opening the dialog up to the smaller of the two (Thanks to Jason Godfrey for the idea!). If it finds a smaller screen size it also opens the dialog at 0,0 (upper left corner of the screen). So far this seems to work well. I'm continuing work on those features plus modifying the rest of the dialogs. This should be ready in the next few weeks for a pull request, sooner if I can manage it.

we7u, 2020-03-04:
The latest code (yet to be pushed to my Xastir fork) looks at the screen size when creating a dialog. 

If dialog is larger than the screen size, it opens the dialog up at size screen-size-minus-decorations, activates scrollbars, and locates it at 0,0 (upper left). I hard-coded numbers of "6" and "27" to compensate for the left/right and top/bottom decoration sizes, but that depends on the window manager selected, its settings, and the font used.

If dialog is smaller than screen size, open it like normal and just set max dialog size. It can be made smaller by the operator at which point scrollbars will appear.

I also hard-coded a "+10" where I set max dialog sizes: This appears to be the size of the added frame just inside the decorations but outside the form. I have more dialogs to convert but will push changes (to my fork) soon so others can test.

I haven't found a general way to compute decoration sizes on the system after trying quite a few methods. It looks like hard-coding decoration sizes is our only option. I'm interested in any comment about that. Perhaps we can later create variables for those, settable on some configuration dialog?

tvrusso, 2021-10-17:
It's been over a year since this issue was last updated, so I've moved it back out of "in progress" and into the "to do" column.

Curt, if you consider it complete (or even "as much as I'm gonna do") then let's close it.

tvrusso, 2024-05-28:
I'm gonna go ahead and remove this from the Xastir 2.2.2 project.  Progress was made on the effort, but it's been 4 years since any work was done on it.