Xastir/Xastir #139: Do we use the Xastir/po and Xastir/intl directories/contents?

we7u, 2019-06-05:
I created a branch and deleted the "po" and "intl" directories from it. Xastir compiles/installs just fine. I've not ever touched this stuff, has anyone else? Can we get rid of them?

tvrusso, 2019-06-05:
These have been present in the repo since the very first "rc1" commit by Chuck Byam back in 2002.  They appear to be related to an old way of doing internationalization via GNU gettext and its libintl library.   Last references to the internal "intl" directory (in which intl/Makefile was configured) were removed from configure and various makefiles in commit c698d17.

Near as I can tell, this is dead code.

I am not sure what the "po" directory was for, but it appears also  to be related to internationalization with gettext. 

But in digging into this, I find that we still say we depend on libintl in configure.ac, but in fact our internationalization is not done using gettext at all, but using a simple system native to Xastir that has also existed since the very first "rc1" commit.  That code uses the functions in lang.c instead of anything provided by gettext.  Oddly, plenty of files in Xastir include "<libintl.h>" but do not appear to use anything provided by that header.

My guess is that not only do we not use the intl and po directories, we don't actually use gettext and libintl at all, even though configure probes for them *AND* we tell everyone we depend on it and we link against it.  I would bet we can remove the code that probes for (and adds to LIBS) libintl and libintl.h from configure.ac, and remove every instance of "#include <libintl.h>" from source code, and Xastir would work just fine with no impact on our internal internationalization stuff.

If you want to get rid of those two directories, you should probably also get rid of all the vestiges of Xastir's one-time use of gettext at the same time.

tvrusso, 2019-06-05:
Digging further:  The files that include libintl.h simply do this:

#ifdef  HAVE_LIBINTL_H
#include <libintl.h>
#define _(x)        gettext(x)
#else   // HAVE_LIBINTL_H
#define _(x)        (x)
#endif  // HAVE_LIBINTL_H

There is EXACTLY one place in which the "_(x)" macro is used, and it's in main.c:

t = _("X Amateur Station Tracking and Information Reporting");

This means that only on one place is there any use of gettext remaining in Xastir, and it's doing nothing because we no longer have any "po" files that actually handle translation of this string.

So if we remove all references to libintl.h from source, we also have to remove the "_()" macro usage from this one line of main.c.

we7u, 2019-06-05:
I see this also:
configure.ac:# Check for intl library (Needed for Cygwin)
However if that one main.c reference is changed it should remove Cygwin's need.

tvrusso, 2019-06-05:
That comment has been present since commit  8ef07b in 2003. 

The comments in configure.ac are probably meaningless in this case, and what's actually important is the code and the history --- we require gettext and its libintl on EVERY system precisely because we use it in that one place in main.c (and it's doing absolutely nothing).

It's probably more likely that most linux systems already had it and the person who wrote that line just assumed it was only Cygwin that needed the probe (because it wasn't already installed there and he had to add the probe to get things right --- gettext is so freakin' common on Linux that you can almost assume it's already there if anything nontrivial is installed).  If you took out the probe *and* didn't take out the dependency, it'd puke everywhere where gettext isn't installed.

we7u, 2019-06-05:
Closed by PR #140.

closed 2019-06-05 by we7u