Xastir/Xastir #141: Delete Xastir/m4/ directory and contents?

we7u, 2019-06-05:
Yet another directory that I don't have any experience with. I created a branch where I deleted it: Had to modify these two files to compensate:
  Makefile.am
  configure.ac
With those minor changes Xastir compiles/installs/runs. Not sure what might have broken that I need to check, or whether this was a partial implementation of something we never used.

tvrusso, 2019-06-05:
Yep, looks like an ancient bit of cruft that someone left lying around after it stopped being useful.  Looks like it involves XmHTML libraries, probably something that was played with prior to 2002, hacks added to find libraries and headers (including m4 files needed to deal with it), then just left to rot.

The only information I can find anywhere about XmHTML in commit logs is commits fixing copyright dates and such.  This stuff is clearly unused.

I say get rid of it with extreme prejudice.

we7u, 2019-06-05:
Closed by PR #143

closed 2019-06-05 by we7u