Xastir/Xastir #142: Travis-CI builds don't build Davis or LaCrosse code

we7u, 2019-06-05:
We build the in the src and callpass directories with Travis-CI but we don't build in the Davis or LaCrosse directories. We should include those in the Travis-CI builds so we build all of the code.

tvrusso, 2019-06-05:
That means installing more dependent libraries --- those codes both need mysql libraries.

we7u, 2019-06-05:
Yea, still researching how to do that correctly with Homebrew on OSX.

we7u, 2019-06-05:
I'll probably need Jason's help on this one. I'm getting tired of trying random things via Travis-CI. Would be good if someone who had direct access to OSX could give me the things to add to the Travis-CI config.

godfreja, 2019-06-06:
The Minnesota MS150 bike ride is coming up this weekend so I am fairly busy. I will try to look at this after that event.

we7u, 2019-06-06:
No problemo. What I've done is here:
  https://github.com/we7u/Xastir/tree/travis
The three Linux builds are compiling Davis & Lacrosse but the OSX build isn't. It's somewhat close to working but I can't figure out where the mysql.h include file is that it needs w/o a running system to play with. By the way: The mysql-connector-c package added a trailing "-l" in the link, apparently a known problem, which is why I switched to the mariadb-connector-c package.

we7u, 2019-06-06:
I'm going to commit my Travis-CI config changes to the master project. Everything's working except the Davis/Lacrosse pieces of the OSX build. I can wait until Jason works on that, it's not critical. Davis/Lacrosse pieces are commented-out for OSX so that it compiles, but those pieces are compiling successfully in the Linux builds.

PR #146 (2019-07-03): Enable Davis builds in travis for Mac
Fixes #142
commit 913fdf83 (2019-07-03): Enable Davis builds in travis for Mac

Fixes #142
closed 2019-07-03 by godfreja