Xastir/Xastir #152: Built-in map is 30 years old

df7cb, 2019-07-28:
The built-in map that is shown the first time xastir is lanched is at least 30 years old. It still shows the German Democratic Republic, the Soviet Union, and Yugoslavia. 
[image]
Please update it to something less embarrassing. Thanks.

tvrusso, 2019-07-28:
Yes, that is pretty appalling.  I guess nobody actually looks at that map anymore.

The ancient APRSDOS worldhi.map file that comes with Xastir is simply the only map we had available that can run with the minimal build of Xastir.  It is better than the map that used to come with Xastir before that was found (nothing).

We would be happy to accept any other map that is less embarrassing.  Unfortunately, whatever we take has to work with a minimal build (no shapefile, no imagemagick, no geotiff, no xpm...  that is, a vector map of the most primitive type, like, say, an ancient APRSDOS), or we have to change what constitutes "minimal."

@we7u , correct me if I'm wrong, but I thought the whole reason why we originally created a shapefile fork in the Xastir source tree was so that we could assume shapefile support, precisely so we could distribute a simple and maintainable shapefile map instead of the horrid worldhi.map thing.  But I can find no shapefile map in the Xastir source tree.

If we make shapelib a required package instead of an optional, then it would be much more straightforward to keep a reasonable and not miserably outdated shapefile as our default map.

tvrusso, 2019-07-28:
Correcting myself:  the internal shapelib (since removed) was added to Xastir not so we could have a good default shapefile map, but so that NWS weather alerts would always work (they depend on shapefiles).

I found this in my old archives:

> On Tue, 14 Nov 2006, Jeremy Utley wrote:
> 
> > The main reason for including shapelib by default is so we can have a
> > shapelib-based map come up right from the very start.
> 
> Well, at this point the main reason is to have weather alerts from
> the start, but it still requires the user to run the get-NWSdata
> script (to download and install the NOAA data files) in order to
> make that functional.
> 
> We have a default map in the distribution now but it's a WinAPRS
> map, created by Keith Sproul.  I would have liked a Shapefile map in
> there but didn't find one I liked that was small enough to add to
> the distribution.  If we come across one later we can add it easily
> enough though.
> 
> --
> Curt, WE7U. 

I would suggest that since we removed the internal shapelib (because all distros have it in package management nowadays), we should make it mandatory instead of optional and refuse to build without it, just like we refuse to build without Motif and X11.  Then we could find a decent world shapefile and just include it instead of the awful APRSDOS map.

tvrusso, 2019-07-28:
A possible sources of world shapefiles we might be able to use:

https://www.naturalearthdata.com/downloads/50m-cultural-vectors/50m-admin-0-countries-2/
which has a version on github:
https://github.com/nvkelso/natural-earth-vector

We wouldn't need the whole thing, just national outlines with names (and a dbfawk file to go with it so it could be rendered, after we resolve #128, making dbfawk and pcre required instead of optional).

But it, too, may become outdated quickly.  It's already a couple of years old, it seems.

df7cb, 2019-07-29:
I heard a lot of good things about naturalearth, so I'd think that would be a good choice.

we7u, 2019-07-29:
+1 on Switching to Shapefile world map with only countries in it.

I don't have any good sources in mind for one. Maybe ESRI has a
downloadable Shapefile map with country outlines that's kept up-to-date and
could be distributed under GPL?

One of my thoughts: We could bring up the APRSdos world map, trace over it
with an Xastir object, with corrections for where it's out of date. Save
the track to a Shapefile. Not sure if that's a viable method, but it would
give us our own map we could distribute under any license we want. Would
probably take some editing of the final file to make it work. Again, not
sure whether that's a viable method.

tvrusso, 2019-07-29:
On Mon, Jul 29, 2019 at 06:45:46AM -0700, we recorded a bogon-computron collision of the <notifications@github.com> flavor, containing:
> +1 on Switching to Shapefile world map with only countries in it.
> 
> I don't have any good sources in mind for one. Maybe ESRI has a
> downloadable Shapefile map with country outlines

Yes, they do

>  that's kept up-to-date and

yes, they do

> could be distributed under GPL?

Nope.

> One of my thoughts: We could bring up the APRSdos world map, trace over it

*ugh*  that sounds like an extremely tedious, error prone, and generally
icky thing to do.

we7u, 2019-07-29:
Something like this probably. Defines 247 countries. Different than
the next entry on the page which is sovereign states. Haven't tried it
yet, but downloaded and looked at the zip file. It's a Shapefile.
http://www.naturalearthdata.com/downloads/50m-cultural-vectors/

we7u, 2019-07-29:
Tried it. Works. Expands to 2.2MB's. Might be another one that's more
detailed yet to try.

we7u, 2019-07-29:
I like this one the best:  ne_10m_admin_0_countries.zip but it's 4.7 MB's
when tar gzipped. The next best is the ne_50m map at 757k, but when you
zoom in it's pretty crappy like the DOS map.

tvrusso, 2019-07-29:
Yeah, the naturalearthdata was the one I mentioned several comments above.  They also have a github repo (it's about 7 comments up).

we7u, 2019-07-29:
I propose we add the 10m NaturalEarthData map to Xastir, as long as the
licensing allows. Also that we include a dbfawk file appropriate to it. If
there's a way to get it via github, perhaps that NaturalEarthData map could
be added somehow to our Git build as well so we always get the latest?

tvrusso, 2019-07-29:
It could potentially be added as a git submodule.  But that would cause the entire git repo to be cloned, which is probably too much.  Better just to have a single shapefile to distribute, and we can just periodically update it when naturalearth does.

License is "public domain" and they don't even require that they be credited or cited (but provide text and logos to do so if desired).

https://github.com/nvkelso/natural-earth-vector/blob/master/LICENSE.md

tvrusso, 2019-07-29:
Then again, naturalearth has 104 open issues (mostly complaints about bad data at the local level) and hasn't been updated in a year.  Eventually, this too will become obsolete.  But at least it'll be a shapefile, and we can use ArcGIS  or some other shapefile editor to hack on it if we needed to merge or rename a country that ceases to exist in the next few years.

halmueller, 2019-07-29:
QGIS (https://qgis.org) is the FOSS4G tool of choice for viewing spatial data, and for manipulating it with a GUI. If you need something scriptable, look at GDAL.

kg4wsv, 2019-08-02:
Late to the party here.

I've been playing with the 50 and 110 series files (I sort of started under the assumption the goal was to replace worldhi with something comparable in detail and file size, just up-to-date). I've started on some dbfawks.

I've hit several glitches in some of the data files (random straight lines, e.g. between points on rivers in NM connecting to the southern part of South America). Xastir is my only GIS software so I don't really have any way to check to see if it's a data issue or an xastir issue. Suggestions?

I'll have to double-check, but I think there are duplicate dbfawk signatures on some of the files so the dbfawks would need to be name-based.

Beyond the "up to date data" requirement, are there any other requirements for file sizes, etc that anyone has in mind?

-Jason

tvrusso, 2019-08-02:
On Thu, Aug 01, 2019 at 06:45:59PM -0700, we recorded a bogon-computron collision of the <notifications@github.com> flavor, containing:
> Late to the party here.
> 
> I've been playing with the 50 and 110 series files (I sort of started under the assumption the goal was to replace worldhi with something comparable in detail and file size, just up-to-date). I've started on some dbfawks.

Nice.

> I've hit several glitches in some of the data files (random straight lines, e.g. between points on rivers in NM connecting to the southern part of South America). 

Weird.  In which file are you seeing that NM to SA glitch?  That one sounds 
bizarre.  

> Xastir is my only GIS software so I don't really have any way to check to see if it's a data issue or an xastir issue. Suggestions?

To call Xastir "GIS software" stretches the definition a bit.

Hal suggests QGIS as the definitive open source solution -- I haven't used
that in years myself, but I remember from when I still kept it around
that it was actively developed and was at the very least a good viewer for GIS 
data, and could be connected to GRASS GIS (which I also have stopped using a 
few years ago) for serious GIS tweaking.  I have a home license of ArcGIS 
(which I began using when our SAR organization started trying to get more 
serious about incident mapping, and kept around even though I'm now out of SAR 
for good), and might go ahead and pull one or two in to look there, especially
to look for the weird data glitches you're describing.

> I'll have to double-check, but I think there are duplicate dbfawk signatures on some of the files so the dbfawks would need to be name-based.

Not necessarily --- if they all have the same signature, they all might
be able to use the same dbfawk for rendering, unless the different files 
use the same values of columns for completely unrelated features.  

> Beyond the "up to date data" requirement, are there any other requirements for file sizes, etc that anyone has in mind?

Curt is now less concerned about filesize than I was at first.

A git clone of Xastir is 44MB as "du" tells me.  I'd say adding 10% to that
for a better default map is about the largest anyone would stand for, but 
perhaps even that isn't too big.  It's not like we've still got a lot of
users using slow dialup for their downloads.

tvrusso, 2019-08-02:
Zowiewow.  I had (not seriously) suggested that we could make naturalearth a git submodule of Xastir, so that git users would just get it all.

Glad I wasn't serious.  A git clone of that repo is freakin' huge.  It's downloaded 1.23GB of data so far, and that's only 14% of the clone.

kg4wsv, 2019-08-02:
On Thu, Aug 1, 2019 at 9:02 PM Tom Russo <notifications@github.com> wrote:

>
> > I've hit several glitches in some of the data files (random straight
> lines, e.g. between points on rivers in NM connecting to the southern part
> of South America).
>
> Weird. In which file are you seeing that NM to SA glitch? That one sounds
> bizarre.

ne_50m_rivers_lake_centerlines IIRC.

There are several glitches in that file. Lots of them are smaller, but a
few stand out. I looked at the data with dbfinfo and all records looked
like features and not some sort of extra thing causing stray lines.

Oddly ne_50m_rivers_lake_centerlines_scale_rank seems to be the same type
data and no obvious glitches.

It's not the only file, but the other(s) I saw yesterday and I've slept
since then.  IIRC one of them was political (countries or states?) at one
of the other scales.  I've mostly looked at coastlines, lakes, rivers,
admin countries, and admin states.

-Jason
kg4wsv

kg4wsv, 2019-08-02:
I also thought about labels, but what size display should I design for for general use? I've got a 800x480 7" pi all the way up to my 24" linux desktop at work.

Something else I noticed, this data has language support  - names are in a dozen or two languages. Is there some sort of environment variable or flag passed in to dbfawk for runtime internationalization?

-j

tvrusso, 2019-08-02:
Unfortunately, there is no such existing flag or environment variable for DBFAWK internationalization.  One could, of course, add such a thing.  DBFAWK has pretty much been unmodified ever since Alan Crosswell added it back in 2003 (a few bug fixes, but no really serious new features).

halmueller, 2019-08-02:
I have some spare time in the next few days and can take a look with QGIS or some other tools. Please add a screenshot to this issue showing a representative artifact, with lat/lon/filename. 

Hal

kg4wsv, 2019-08-02:
Here are two samples.  Files are from 50m_physical set.  ne_50m_rivers_lake_centerlines is shown in black, ne_50m_coastlines is shown in blue. I've been saying "glitch" but I believe it's a data error. Note there's at least one error on the coastline data and lots on the river data.

[image]

[image]

tvrusso, 2019-08-02:
I just pulled 50m_physical/ne_50m_rivers_lake_centerlines, 50m_physical/ne_50m_coastline, and 50m_cultural/ne_50m_admin_0_countries into ArcGIS and see absolutely none of these lines.

In fact, they look clean.  (attaching 3 map renderings of various pieces, with graticule overlaid)
[world.pdf](https://github.com/Xastir/Xastir/files/3462540/world.pdf)
[NorthAmerica.pdf](https://github.com/Xastir/Xastir/files/3462542/NorthAmerica.pdf)
[EuropeLabeled.pdf](https://github.com/Xastir/Xastir/files/3462544/EuropeLabeled.pdf)

But in fact, it appears that the "river in NM" you're looking at is actually the Colorado river in Arizona, and it is in present as a single shape combined with the Colorado river in Argentina.  When they generated the shapefile, they must have merged features with identical names.

[ColoradoRiverUS.pdf](https://github.com/Xastir/Xastir/files/3462589/ColoradoRiverUS.pdf)
[ColoradoRiverArgentina.pdf](https://github.com/Xastir/Xastir/files/3462590/ColoradoRiverArgentina.pdf)

Xastir is doing exactly what it's told, connecting all points of a line together.

I think the only way to fix *that* issue is to change Xastir's shapefile rendering to skip drawing lines between points a certain distance apart (expressed in degrees, probably), possibly controlled by a dbfawk option.  That would be some coding.

tvrusso, 2019-08-02:
Notably, it's shape number 155 that has both a US Colorado river and an Argentina Colorado river combined into one shape.  There's a second shape, shape number 5, that is also Colorado River, but only has extent in North America.

tvrusso, 2019-08-02:
Note, this sort of data error is somewhat deadly to Xastir's shapefile rendering optimizations (such as rtree), which are supposed to restrict its reads of shapefiles to shapes whose bounding boxes intersect the current display extents.  An enormous, discontinuous polyline like this means an enormous bounding box that pretty much guarantees the shape will be read no matter where in the western hemisphere you might be.

tvrusso, 2019-08-02:
FWIW, the 10m river/lake shapefile does NOT have the same data error of combining the North American Colorado River with the one in South America.  It may have other issues, such as being about 13MB between the .shp and .dbf file.  And of course it could have some other ridiculous data error, too.

The 10m coastline is pretty, but it, too, is quite large (over 7MB for the shp and dbf files).

We could potentially work to clean these up and feed back the cleanup to the natural-earth git repo as issue reports.  There are a lot of such errors showing up in your second image, such as ridiculous straight lines connecting bits and pieces of the Rio Grande where there shouldn't be any lines.

It could also be that we're handling polylines in Xastir incorrectly.  ogrinfo is showing the Colorado River shapes as type "MULTILINESTRING" which *tells* us that there are multiple, unconnected lines in them.  We appear to be treating all SHPT_ARC shapefiles (as indicated from the Shapelib function "SHPGetInfo") as if they're plain "LINESTRING".  

Scanning over map_shp.c, we do not appear to be respecting the "nParts" element of a shape object returned by SHPReadObject, which would tell us the difference between these two types.  We go through great lengths to take care of POLYGON shapefiles that are in multiple parts, but do no checking of a multipart line --- we treat the list of vertices as if it was a single shape to draw.

tvrusso, 2019-08-02:
Confirmed, the code starting around line 2972 of map_shp.c is very clearly constructing a single polyline out of all of the vertices present in the shape object, paying no regard whatsoever to the fact that the shape might be in multiple parts.  It should only be doing that if nParts is 0, and if nParts is nonzero, it should loop over parts and make separate polylines out of the vertices, using the start points in the panPartStart array to determine which part starts at what vertex offset in the full list.

But the code here is clearly structured as assuming a single part, and some careful refactor should be done.

It would help if bug #128 were resolved before doing that work --- the presence of dbfawk/non-dbfawk code in this file makes it extraordinarily difficult to read (and the non-dbfawk code is much more of the file than the dbfawk code).

I'm going to open a new issue for the improper rendering of muiltipart polylines.  I will certainly not be able to work on either bug #128 or that bug anytime soon, but if we get it into the tracker maybe someone will leap at it and do the work.

halmueller, 2019-08-02:
There's nothing wrong with the data, nothing to "clean up". Multipart strings are common.

In the interest of getting a better default map out sooner rather than later, I suggest converting the MULTILINESTRING elements to single LINESTRING elements. 

I note:
% ogrinfo -al  ne_50m_coastline.shp|grep MULTI|wc -l
1
% ogrinfo -al  ne_50m_rivers_lake_centerlines.shp|grep MULTI|wc -l
182

I don't have a working/built Xastir repo at the moment to test this fix. @kg4wsv or @tvrusso, could you try these commands (assuming you have GDAL/OGR installed):
% ogr2ogr -f "ESRI Shapefile" -nlt LINESTRING -explodecollections -lco ENCODING=UTF-8 exploded_rivers.shp ne_50m_rivers_lake_centerlines.shp
% ogr2ogr -f "ESRI Shapefile" -nlt LINESTRING -explodecollections -lco ENCODING=UTF-8 exploded_rivers.shp ne_50m_rivers_lake_centerlines.shp

and then try those files in Xastir?

kg4wsv, 2019-08-02:
On Fri, Aug 2, 2019 at 2:33 PM Tom Russo <notifications@github.com> wrote:

> FWIW, the 10m river/lake shapefile does NOT have the same data error of
> combining the North American Colorado River with the one in South America.
> It may have other issues, such as being about 13MB between the .shp and
> .dbf file. And of course it could have some other ridiculous data error,
> too.
>

In this particular case there's ne_50m_rivers_lake_centerlines_scale_rank.
It has about 4x the data (by file size and number of records), but looks
more or less the same at the world/continent view.

> The 10m coastline is pretty, but it, too, is quite large (over 7MB for the
> shp and dbf files).
>

Yeah, the 10m stuff looks nice, but i think it's too big to roll into the
distro by default.

I'll continue to mess with the 50m data, specifically continue to view the
different datafiles and more or less arbitrarily decide if I like the looks
of them enough to make a dbfawk.  I'll also look at adding labels.

-Jason
kg4wsv

tvrusso, 2019-08-02:
Yes, there's nothing wrong with the data (I had already concluded that and said so after initially postulating that there was a data error).  Xastir is rendering it incorrectly.  Multipart shapefile lines are not processed correctly.  I have already opened bug #154 for that.

The ogr2ogr explosion command you give does indeed break up all the MULTILINESTRING shapes into individual LINESTRINGs, and should no longer render with incorrect connecting lines in Xastir.

Doing this has another advantage besides getting Xastir to render nicely, it lets the rtree optimization (which is based on individual shape bounding boxes, and doesn't know anything about multi-part shapes, either) select more appropriate subsets of the shapefile for rendering.

BTW, your two ogr2ogr commands are identical.  I presume you meant to apply one to the river/lake centerlines and the other to the coastlines?

halmueller, 2019-08-02:
Yes, fixed it, thanks.

kg4wsv, 2019-08-02:
confirmed, ogr2ogr fixed issues with both files.

tvrusso, 2019-08-02:
FWIW, while the existence of MULTILINESTRING features in this file is not a data error, and the failure to render these multilines as separate polylines is a bug in Xastir, it is certainly wrong for two rivers with the same name on different continents to be lumped into the same shape.

So while the Rio Grande rendering errors (with straight line segments between the parts) in Xastir are not the sign of a data error, one can definitely argue that the natural-earth shapefile should be "cleaned up" to separate the two completely unrelated "Colorado" rivers into separate shapes (each of which might very appropriately be a multilinestring) --- that might be a reasonable issue report for the natural-earth github repo at  https://github.com/nvkelso/natural-earth-vector

kg4wsv, 2019-08-06:
On the question of map file sizes, what would be an acceptable size to include in the distribution?

On my first pass I've whittled it down to a set of about half a dozen of the 50m files that includes countries, states, rivers, and populated places.  The files are ~16.5M uncompressed, 3.1M in a gzipped tarball. It's slightly more detailed than worldhi.map; it includes more in the way of rivers and lakes, has labels, and lists larger cities.  It does not include any roads. I was shooting for "roughly the same but maybe a little better than worldhi".

tvrusso, 2025-12-11:
Wahoo!

I have implemented the simple label point selection method of issue #291, and it works wonderfully for the Natural Earth Vector 50m_cultural/ne_50m_admin_0_countries shapefile.  I've even crafted a nice new dbfawk for it, and the world looks like this:

[image]

I've even got it set up to draw the labels for smaller countries only when you're zoomed in closer, because the Natural Earth shapefiles have a "MAX_LABEL" field that I've mapped onto Xastir's label_level dbfawk variable.

[image]
and further in...

[image]

I added a set of colors to Xastir's color palette based on the ColorBrewer "color advisor for cartography" "Set1" color palette, and mapped Natural Earth's "MAPCOLOR9" value to the 9-element Set1 colors.  I think this palette looks nice, though apparently it is not colorblind friendly.  I also added the "cividis" colorpalette, which is what the EU mandates for accessible public sector maps, but I think those colors are pretty ugly, ranging from navy blue through an olive green down to yellow.
We can fight that out later, or let users change the dbfawk to suit.

When PR #293 is merged, we can possibly move on adding the Natural Earth Vector shapefiles as included maps with Xastir, to replace the ancient worldhi.map.

tvrusso, 2025-12-11:
FWIW, this is my DBFAWK file for the Natural Earth Vector 50m_cultural/ne_50m_admin_0_countries shapefile:
```
BEGIN {
dbfinfo="featurecla:scalerank:LABELRANK:SOVEREIGNT:SOV_A3:ADM0_DIF:LEVEL:TYPE:TLC:ADMIN:ADM0_A3:GEOU_DIF:GEOUNIT:GU_A3:SU_DIF:SUBUNIT:SU_A3:BRK_DIFF:NAME:NAME_LONG:BRK_A3:BRK_NAME:BRK_GROUP:ABBREV:POSTAL:FORMAL_EN:FORMAL_FR:NAME_CIAWF:NOTE_ADM0:NOTE_BRK:NAME_SORT:NAME_ALT:MAPCOLOR7:MAPCOLOR8:MAPCOLOR9:MAPCOLOR13:POP_EST:POP_RANK:POP_YEAR:GDP_MD:GDP_YEAR:ECONOMY:INCOME_GRP:FIPS_10:ISO_A2:ISO_A2_EH:ISO_A3:ISO_A3_EH:ISO_N3:ISO_N3_EH:UN_A3:WB_A2:WB_A3:WOE_ID:WOE_ID_EH:WOE_NOTE:ADM0_ISO:ADM0_DIFF:ADM0_TLC:ADM0_A3_US:ADM0_A3_FR:ADM0_A3_RU:ADM0_A3_ES:ADM0_A3_CN:ADM0_A3_TW:ADM0_A3_IN:ADM0_A3_NP:ADM0_A3_PK:ADM0_A3_DE:ADM0_A3_GB:ADM0_A3_BR:ADM0_A3_IL:ADM0_A3_PS:ADM0_A3_SA:ADM0_A3_EG:ADM0_A3_MA:ADM0_A3_PT:ADM0_A3_AR:ADM0_A3_JP:ADM0_A3_KO:ADM0_A3_VN:ADM0_A3_TR:ADM0_A3_ID:ADM0_A3_PL:ADM0_A3_GR:ADM0_A3_IT:ADM0_A3_NL:ADM0_A3_SE:ADM0_A3_BD:ADM0_A3_UA:ADM0_A3_UN:ADM0_A3_WB:CONTINENT:REGION_UN:SUBREGION:REGION_WB:NAME_LEN:LONG_LEN:ABBREV_LEN:TINY:HOMEPART:MIN_ZOOM:MIN_LABEL:MAX_LABEL:LABEL_X:LABEL_Y:NE_ID";

dbffields="NAME_LONG:MAPCOLOR9:MAX_LABEL:LABEL_X:LABEL_Y";}

BEGIN_RECORD {key=""; lanes=1; color=8; name=""; filled=1; fill_style=2; fill_stipple=0; pattern=0; display_level=2147483647; label_level=50000; label_color=8; symbol=""; font_size=3; label_method=1;}

/^NAME_LONG=(.*)$/ {name="$1";}

# Color each polygon based on its "MAPCOLOR9" value.  Use the Set1 colors

/^MAPCOLOR9=1$/ {fill_style=2; fill_stipple=0; fill_color=127;}
/^MAPCOLOR9=2$/ {fill_style=2; fill_stipple=0; fill_color=128;}
/^MAPCOLOR9=3$/ {fill_style=2; fill_stipple=0; fill_color=129;}
/^MAPCOLOR9=4$/ {fill_style=2; fill_stipple=0; fill_color=130;}
/^MAPCOLOR9=5$/ {fill_style=2; fill_stipple=0; fill_color=131;}
/^MAPCOLOR9=6$/ {fill_style=2; fill_stipple=0; fill_color=132;}
/^MAPCOLOR9=7$/ {fill_style=2; fill_stipple=0; fill_color=133;}
/^MAPCOLOR9=8$/ {fill_style=2; fill_stipple=0; fill_color=134;}
/^MAPCOLOR9=9$/ {fill_style=2; fill_stipple=0; fill_color=135;}
/^MAX_LABEL=5.(.*)$/ {label_level=60000;}
/^MAX_LABEL=6.(.*)$/ {label_level=40000;}
/^MAX_LABEL=7.(.*)$/ {label_level=30000;}
/^MAX_LABEL=8.(.*)$/ {label_level=20000;}
/^MAX_LABEL=9.(.*)$/ {label_level=10000;}
/^MAX_LABEL=10.(.*)$/ {label_level=5000;}
/^LABEL_X=(.+)$/ {label_lon=$1;}
/^LABEL_Y=(.+)$/ {label_lat=$1;}
```

Here's the one for the rivers:
```
BEGIN {
dbfinfo="scalerank:featurecla:name:note:min_zoom:name_alt:min_label:name_en:label:wikidataid:name_ar:name_bn:name_de:name_es:name_fr:name_el:name_hi:name_hu:name_id:name_it:name_ja:name_ko:name_nl:name_pl:name_pt:name_ru:name_sv:name_tr:name_vi:name_zh:ne_id:name_fa:name_he:name_uk:name_ur:name_zht";

dbffields="featurecla:name";
}

BEGIN_RECORD {key=""; lanes=1; color=26; fill_color=26; name=""; filled=1; pattern=0; display_level=2147483647; label_level=10000; label_color=9; font_size=0; symbol=""; fill_style=0;}

# per-field rules are applied to the dbffields that are read from
# each record.

/^name=(.+)$/ {name="$1";next}
/^featurecla=River/ {lanes=1; font_size=3; pattern=0; next}
/^featurecla=Lake Centerline/ {lanes=1; font_size=2; pattern=1; next}
/^featurecla=Canal/ {lanes=1; font_size=2; pattern=2; next}
```

and here's the one for coastlines:
```
BEGIN {
dbfinfo="scalerank:featurecla:min_zoom";

dbffields="featurecla";
}

BEGIN_RECORD {key=""; lanes=1; color=8; fill_color=26; name=""; filled=1; pattern=0; display_level=2147483647; label_level=10000; label_color=9; font_size=0; symbol=""; fill_style=0;}

/^featurecla=Coastline/ {lanes=1; pattern=0; next}
```

tvrusso, 2025-12-11:
So, work needed to get Xastir to start up with nicer maps than the worldhi.map:

 - we need to include a directory of suitable map data in the repo and install that whole directory to ${pkgdatadir}/maps/natural_earth when Xastir is installed by the user.
   - we need to decide how much map data to have as a default map.  
   - are coastlines and country boundaries enough?  Do we need hydrography?
 - we need dbfawks for each of the shapefiles we choose to include
 - The code in main.c that writes "worldhi.map" into the "selected_maps.sys" file on first time startup needs to be smarter, and only write that if shapefile support is not compiled in.  Otherwise, it should write "natural_earth" to pull in the whole new shapefile directory.

tvrusso, 2025-12-11:
Better?

[image]

[image]

[image]

The US states and counties you see here are from another layer I forgot to turn off that would not be part of this default map set.

tvrusso, 2025-12-11:
> [@tvrusso](https://github.com/tvrusso) thanks for not forgetting about this issue and keeping us updated!

Thanks for not letting us forget.

I think we're very close to having a usable map for first-time users to replace worldhi.map.

If you have comments or complaints about the map I've got so far, please say so.

chicoreus, 2025-12-11:
@tvrusso That's looking good.

tvrusso, 2025-12-11:
Sorry about the permissions.  I have my umask set for "read by nobody but me" and forgot to fix that before uploading the tarball.  I will replace it.

Even so, I can't reproduce what you see.  I went over to a machine other than the one where I keep my main xastir instance and did this:
```
mv .xastir dotxastir_save
cd /usr/local/share
sudo mv xastir xastir_save
cd /where/I/do/my/xastir/build
sudo make install
cd /usr/local/share/xastir
sudo tar xzf ~/NaturalEarthVector.tgz
sudo chmod -R a+rX NaturalEarthVector
```
That gives me an environment that is completely clean, with no old maps or configs, into which I've dumped the new files.

Initial startup looks like this, showing the old map and the microscopic Xastir window:

[image]

After configuring a callsign and initial position:

[image]

The map chooser looks like this:

[image]

I choose the NaturalEarthVector directory and deslect the worldhi.map and get this:

[image]

and zoomed in I get

[image]

I bet the problem is still permissions on your end.  Try that `sudo chmod -R a+rX /usr/local/share/xastir/NaturalEarthVector` and report back if that fixes it.  In the meantime, I'll put a new tarball up without broken permissions.

lizdodd, 2025-12-11:
The mapchooser contains

- GPS/
- Online/
- Online/geogratis/gc/ca
- Online/nationalmap.gov
- worldhi.map

Configure gave me this report

> xastir 2.2.3 has been configured to use the following
> options and external libraries:
> 
> MINIMUM OPTIONS:
>   ShapeLib (Vector maps) .................... : yes
> 
> RECOMMENDED OPTIONS:
>   Xpm / Snapshots ........................... : no
>   GraphicsMagick/ImageMagick (Raster maps) .. : no
>   pcre (Shapefile customization) ............ : yes (PCRE2)
>   Berkeley DB map caching-Raster map speedups : no
>   internet map retrieval .................... : yes (libcurl)
> 
> FOR THE ADVENTUROUS:
>   AX25 (Linux Kernel I/O Drivers) ........... : no
>   libproj (USGS Topos & Aerial Photos) ...... : yes
>   GeoTiff (USGS Topos & Aerial Photos) ...... : no
>   Festival (Text-to-speech) ................. : yes
>   GPSMan/gpsmanshp (GPS downloads) .......... : no
> 
> xastir will be installed in /usr/local/bin.

At the moment I have no other maps installed.
I found that I had changed file ownership to 1001:1001 with permissions 777
so I altered that to root:root still with permissions 777
reindexed the files and lost worldhi.map

Then tried sudo chmod -R a+rX /usr/local/share/xastir/
reindexed and worldhi.map returned.

The terminal gives no errors
```
Indexing maps...
XPM and/or ImageMagick support have not been compiled in.
[repeated 13 more times]
Finished indexing maps
```
I think I need a break at this point.

tvrusso, 2025-12-11:
This is obviously still a work in progress...

lizdodd, 2025-12-12:
I've tried with a new download of the NaturalEarth data, this is the source of the owner 1001:1001.
No joy.
Changed ownership to root.
No joy.
Restarted and reindexed and still no new shapefiles to be seen.

[image]

Hopefully someone else is more successful and will be able to point me past this impasse.

godfreja, 2025-12-12:
Reading the description of what the purpose of the 1:50 scale is meant for maybe this isn't a surprise.

tvrusso, 2025-12-12:
This is in fact intentional, but I understand why it might seem like an error.

My intent was that as you zoom way in, you are probably trying to get more detailed maps.  I found that if I did in fact *have* more detailed maps, the shapes from this small-scale global map were distracting.  For example, say you have road maps for an area.  You generally don't care about seeing all of those in high detail if you're zoomed all the way out so that you see an entire country.  But if you're zooming in to see a route or something, the stippling that colors the entire country gets in the way.

That's why I implemented "min_display_level" in commit f509b4 in the first place --- precisely to make it easier for these things to get out of the way when they were no longer helpful.

Maybe that's not desirable behavior, but I found when using these in conjunction with other maps that it was really more pleasant.

I figure, if you're zooming in to a country so far that the nearly the entire country is in your screen, you don't need to see the shading.  Maybe I didn't get the exact zoom level choices right.

They are controlled by the `min_display_level` variables of the dbfawk, which are set based on the "MIN_ZOOM" field of the shapefile's attributes.  I just sort of came up with a mapping between MIN_ZOOM values and `min_display_level` values by a careful process of rectal extraction.

closed 2025-12-12 by tvrusso

lizdodd, 2025-12-12:
I've done another git fetch, recompiled with almost all map types enabled and I still am in a map-free environment. There remains no option to choose the NaturalEarth maps and I don't even have online maps downloading yet. 
My reason for playing Xastir today was to check the data flow through the Wx3in1 on the network, and that is working.

[image]

lizdodd, 2025-12-12:
I managed to see the new maps running as root, so I deleted the entire ~/.xastir directory for a restart.
With the continent boundary "Australia" in blue and the major island boundaries  eg "Tasmania" in green, it looks a bit odd.

[image]

I also find that the stippling is present at too high a zoom level. I'm not actually sure the number of the zoom level.
[image]

tvrusso, 2025-12-12:
Indeed, *I* don't see Australia and Tasmania in different colors:

[image]

Oh, and I just reread your last post as well, and I think the hand-editing of the config files is the clue about why things weren't showing up --- you installed Xastir and the shapefiles were given the time of install for their last modification dates.  You then hand-edited config files, making them newer than the files.  "Add new maps" wouldn't think the maps were new, because they were older than your hand-edited files.   "Reindex ALL Maps" should have done it, though.

tvrusso, 2025-12-12:
> I did notice an oddity as I zoomed in.
> 
...
> Zoom in more and the country boundaries disappear: <img alt="Image" width="1421" height="813" src="https://private-user-images.githubusercontent.com/1238497/525690066-0b535415-dfeb-4c86-a7a9-c3867dcc2393.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjU1NTc2ODgsIm5iZiI6MTc2NTU1NzM4OCwicGF0aCI6Ii8xMjM4NDk3LzUyNTY5MDA2Ni0wYjUzNTQxNS1kZmViLTRjODYtYTdhOS1jMzg2N2RjYzIzOTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEyMTJUMTYzNjI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2YxNWRkN2Q5NjQxOWU4ZDIxOTk0OGNlZjQ4Yzk2ZGI4YTUyMjc4ODcxN2UyM2UxNGMzNDA1NTUzNDAxMjJkYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.XYFQmkXYy3xhlXlkEjlvt4BFE9xBEw-rlTpG5zjLq0M">

One thought for an enhancement:  What if I added a `min_fill_level` variable to dbfawk, so that instead of turning off the shape entirely at low zoom levels, we just stopped filling it below those levels?

This would be a little bit more work in `draw_shapefile_map`, but I think it would be doable.  Then I could change the dbfawk so that instead of setting `min_display_level` it would set `min_fill_level`, and the code would continue to draw those shapes, but only as unfilled polygons (i.e. just draw their boundaries).

Does that seem like a worthwhile enhancement?