Xastir/Xastir #153: "API Key Required" Shows up on OSM_tiled_cycle.geo Maps

KB3EFS, 2019-07-29:
Even with the recent cleaning up of the code and related items regarding API Keys I am still seeing "API Key Required" on OSM_tiled_cycle.geo maps. Perhaps this is a new requirement given the fluid nature of the way OSM has been trending of late. (This is my first time filing a "Git-Issue".) This issue is reproducable. I am running Xastir V2.1.5 (9692556) on
-  uname -m = x86_64
- uname -r = 4.4.0-148-generic
- uname -s = Linux
- uname -v = #174~14.04.1-Ubuntu SMP Thu May 9 08:17:37 UTC 2019

tvrusso, 2019-07-29:
This has nothing to do with Xastir code or the machine you're on.  Clearly, OSM has changed their terms of use for that map, and require an API key to get an unmolested tile set, probably because annoying codes like Xastir keep using their tile servers and bogging them down.  My guess is that eventually they'll not just put a watermark over the image, but deny access entirely unless a proper API key is given.

Fixing this would require figuring out not only how to get an API key for Xastir use, but how to wedge it into the tile requests.

tvrusso, 2019-07-29:
https://support.goldensoftware.com/hc/en-us/articles/360003642753-How-can-I-remove-the-API-key-required-watermark-from-my-OSM-map-layer-?mobile_site=true

tvrusso, 2019-07-29:
I take it back --- it looks like you can get an API key for an app or a web site and have the app or web site use that key for the download.

We'll have to add a keyword for .geo files so that we can specify the key, and add code to pass it down to the OSM code and download manager so they can append the necessary apikey keyword and value.