Xastir/Xastir #156: shapefile defaults do not display maps at large zoom levels

kg4wsv, 2019-08-03:
When zooming out to large (e.g. multiple continents, entire world) zoom levels shapefile maps do not display. creating a dbfawk and setting display_level to an extremely large number in BEGIN_RECORD gives the expected behavior.

Browsing map_shp.c leads me to think that the default maximum display level for a shapefile (without a dbfawk override) is 65536 if xastir is compiled with dbfawk support, and possibly 8192 without dbfawk.

Seems to me that the default should be MAXINT unless there's a range defined elsewhere for display_level (or scale_y?)

in map_shp.c I see:

`/* default dbfawk rule when no better signature match is found */`
...
`static awk_rule dbfawk_default_rules[]`
...
`display_level=65536;`

and later on in map_shp.c:

`void draw_shapefile_map()`
...
`    static int color,lanes,filled,pattern,display_level,label_level;`
 ...
`    display_level=8192;`
...
`         skip_it = (map_color_levels && (scale_y > display_level));`

tvrusso, 2019-08-03:
The defaults could easily be changed, but the entire point of dbfawk
was to offload rendering rules out of the source code and into the 
dbfawk file.  It was presumed that the BEGIN_RECORD line of a dbfawk
would be used to set the defaults for each individual file type.  It was
also presumed that there would be dbfawks for everything.

That said, setting a reasonable "always display" default for maps that *have* 
no dbfawk file would make a lot of sense.

kg4wsv, 2019-08-03:
Assuming the dbfawk will handle it is reasonable, but for me the situation violates principle of least astonishment. My maps didn't display when I zoomed out, I didn't understand why, looked at the code to figure it out, then looked at the code some more to figure out what the maximum should be.

I think the argument could be made that a dbfawk has no need to set a maximum display level at all, since map properties handles that for all map types, not just shapefiles.  I'm certainly not saying anything crazy like remove support for display_level from dbfawk/map_shp, but I think the default should be "always display everything unless the user says different".

tvrusso, 2019-08-03:
Just did some git paleontology, and the "display_level=8192" line you point to in draw_shapefile_map was put in because it is necessary to re-initialize the dbfawk structures to prevent prior maps' dbfawk info from being applied to a new map (because the new map didn't set some variable or other).  It was a real problem with the code before commit 4999d108 in 2004 --- some dbfawk that didn't set every possible attribute in its BEGIN_RECORD was applied to a map that was drawn *after* one that had a more unusual BEGIN_RECORD setting, and wound up getting rendered funny depending on what prior maps you turned on or off.

Clearly, the default used for this reinitialization was never carefully aligned with other defaults.  There should really be only one set of values used for defaults, and they should be the same as the ones in the "dbfawk_default_rules" struct.  Hell, perhaps they should just be taken OUT of the default rules struct so there's only one place where constants are enumerated.  The block of code beginning around line 804 of the current version of map_shp.c (as of commit 6d3c8174) should probably just pull data out of the static dbfawk_default_rules instead of having hard-coded defaults.

I agree, the default should be to always display unless a dbfawk overrides.  Your suggestion of using the maximum int value is a good one, and if you care to do it, test it out to make sure it works in an intuitive manner, and make a PR ...  (the "test it out" part is why I'm not leaping to do it myself, even though it amounts to about two lines of changes in map_shp.c)

kg4wsv, 2019-08-03:
Yeah, I'll take it.  It may take a day or three.

Portability question: INT_MAX is defined in limits.h isn't provided by the
current set of include files. (apparently the cool kids prefer this over
MAXINT, at least according to /usr/include/values.h)

Would using the compiler defined __INT_MAX__ be preferable to adding
#include limits.h?

Are there autoconf (or other) ramifications to adding limits.h ?

-Jason, portability noob

tvrusso, 2019-08-03:
On Sat, Aug 03, 2019 at 11:44:15AM -0700, we recorded a bogon-computron collision of the <notifications@github.com> flavor, containing:
> Yeah, I'll take it.  It may take a day or three.
>
> Portability question: INT_MAX is defined in limits.h isn't provided by the
> current set of include files. (apparently the cool kids prefer this over [...]

No, that would not be preferrable.  It would be best to use the correct
symbol from limits.h (not values.h, which isn't a standard header and isn't
present on all systems).  https://stackoverflow.com/questions/53985418/c-is-limits-h-cross-platform

I think you are safe to require a standards compliant C compiler these days,
and the days of hunting around for non-standard alternatives seem to be over.

Go ahead and include limits.h in the map_shp.c file, and use INT_MAX.

> Are there autoconf (or other) ramifications to adding limits.h ?

Xastir's autoconf.ac already checks for the existence of limits.h, but I'm
not sure why it bothers, since no place in the code actually looks for
the HAVE_LIMITS_H macro that this probe would set, and no bit of code to
date has included it.  Go figure.  Looks like we've got a lot of that, 
testing for the existence and usability of tons of header files we don't even
use, and then blissfully proceeding whether they exist or not.

If you're concerned about compatibility with ancient OSen and compilers where
limits.h might not define INT_MAX, you could use this pattern in map_shp.c:

#ifdef HAVE_LIMITS_H
#include <limits.h>
#define MAX_ZOOM INT_MAX
#else
#define MAX_ZOOM 65536
#endif

and then use MAX_ZOOM where you want to set the display_level default.  
configure is setting HAVE_LIMITS_H if limits.h is found, and I'm proposing 
that you simply presume that it actually contains an INT_MAX definition.  One
could go overboard and actually check that with AC_CHECK_DECL, but it seems
way too much effort for the number of archaic systems that might still not
have this standard definition.  Any modern system with a C99 standard compiler
will have that.

If you don't do this ifdeffing, then the lack of limits.h should be an error 
at configure time, so you would have to change the AC_CHECK_HEADERS thing to 
error out if limits.h isn't found.

tvrusso, 2019-08-03:
There is a lot of weirdness in how zoom levels are defined and what the "entire world" zoom level is.  The pull-down menu in the right-click "zoom level" menu has an "Entire World" selection, which invokes this set of lines in main.c's "Zoom_level" function:

```
      case(7):

//WE7U
// Here it'd be good to calculate what zoom level the entire world
// would best fit in, instead of fixing the scale to a particular
// amount.  Figure out which zoom level would fit in the X and the Y
// direction, and then pick the higher zoom level of the two (to
// make sure the world fits in both).  We should probably center at
// 0.0N/0.0W as well.

        new_scale_y = 500000;

        // Center on Earth (0/0)
//                new_mid_x = 12900000l / 2;
//                new_mid_y = 6480000l  / 2;
        break;

So in fact, 65536 ain't right for a default of "always display", but when I actually click "Entire World" Xastir tells me I'm at zoom level 75k, not 500k.  And it doesn't center the map, as Curt's comments say it doesn't: the result is that zooming to "Entire World" shows me somewhat less than the whole world, and a big blank spot to the left.
```

kg4wsv, 2019-08-05:
I've noticed an inverse relationship between the zoom level that shows the entire world is larger when the window is smaller - e.g. on the computer in front of me the world zoom is 84k if the window is full height, but if I resize the window so that it's only a couple inches tall, world zoom is 336k.

tvrusso, 2019-08-06:
Related to bug #152

tvrusso, 2023-07-27:
This is so trivial I'm just going to do what was last suggested, using INT_MAX as the default display_level instead of 8192.

8192 was the default in code for a long time, then 65536 was put into the dbfawk_default_rules struct.  Unfortunately, it's a *string* in dbfawk_default_rules, so not easy to make dependent on a defined constant.  I'm going to leave it as 65536 there and update the code.

I'm going to wait until #207 is approved and merged, though, so I don't have multiple PRs and multiple branches to complicate the merge.

tvrusso, 2023-07-27:
I'm going to set display_level to 2147483647 in default_dbfawk_rules and INT_MAX in the code.

That will make it behave fine if an int is at least 16 bits, which it will always be now.   65536 in the default rules is way to conservative when an int is 16 bits or more and would be an error if one were running on an absolutely ancient system where an int was a mere 8 bits --- I rather doubt Xastir would be running in any such retro computer anyway.  This will at least allow the thing to work correctly at all zoom levels if no dbfawk file was found, and still work at all zoom levels even if a dbfawk file existed that failed to set display_level anywhere.

We already know that 65536 is too small to cover all zoom levels when windows are small (c.f. [the earlier comment](https://github.com/Xastir/Xastir/issues/156#issuecomment-518055402)) so we should at least make it less wrong.  That default rule is only used when no dbfawk with matching signature is found, but it means that for really high zoom levels such a shapefile would not display.

commit a06e74df (2023-07-27): Merge pull request #208 from tvrusso/issue-156

Issue 156 -- Fix default display_level to be consistent and huge

Closes #156
closed 2023-07-27 by tvrusso