Xastir/Xastir #157: shapefile polygon labels can be located poorly

kg4wsv, 2019-08-06:
Shapefile labels for polygons can appear nowhere near the object in question. The existing algorithm uses the center of the bounding box for the polygon.  This results in, for example, the label for "United States" being placed over France due to Alaska being on the other side of xastir's world.

This is mentioned in the code (line 4362 of map_shp.c).  I'm opening an issue because it will become more of an issue as more shape files are available and so there's a git issue associated with the problem.

tvrusso, 2019-08-06:
On Mon, Aug 05, 2019 at 08:00:57PM -0700, we recorded a bogon-computron collision of the <notifications@github.com> flavor, containing:
> Shapefile labels for polygons can appear nowhere near the object in question. The existing algorithm uses the center of the bounding box for the polygon.  This results in, for example, the label for "United States" being placed over France due to Alaska being on the other side of xastir's world.

My guess is that this shape is a multipart polygon shape, and 
this is just more evidence of Xastir not correctly dealing with those.

There is some work done in the map_shp.c to deal with the fact that there
are multiple "rings" in some polygons, with clockwise coordinates denoting
polygons to fill, and counterclockwise represent "holes" inside those.
Each part of a multipart polygon shape *is* drawn separately and not connected
like multipart polylines are, but only a single label is created for the whole
shape.

It might be possible to base the label location on the centroid of the
"fill" rings rather than the entire shape's bounding box --- putting one 
label in each fill ring rather than one in the entire shape, which, as you say,
is Just Wrong for a multipart polygon.

> This is mentioned in the code (line 4362 of map_shp.c).  I'm opening an issue because it will become more of an issue as more shape files are available and so there's a git issue associated with the problem.

There is no longer a line 4362 of map_shp.c after merge of PR #155 (removal
of non-dbfawk code from Xastir).  I can't find any obviously relevant comment
at line 4362 of the version that existed prior to that merge, either.
Which comment are you referring to here?

kg4wsv, 2019-08-06:
On Mon, Aug 5, 2019 at 10:23 PM Tom Russo <notifications@github.com> wrote:

> This is mentioned in the code (line 4362 of map_shp.c). I'm opening an
issue because it will become more of an issue as more shape files are
available and so there's a git issue associated with the problem.

There is no longer a line 4362 of map_shp.c after merge of PR #155 (removal
of non-dbfawk code from Xastir). I can't find any obviously relevant comment
at line 4362 of the version that existed prior to that merge, either.
Which comment are you referring to here?

Sorry, typo. That should have been 4632.

       // XXX - todo: center large polygon labels in the center (e.g.
county boundary)
        /* center label in polygon bounding box */

tvrusso, 2019-08-06:
BTW, it would be useful to know which shapefile you're working on that exposed this issue, so that if anyone wants to start working on it they can try to reproduce your issue and then try to make it look nice.

tvrusso, 2019-08-06:
Related to issue #152 (trying to capture bug interdependencies)

kg4wsv, 2019-08-06:
Examples from Natural Earth data:

50m_physical/ne_50m_geography_marine_polys.shp - North Pacific Ocean example

50m_cultural/ne_50m_admin_0_countries - United States / France examples

tvrusso, 2025-12-07:
For what it's worth, I played a bit with the polylabel idea I mentioned in the [previous comment](https://github.com/Xastir/Xastir/issues/157#issuecomment-3619671775).  Not in Xastir, just in a toy program.

It was not difficult to craft a C-callable function in C++ that implemented a  simple polylabel use (pass it double arrays of X and Y coords, a total number of points, a number of "parts" (rings), an array of "part start" indices, and variables to return the POI).  I even wrote a quick hack that can read a shapefile with a single feature and return its POI -- my little hack will even then create a point shapefile with that POI, which I can display in QGIS.    I then extracted only the one record from the natural-earth-vector/50m_cultural/ne_50m_admin_0_countries shapefile that was for the United States and ran my hack on it.

However, when I extracted the single shape for the United States from the natural-earth-vector/50m_cultural/ne_50m_admin_0_countries.shp file, it finds a POI that is in Alaska.  This is because the shapefile has a single record in the shapefile with 127 parts ("rings") for the United States, and if you use all of those parts and pass them to the polylabel function as separate rings, that's just what you get.

It's still orders of magnitude better than having the label appear in France, but it's not ideal.   

My idea of creating a label for each non-hole polygon ring is probably better, but then every single island in Hawaii or off the coast of Alaska will get labeled as "United States of America."  Given that this file *only* contains country boundaries, that's probably OK.  

Another option might be check the area of each ring and only generate its POI for label plotting if it's big enough.  The threshold might be hard to figure out.  Alternatively, calculate the areas of all rings and only label, say, the two largest?  Or maybe only the largest?

Such a thing might be a lot more processing.  However, a good refactor of the polygon code in Xastir *might* enable us to do POI computations once, perhaps at the same time we build the shapefile hash and the RTree structure for the file.  Then, the up-front cost of checking the area of all polygon rings before shipping them off to the POI calculator might not be so bad.

This, by the way, is what I got by turning the output of polylabel into a point shapefile and overlaying it on the single shape for the United States in QGIS.  You can see the POI marked somewhere around Mitkoff Island in Alaska.

[image]

Now, there are Issues with linking the C-interfaced C++ file (you either have to do the final link with the C++ compiler so you get the correct C++ libraries linked in, or you have to make a shared library out of the C++ file and link it in with Xastir).  We can work that out later.

tvrusso, 2025-12-07:
*facepalm*

I just looked at the DBF fields for ne_50m_admin_0_countries.

There are fields for "label_x" and "label_y" that are basically intended as the place where a feature label should be placed.  The values in those fields are pretty much the same as the ones that polylabel computed for the "largest ring" case.

So the *simplest* approach here would be to add DBFAWK variables for just such use and to use them instead of the center of the bounding box.  There is some work to do to get that done (one would have to have a fall-back when the DBFAWK file doesn't provide any label hint location), and it definitely can't be touched until the draw_shapefile_map stuff is more manageable.   But if the shapefile itself has fields that tell us where the label should go, there should be a way to use that information.

Finding a better default placement when the shapefile DOESN'T throw us a bone is still important.  But this might be the quicker way to get past #152 .