Xastir/Xastir #162: get-gnis

texnusn, 2019-12-02:
another small correction for the scrips. this one is for get-gnis. 

line 28 
SUFFIX=_Features_20180601
should be just
SUFFIX=_Features

tested and works after change
ROBERT

godfreja, 2019-12-03:
I just tried that change with Minnesota, Wisconsin, and California. It seems like the change to be to have a suffix of _Features_20191101.

Did it complete all the way to installing in the GNIS directory with just _Features?

texnusn, 2019-12-03:
For the TX yes it did. I looked directly at the website, that us what it
called for was just features. It unzipped to the current date text file.

texnusn, 2019-12-03:
This is a copy and paste from the terminal after I made the change to the script.

Retrieving GNIS file for TX
--2019-12-03 16:32:18--  https://geonames.usgs.gov/docs/stategaz/TX_Features.zip
Resolving geonames.usgs.gov (geonames.usgs.gov)... 137.227.239.220, 2001:49c8:8000:121d::76
Connecting to geonames.usgs.gov (geonames.usgs.gov)|137.227.239.220|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4501628 (4.3M) [application/zip]
Saving to: ‘TX_Features.zip’

TX_Features.zip     100%[===================>]   4.29M  1.38MB/s    in 3.1s    

2019-12-03 16:32:21 (1.38 MB/s) - ‘TX_Features.zip’ saved [4501628/4501628]

Archive:  TX_Features.zip
  inflating: TX_Features_20191101.txt  
File for TX not successfully downloaded.

godfreja, 2019-12-03:
The "File for TX not successfully downloaded." is the part I was wondering about - you hit the same problem I did. The change I have staged for commit fixes that. I plan on leaving the PR open for a bit in case there is any comments and then merge it either later tonight or tomorrow.

texnusn, 2019-12-04:
i think i have figured it out. 
Retrieving GNIS file for TX
--2019-12-03 19:14:05--  https://geonames.usgs.gov/docs/stategaz/TX_Features.zip
Resolving geonames.usgs.gov (geonames.usgs.gov)... 137.227.239.220, 2001:49c8:8000:121d::76
Connecting to geonames.usgs.gov (geonames.usgs.gov)|137.227.239.220|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4501628 (4.3M) [application/zip]
Saving to: ‘TX_Features.zip’

TX_Features.zip     100%[===================>]   4.29M  97.5KB/s    in 29s     

2019-12-03 19:14:36 (149 KB/s) - ‘TX_Features.zip’ saved [4501628/4501628]

File successfully downloaded. Moving to /usr/local/share/xastir/GNIS

what do i need to do to properly send the script for your review

I added an MV command to do a rename so it could pass its validation and a couple of of options to the unzip command.

godfreja, 2019-12-04:
Thanks for the effort. I already have some changes ready to merge that removes the date stamp dependency for downloading zip files. You can see the changes by looking at the pull request under the second comment for this issue ("Update gins..."). If you are happy with those changes we can just go with those. This is probably the easiest option.

Otherwise, if you would like to submit your changes, the "official" way is described in https://github.com/Xastir/Xastir/blob/master/CONTRIBUTING.md#contributing-code-changes-to-the-project. This would require some familiarity with using git or a willingness to learn. We can then go with whichever of our change looks most appropriate. If that is too much we can figure out a way for you to send a diff.

Either option is fine by me.

PR #163 (2019-12-09): Update get-gnis to not use date stamps for zip files
Zip files can now be downloaded without a date stamp,
but the unzipped file still has a date stamp in the filename.
Determine the unzipped filename on the fly so the date stamp
does not need to be hard coded.

Closes #162
commit 12dba3d1 (2019-12-09): Update get-gnis to not use date stamps for zip files

Zip files can now be downloaded without a date stamp,
but the unzipped file still has a date stamp in the filename.
Determine the unzipped filename on the fly so the date stamp
does not need to be hard coded.

Closes #162
closed 2019-12-09 by godfreja