Xastir/Xastir #173: configure checks for deprecated pj_init()

bwarden, 2021-03-03:
By proj 7.2.1, `pj_init()` was listed as a deprecated function. In proj 8.x, it's no longer present.

xastir's `configure` checks for the presence of proj via `pj_init()`:

```
checking for pj_init in -lproj... yes
```

This check needs to be updated to use a different proj function.

bwarden, 2021-03-03:
The [migration guide](https://proj.org/development/migration.html) suggests using `proj_create()` or `proj_create_crs_to_crs()`

closed 2021-03-03 by tvrusso

tvrusso, 2021-03-03:
Thank you.  As described in the commit log for 6ee6348, Xastir doesn't even use libproj at all, but probes for it because geotiff needs it.  In older days geotiff was not in any package management systems and therefore had to be hand-rolled by every Xastir user --- this usually resulted in static libraries instead of shared libraries, and therefore proj needed to be linked in explicitly if you wanted geotiff support.

This is in general never necessary anymore because almost all package systems have geotiff and it is a shared library that pulls in proj as a run-time dependency, not a link time dependency.  But we retain the capability for those who wish to do things the hard way.

Our configure now first checks for proj_create_crs_to_crs.  If it finds it, great, we have libproj.  If it doesn't find that it looks for pj_init --- and if it finds that it means we have an old proj and we still use it.

Xastir is full of stuff backward-compatibility stuff like that, because some of our users run it on very old systems and can't always run the latest, greatest packages. The backward compatibility lingers many years past its sell-by date, but it does mean that the lastest Xastir can generally work on systems that people collect up in junkyards, garage sales, and hamfests.

tvrusso, 2021-03-03:
Good thing I set this up for backward compatibility.

Ubuntu 18.04 and 16.04, both of which we're testing in our Travis setup, have old proj libraries that do NOT have proj_create_crs_to_crs, but do have pj_init.  The system on which I build has the newer one.  It compiles in both places.

We'll continue supporting the old proj as well as newer versions for a lot longer than it should be necessary.

tvrusso, 2021-03-03:
Ubuntu 20.04 has a more recent proj and I just added it to our travis config to verify that it, too, works.  It does.

commit 6ee6348c (2021-03-03): Update proj probe to look for modern function

Recent versions of libproj deprecated the decades-old PROJ.4 API in
favor of a much more flexible and modern version.

Xastir has always looked for one of the functions (pj_init) in the original API
in its configure probe.  Since this function will be going away in
proj 7 we can't just keep doing that.

Now, configure probes for the function proj_create_crs_to_crs
instead.  If it finds it, we have a newer libproj and we use it.  If
it doesn't, it tries again with pj_init -- and if it finds *THAT* then
we have an older libproj and use it.

Xastir d [... cut]