Xastir/Xastir #174: relay_digipeat causes core dump

mobilinkd, 2021-03-05:
```
$ rpm -q xastir
xastir-2.1.4-8.fc33.x86_64
```

relay_digipeat causes Xastir to crash when compiled with stack fortification because:

```
sprintf(small_string,"relay_digipeat: outputs:\n\tport: %d\n\tcall: %s\n\tdest: %s\n\tpath: %s\n\tinfo: %s\n",
          port, call, destination, new_path, info);
```

Overflows the 200 byte buffer of small_string when trying to create the following string:

```
relay_digipeat: outputs:\n\tport: 6\n\tcall: K9IQP-13\n\tdest: APWW10\n\tpath: WX9O-3*,WIDE2-1\n\tinfo: }KD9QEN>APWW11,TCPIP,K9IQP-13*:}KD9QEN>APWW11,TCPIP,K9IQP-13*:@185408h4220.59N/08807.61W-PHG5100APRS-IS f
```

Which is 209 bytes long.

Backtrace follows.

```
#0  0x00007faa96c4d9d5 in raise () from /lib64/libc.so.6
#1  0x00007faa96c36954 in abort () from /lib64/libc.so.6
#2  0x00007faa9728090b in MagickPanicSignalHandler () from /lib64/libGraphicsMagick-Q16.so.3
#3  <signal handler called>
#4  0x00007faa96c4d9d5 in raise () from /lib64/libc.so.6
#5  0x00007faa96c368a4 in abort () from /lib64/libc.so.6
#6  0x00007faa96c90177 in __libc_message () from /lib64/libc.so.6
#7  0x00007faa96d214ca in __fortify_fail () from /lib64/libc.so.6
#8  0x00007faa96d1fd66 in __chk_fail () from /lib64/libc.so.6
#9  0x00007faa96c8893f in _IO_str_chk_overflow () from /lib64/libc.so.6
#10 0x00007faa96c94441 in _IO_default_xsputn () from /lib64/libc.so.6
#11 0x00007faa96c7e24b in __vfprintf_internal () from /lib64/libc.so.6
#12 0x00007faa96c889e4 in __vsprintf_internal () from /lib64/libc.so.6
#13 0x00007faa96d1f84f in __sprintf_chk () from /lib64/libc.so.6
#14 0x00005632636e7b7e in sprintf (
    __fmt=0x5632637b43d8 "relay_digipeat: outputs:\n\tport: %d\n\tcall: %s\n\tdest: %s\n\tpath: %s\n\tinfo: %s\n", 
    __s=0x7ffc9dd8d4a0 "relay_digipeat: outputs:\n\tport: 6\n\tcall: K9IQP-13\n\tdest: APWW10\n\tpath: WX9O-3*,WIDE2-1\n\tinfo: }KD9QEN>APWW11,TCPIP,K9IQP-13*:}KD9QEN>APWW11,TCPIP,K9IQP-13*:@185408h4220.59N/08807.61W-PHG5100APRS-IS f")
    at /usr/include/bits/stdio2.h:38
#15 relay_digipeat (call=<optimized out>, path=<optimized out>, info=<optimized out>, port=<optimized out>) at db.c:21124
#16 0x00005632637a71e1 in decode_ax25_line.constprop.0.isra.0 (line=line@entry=0x7ffc9dd8f2e0 "K9IQP-13", from=from@entry=84 'T', 
    port=6, dbadd=1) at db.c:21441
#17 0x000056326373050a in UpdateTime (clientData=0x56326444d2f0, UNUSED_id=<optimized out>) at main.c:13816
#18 0x00007faa975114c8 in XtAppProcessEvent () from /lib64/libXt.so.6
#19 0x00007faa975065c3 in XtAppMainLoop () from /lib64/libXt.so.6
#20 0x00005632636cc4ea in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at main.c:30580
```

tvrusso, 2021-03-05:
Sigh.  This is some debugging output that was added to the code in 2003 to replace an fprintf to stderr with this sprintf (commit 59798908).  In later years more care was taken to use snprintf (and variants thereof) to prevent such buffer overflows, but not here.

What's really weird is that all this effort is taken to stuff information into "big_string" (partially by way of small additions to "small_string") but this string is then never actually output (the fprintf is commented out).

The ideal would be to remove all that debugging stuff altogether rather than leave a mix of broken string operations followed by and output statement that is commented out.  This has been in the code for almost 18 years now in this broken state.  It is not clear that any of this debugging output has been at all relevant since 2008, the last time any substantive changes have been made to relay_digipeat.  My preferred solution would be to remove it.

If there is great objection to removing such old debugging code, then the second choice would be to fix the sprintfs so that they are buffer-overrun safe snprintfs instead.  And as a last resort, all of the stuff involving "small_string" and "big_string" in the relay_digipeat function could just be commented out, since it's not doing anything anyway (I am strongly opposed to commenting out code and leaving it in place and would remove it, but mine is not the last word here).

This is not likely to get anyone's rapt attention anytime soon.  If you would like to take a stab at fixing this by replacing the sprintfs with appropriate snprintfs that prevent buffer overrun, please do.  We will gladly review any pull request you make to help along these lines.

If this is really causing you trouble, just comment out all the lines that make use of big_string and small_string anywhere in that function, as the strings themselves are never, ever used.

closed 2021-03-05 by tvrusso

tvrusso, 2021-03-05:
I've merged the associated pull request, and taken care of removing the few remaining sprintfs and buffer declarations that could have caused problems for others.

commit 41eca5e6 (2021-03-05): Merge pull request #175 from mobilinkd/master

Remove buggy, unused diagnostic code.

Closes #174