Xastir/Xastir #178: For some reason i am getting parts of the packet data messed up

KA5TTZ, 2021-05-27:
[image]

I am running it on a Raspberry Pi 3.
Then TNC is a PITNC-9K6 using the teensy 3.2 board
I have set everything up per the documents. I have change the Baud rate from 9600 to 1200 since i am on 144.390.
I have the jp1 set to 1200. 
The radio is a Yaesu FT857D so i can use the 6 pin plug for both the radio and the TNC. If anyone know the issues or where i can go to find out what is going on please let me know. This is the third time I have in stall the software. Each is using a new sd card and new Raspbian os.
73
KC5DP

tvrusso, 2021-05-27:
This looks like an incorrect baud rate in the connection to the tnc (the baud rate for communication to the TNC can be (and usually is) different from the baud rate the TNC uses over the air, which is correctly 1200 baud on VHF).

Check that your  communication rate with between the Pi and the TNC is correct.

tvrusso, 2021-05-27:
This _really_ sounds more like an issue with TNC settings than with Xastir, and you might have better luck contacting a group that supports the PiTNC-9k6 with getting the settings correct for communication with Xastir.

From the look of it at http://www.tnc-x.com/TNCPi9k6, there is a recommendation for how to set the on-air baud rate using pitnc_getparams, but the TNC-to-Pi serial port speed is done through raspi-config.  Make sure that the baud rate there corresponds to the baud rate you've set in the Xastir interface box (or vice-versa).

tvrusso, 2021-08-24:
@KA5TTZ have you managed to get this working or to diagnose it any further?  I am inclined to close this issue because your woes do not appear to be an Xastir bug.