Xastir/Xastir #179: KISS-over-TCP Port

leonardinux, 2021-08-23:
KISS TNCs can be accessed over a TCP/IP connection. Xastir miss this port type support. There are many software that can interface KISS modem to TCP ports, such as ser2net or share-tnc or tnc-server (all source are hosted here in GitHub). How can be defined a remote TCP link to a KISS device in Xastir? Simply read a TCP address that read/write KISS frames... but I cannot find a solution into Interfaces menu...
Any trick? have I missed a previous thread about it?

73 de Leo IZ5FSA

tvrusso, 2021-08-24:
There is no capability built into Xastir to connect to a KISS-over-TCP port.  One could clearly be added, but is is unlikely that anyone has the cycles to do it anytime soon.

That said, there may be simple ways to get the job done, but you'll have to involve an intermediary.

One could cobble together a kludge using "socat" to connect a KISS-over-TCP TNC to a pseudo TTY, then connect Xastir to the PTY as if it were a serial KISS TNC.  See http://www.dest-unreach.org/socat/doc/socat.html for the man page.  Simply have socat talk to the KISS tnc's TCP/IP port for one of its interfaces and a PTY for the other, then tell Xastir to use the slave side of the PTY as if it were a serial KISS TNC.

leonardinux, 2021-08-24:
Thanks!!!
Googling a bit I've found this solution:
$ socat PTY,link=/tmp/tnc TCP:192.168.1.10:8001 &
This command "pipe" /tmp/tnc to a network TCP address.
In Xastir you can use a Serial KISS TNC on /tmp/tnc device and all can work!
I think I'll work around this feature to create a "script" that can be
started/stopped to render it fully automated by Xastir itself.
Stay tuned!

[image: immagine.png]

73 de Leo IZ5FSA

leonardinux, 2021-08-31:
Hi OM,
I've create this script to run my Xastir and it seems to be ok:
`#!/bin/bash
socat PTY,link=/tmp/tnc TCP:192.168.1.10:8001 &
sleep 1
xastir
sleep 1
kill -SIGTERM $(pidof socat)
#
`
73 de Leo IZ5FSA

closed 2021-09-10 by leonardinux

tdhart, 2024-04-11:
Thanks for the info, this was very helpful. YAAC includes the option to connect to a network KISS TNC, I wish Xastir did. With your workaround, it does now.

tdhart, 2024-04-11:
> Hi OM, I've create this script to run my Xastir and it seems to be ok: `#!/bin/bash socat PTY,link=/tmp/tnc TCP:192.168.1.10:8001 & sleep 1 xastir sleep 1 kill -SIGTERM $(pidof socat)
> 
> ` 73 de Leo IZ5FSA

Thank you Leo!
This was very helpful. YAAC includes the option to connect to a network KISS TNC, I wish Xastir did. With your workaround, it does now!