Xastir/Xastir #189: Problem sending packet with xastir_udp_client

deejdeej, 2022-01-16:
I'm using an APRS extension in WeeWX to generate APRS formatted weather packets, for example:

/161958z4355.97N/06956.34W_125/...g001t021r000h56b10134RaspberryPi-Xastir-Wx

When I run the xastir_udp_client command to send this packet out via Xastir, it adds a "}" to the beginning of the APRS packet which causes sites to complain "[Unsupported packet format]", ie:

2022-01-16 15:01:06 EST: N1JOV-7>APX219,KQ1L-1,KQ1L-8*,WIDE2,qAR,N1QQA-10:}/161958z4355.97N/06956.34W_125/...g001t021r000h56b10134RaspberryPi-Xastir-Wx [Unsupported packet format]

Making the following changes in main.c corrects this issue:

$diff main.c main.c.orig 
13519a13520
>               *(line + line_offset - 1) = '}';
13522c13523
<                 (char *)(line + line_offset),  // Raw data line
---
>                 (char *)(line + line_offset - 1),  // Raw data line

 - Remove line 13520, "*(line + line_offset - 1) = '}';"
 - Modify line 13522 to remove the " - 1"

The resulting section of code looks like this:

// Change to a third-party packet.  In this case we know we have a
// line_offset, so backing up one to insert a char is ok.

              output_my_data(
                (char *)(line + line_offset),  // Raw data line
                -1,    // ports, -1=send out all interfaces
                0,     // type: 0=cooked, 1=raw
                0,     // loopback_only
                0,     // use_igate_path
                NULL); // path

Packets now sent with xastir_udp_client are accepted by other sites.

tvrusso, 2022-01-16:
How, exactly, are you using the udp client to inject this packet?

The code that inserts the "}" is there to create third-party packets, and is only supposed to happen when the callsign in the packet doesn't match Xastir's call sign.  If it's being inserted incorrectly, perhaps you're not specifying the callsign in the injected packet correctly.

Please show us exactly how you're creating the packet, and how you're invoking xastir_udp_client, and we will better be able to tell whether the suggested code change is correct.

Sadly, the main.c code is a holy mess of commented-out code that makes it difficult to read, and some of the commented-out code has been commented out since 2005.  You may be correct that inserting that brace is not the right thing to do, but comments in the code suggest that it is *supposed* to be there precisely to get the third-party packet correctly formatted.  Others have used the xastir_udp_client code to inject packets into Xastir for years, so it would surprise me if it were so completely broken.

tvrusso, 2022-01-16:
It appears that while there are comments indicating the packet should only be third-party if the callsign of the incoming packet doesn't match the callsign of the Xastir instance, all that code is commented out and the packets are unconditionally sent as third party.

That means that you MUST provide a *complete* packet, not just the payload.  In the example packet you gave in the initial description, you didn't have either a FROM, TOCALL, or path and only had the payload.  You need all of those (though the path can be null, of course).  So, for example, I just did this:

```
xastir_udp_client 127.0.0.1 2023 KM5VY 21753 -to_rf -to_inet "KM5VY>APZ0:@162200z3458.19N/10618.28W_270/002g010t047r000P000p000h29b10173"
```

And Xastir transmitted this:
```
KM5VY>APX219,WIDE2-1:}KM5VY>APZ0:@162200z3458.19N/10618.28W_270/002g010t047r000P000p000h29b10173
```
And aprs.fi correctly shows this as a valid packet. 

So when correctly inserting a full packet via xastir_udp_client, Xastir's inserting the brace is in fact correct.  So we won't likely want to mess with the code that inserts the brace as you suggest doing.

Try:

```
xastir_udp_client 127.0.0.1 2023 N1JOV-7 <yourpass> -to_rf -to_inet "N1JOV-7>APX219:/161958z4355.97N/06956.34W_125/...g001t021r000h56b10134RaspberryPi-Xastir-Wx"
```
*without* your modified main.c, and it should work correctly.  Your packet will be treated by Xastir as a third-party packet, but it'll be transmitted in a valid APRS packet.

closed 2022-01-16 by tvrusso

tvrusso, 2022-01-20:
Just for the record and to round out my response here a little:  I just looked up the WeeWX aprs extension, and see that the example they use to inject their output data into APRS uses the "beacon" feature of aprx.

This aprx feature apparently adds a source and destination address to the payload before transmitting.  xastir_udp_client doesn't do that, so you have to do it yourself.

So where the weewx aprs extension's documentation has this:
```
Example of integration in the aprx configuration:
    <beacon>
    beaconmode both  # Send packet via APRS-IS and radio.
    cycle-size  5m
    beacon srccall N0CALL-1 via WIDE2-1 file "/dev/shm/aprs.pkt"
    </beacon>
```

and assuming you've got everything *else* about that extension set up as in https://github.com/cavedon/weewx-aprs, you should do this:
```
  xastir_udp_client <hostname> <port> <srccall> <passcode> -to-rf -to-inet "SRCCALL>DESTCALL:"`cat /dev/shm/aprs.pkt`
```

where "SRCCALL" is your call sign/SSID and "DESTCALL" is any valid APRS ToCall (use APX219 to say your third-party packet is from Xastir).

The main point is that xastir_udp_client does not add to or modify the data passed to it in any way --- it is meant to inject complete APRS packets into Xastir, and as such does not add the "SRCCALL>DESTCALL:" part if you've left it out.  

All the problems you've had with this are because of that misunderstanding.  I just dug through old Xastir mailing lists on this issue and see that you first posted a question about it in 2018 and this was never clarified in that thread.  Perhaps this points to a problem in documentation of xastir_udp_client that I will look into.

tvrusso, 2022-01-20:
Looks like the only documentation of xastir_udp_client is in its man page, installed when you build Xastir.  That only says:  "
```
EXAMPLES
        xastir_udp_client localhost 2023 <callsign> <passcode> "APRS Packet
       Goes Here"

       Currently that will inject the packet into Xastir's decoding routines
       and send it to any TCP-connected clients. It will also igate it to the
       INET if you have igating enabled.  It will send the packet out the RF
       ports as third-party packets only if you add the "-to_rf" flag after
       the passcode like this:

        xastir_udp_client localhost 2023 <callsign> <passcode> -to_rf "APRS
       Packet"

       The UDP client is useful for generating and injecting APRS packets from
       external scripts.  It can also be used to fetch the callsign of the
       remote xastir server by using the -identify flag:

        xastir_udp_client localhost 2023 <callsign> <passcode> -identify

NOTES
       This is a very simple utility that provides no validation of message
       content.
```

Which leaves unstated the fact that it feeds Xastir the "APRS Packet Goes Here" text unmodified.  I will augment that.

tvrusso, 2022-01-20:
Now the documentation says this in the NOTES section:

```
NOTES
       This is a very simple utility that provides no validation of message
       content.  The text passed as the last argument is directly injected
       into Xastir's incoming data stream and processed exactly as if it had
       come out of a TNC.

       As such, the text passed to it as "APRS Packet" must be a complete APRS
       packet, including FROM and TO call signs, not simply the payload.  Thus
       to have it work properly, you should pass "MYCALL-0>APX219:payload goes
       here" as the string, not "payload goes here" (where obviously you
       should replace "MYCALL-0" with your own callsign and SSID).
```

deejdeej, 2022-01-20:
Tom, thank you for looking into this.  I haven't had the time to test yet but your comments make sense.