Xastir/Xastir #190: xastir.org down

PhantomNomad, 2022-03-03:
Was trying to get to the documentation but it looks like xastir.org is down.  Anyone know what's happening?

tvrusso, 2022-03-03:
@we7u is the guy who runs xastir.org, and it has gone down a number of times recently (either because of DNS issues or otherwise).  He is the only one with control of it, and the only one who can fix it.

I am not sure how much time he has to pay attention to github issues, so I'll email him privately and let him know what's going on.

we7u, 2022-03-03:
DNS. I kicked the master named server (Santa Cruz, CA). John Ronan and I
will work to assure 2 of the slave DNS servers (in Ireland) are accessible
from outside the VM facility. There's a third slave server also (Bellevue,
WA): I'll have the guy who owns that machine re-check that one.

I won't be able to do much for a couple of hours. Am still at work. However
it's resolving now.

godfreja, 2022-03-03:
IF managing DNS servers is becoming a pain point, I have had success letting Cloudflare or Hurricane Electric handle my DNS needs for free.

we7u, 2022-03-04:
Xastir.org has been up for 190+ days. Looks like it was perhaps a CentOS7
update recently that caused the master "named" to fall over. After 2 days
the secondary servers stopped serving the zone.

Master is back up (Santa Cruz), one secondary server is serving up the
zones just fine (Bellevue, WA). #2 and #3 slave servers are not (Ireland).
Will work with John Ronan when he wakes up to assure the routing for DNS
UDP packets actually makes it to the VM's. VM's named's and firewalls
appear to be set up correctly and the named's are serving up the zones
locally, but not outside the VM's. Should get that fixed-up within the next
day or two to get 1 primary and 3 secondary servers fully functional. If I
can't get #2 and #3 to receive the packets I'll set up yet another VM
that's running on different hardware (but also in Ireland).

we7u, 2022-03-04:
Heard from John Ronan this morning. He got external DNS UDP packets to
route to the VM and it started working. The .6 machine is not working yet
but he had to run. I reorganized the DNS list to pull the .6 machine from
the zone record. All three servers in the zone are serving up DNS
responses. Tested each one with "dig".

closed 2022-05-04 by tvrusso