Xastir/Xastir #193: NO MOTIF HEADERS FOUND

Jujified, 2022-05-03:
I am trying to install Xastir on Macos. When I get to this command
 >../configure CPPFLAGS="-I/usr/local/opt/berkeley-db@4/include" LDFLAGS="-L/usr/local/opt/berkeley-db@4/lib" 
it throws this error:

>"checking for Motif headers... configure: error: **** NO MOTIF HEADERS FOUND **** install Motif  development headers or use ->-with-motif-includes to specify location of Xm/Xm.h"

I tried using the flag/s --with-motif-includes and --with-motif-lib than specified the path to my x11 folder(in homebrew's cellar). It gives these warnings 
>"configure: WARNING: you should use --build, --host, --target
>configure: WARNING: invalid host type: /usr/local/opt/homebrew/cellar/openmotif/2.3.8_1/lib" than this error "checking build >system type... /usr/local/opt/homebrew/cellar/openmotif/2.3.8_1/lib
>configure: error: invalid value of canonical build"

Please help! Thank you.

tvrusso, 2022-05-03:
Please post the exact command you used to configure with "--with-motif-includes" and "--with-motif-lib".  The correct syntax should be "--with-motif-includes=/path/to/where/motif/includes/are --with-motif-libs=/path/to/where/motif/libs/are".  If you left out the equals sign that could have caused the error you're seeing, but it's hard to be sure without seeing the exact command you used.

tvrusso, 2022-05-03:
If you can't get *that* to work, you could always try just adding the include and lib directories to CPPFLAGS and LDFLAGS just as you are for Berkeley DB:
```
  CPPFLAGS="-I/usr/local/opt/berkely-db@4/include -I/usr/local/opt/homebrew/cellar/openmotif/2.3.8_1/include" LDFLAGS="-L/usr/local/opt/berkeley-db@4/lib -L/usr/local/opt/homebrew/cellar/openmotif/2.3.8_1/lib"
```
which effectively does the same thing that the "--with-motif-" options do in the end.  I'm assuming the layout of your openmotif package, and might be off a little.  Note that Xastir's configure is looking for a file "Xm/Xm.h" in your search path, and if there is no such file (as in, if the file is just Xm.h instead of being in a subdirectory) it'll be confused and not work.  Headers for X-related packages are all supposed to be organized as subdirectories of the X11 directory (wherever that is) and custom packages that don't do that will confuse standard Motif codes like Xastir.

So if you try both the --with-motif-headers and CPPFLAGS thing that I suggest and it still doesn't work, you may have some kludges to implement.  But let's burn that bridge when we come to it.

godfreja, 2022-05-03:
I actually dealt with this last weekend trying to compile xastir on a new Mac. Tom's suggestion would help. but you will just hit another problem after. Configure isn't looking for any of the homebrew includes. Here is the configure command that worked for me:

./configure 'CPPFLAGS=-I/opt/homebrew/include -I/opt/homebrew/opt/berkeley-db@4/include' 'LDFLAGS=-L/opt/homebrew/lib -L/opt/homebrew/opt/berkeley-db@4/lib'

tvrusso, 2022-05-03:
Ah.  This is actually a fairly common problem --- configure doesn't look for headers anywhere but where the C compiler looks, and CPPFLAGS "-I" flags simply tell the C compiler extra places to look.   On systems where headers live anywhere other than /usr/include you have to add to CPPFLAGS.  On my system, third party packages always go in /usr/local, so I have to have "-I/usr/local/include" in CPPFLAGS all the time.  

But a Homebrew-specific installation note on the wiki sounds like it's in order.

godfreja, 2022-05-03:
I believe this is a semi-new wrinkle with homebrew. It used to install into /usr/local but moved to /opt/homebrew. On my old Mac I never had to specify extra paths to configure.
Agreed that some wiki updates are in order.

tvrusso, 2022-05-03:
Hmmm.  Funny you should say that.  I *thought* I always needed "-I/usr/local/include" and "-L/usr/local/lib" for Xastir, but I don't.  That's because Xastir's configure has some goofy, nonstandard stuff that adds /usr/local/include and /usr/local/lib to the default search path automatically.  

I need to add CPPFLAGS="-I/usr/local/include" and LDFLAGS="-L/usr/local/lib" to *most* autoconf projects' configure invocations, but not Xastir's for this reason --- a kludge is already adding it for me.

Xastir does some really weird stuff to look for Motif, too.  Loops over a huge number of possible locations for it, one for each weird place that any obscure system happened to have had back in the late 90s/early 2000s.

There was a big push 20 years ago to make it so that Xastir's configure just worked as "./configure" with no options on every platform that any Xastir user had, which meant it had a lot of weird hard coded things that took care of some of the obscure ones. That was a nice thought, but it meant that it only worked properly on the systems where it's been set up already, and was not in the spirit of autoconf at all --- it is more like the ancient hand-porting of Makefiles that autoconf was designed to replace.  

It also does users a disservice when they come to expect "./configure" to do everything without them having to add system customizations, and then suddenly the system changes out from underneath them or they move to some new one and they have to add new stuff anyway.

So when homebrew changed from a base prefix of /usr/local/ to /opt/homebrew, it stopped benefiting from this particular leftover kludge in Xastir.

tvrusso, 2022-05-05:
@Jujified:  Were you able to get up and running with the combined information from me and Jason?  If not, please post your exact configure line, and perhaps Jason, who has experienced exactly the same issue with Homebrew can post his.  Between the two of those things we should be able to converge on a solution pretty quickly.

Jujified, 2022-05-05:
Yes, I was able to get Xastir up and running. TYSM

closed 2022-05-05 by Jujified