Xastir/Xastir #195: snapshot.png fails

mathisono, 2022-07-11:
Warning: XtRemoveGrab asked to remove a widget not on the list
/usr/bin/gm convert: Improper image header (/root/.xastir/tmp/snapshot.xpm).
System call return error: convert: /root/.xastir/tmp/snapshot.xpm -> /root/.xastir/tmp/snapshot.png
/usr/bin/gm convert: Improper image header (/root/.xastir/tmp/snapshot.xpm).
System call return error: convert: /root/.xastir/tmp/snapshot.xpm -> /root/.xastir/tmp/snapshot.png
Socket connection for interface 2 type (2, 1, 6) failed: Connection refused
This is OK since we have more to try.
/usr/bin/gm convert: Improper image header (/root/.xastir/tmp/snapshot.xpm).
System call return error: convert: /root/.xastir/tmp/snapshot.xpm -> /root/.xastir/tmp/snapshot.png
Socket connection for interface 2 type (2, 1, 6) failed: Connection refused
This is OK since we have more to try.
Socket connection for interface 2 type (2, 1, 6) failed: Connection refused
This is OK since we have more to try.
Socket connection for interface 2 type (2, 1, 6) failed: Connection refused
This is OK since we have more to try.
Socket connection for interface 2 type (2, 1, 6) failed: Connection refused
This is OK since we have more to try.

tvrusso, 2022-07-11:
Why is your Xastir install trying to write snapshots into root's home directory?  Clearly, GraphicsMagick is having trouble doing the conversion, possibly due to a permissions problem.

I use snapshots every day and they always drop into my own home directory.  They have been working correctly for years.

How are you building/installing Xastir, and how are you running it?  There is clearly something wrong.

Please provide more details on your install and environment.  Without more information, it will be impossible to determine what is going on.

The "Socket connection" warnings have nothing to do with snapshots.

mathisono, 2022-07-13:
Im running on debian for a rpi. Im now running xastir as the pi user. The .xastir dir is present and I can access the tmp folder.  I shouldn't have been running this as root! with sudo!!! The permissions for serial device... and that looks to be fixed.

tvrusso, 2022-07-13:
So snapshots work now?

If so, we can close this issue.  Running anything as root is a Very Bad Idea.

tvrusso, 2022-09-03:
Closing as this appears to have been a user error that resulted from trying to run Xastir as root.  No comment from original reporter since changing to a non-root user, but it appears that that fixed the problem.

closed 2022-09-03 by tvrusso